zephyrproject-rtos/zephyr #71451: Is there any interest in USB CDC-NCM driver?

rgrr, 2024-04-12:
Before I waste my time issuing a PR: is there any interest in an CDC-NCM driver for Zephyr?  As far as I understand #42066 it should be.

Some time ago I made a bigger rewrite of the NCM driver for TinyUSB and offered it to the project (https://github.com/hathach/tinyusb/pull/2227).  Til now this PR has not been merged.

I'm not complaining... so another approach: I adopted this driver to Zephyr.  You can check the current status here: https://github.com/rgrr/ncs-games/tree/zperf/ncm-start/zperf/src/netusb

If you are interested in it, I will rewrite it according to your style guide and create a PR.  If not, would also be nice to hear that.

carlescufi, 2024-04-12:
CC @jfischer-no @tmon-nordic 

@rgrr did you target the next USB stack or the current one?

tmon-nordic, 2024-04-12:
> @rgrr did you target the next USB stack or the current one?

The linked code is using the old stack, so the code would have to be not just rewriten to match the style guide, but also ported to the device-next stack.

rgrr, 2024-04-16:
Trying to port to the next USB stack.  Therefor I want to use the zperf sample as test environment.  Unfortunately this does not start the interface when doing

    west build --build-dir _build . --pristine --board nrf52840dk_nrf52840 --no-sysbuild -- -DNCS_TOOLCHAIN_VERSION=NONE -DBOARD_ROOT=. -DCACHED_CONF_FILE=prj.conf -DEXTRA_CONF_FILE=overlay-usbd_next_ecm.conf -DDTC_OVERLAY_FILE=usbd_next_ecm.overlay

log output:

    [19:37:04.899681 0.001739] *** Booting nRF Connect SDK v3.5.99-ncs1 ***
    [19:37:04.900630 0.000950] [00:00:00.381,713] <inf> net_config: Initializing network
    [19:37:04.901997 0.001368] [00:00:00.381,744] <inf> net_config: Waiting interface 1 (0x20000fc4) to be up...

Is this supposed to work out of the box?  If not, how to make it run?

rgrr, 2024-05-03:
If anyone wants to inspect: https://github.com/zephyrproject-rtos/zephyr/pull/72305.  This version of the NCM driver uses usb-next and is kept very simple: only one datagram for RX/TX.

Nevertheless it produces errors with iperf (monitored with Wireshark) and my knowledge seems to be a little bit too limited to understand this fully.  The code does more or less ECM (but packed into NCM) and generates errors which does ECM not.

The code has also been more or less taken from my TinyUSB NCM driver which does not have those problems.

Confuses me...

rgrr, 2024-05-05:
Problem with iperf has been resolved by setting `CONFIG_NET_BUF_VARIABLE_DATA_SIZE=y`.

It is not clear to me, if this has something to do with the host driver or whatelse.

Nevertheless it is now working.

Would be great, if someone could do performance measurements other than nRF52840.  I'm doing my benchmarking with

    for MSS in 10 100 800 1450; do iperf -c 192.168.2.1 -e -i 1 -M $MSS -l 8192 -P 1 ; sleep 2; done

or

    for MSS in 10 100 800 1450; do iperf -c 192.168.2.1 -e -i 1 -M $MSS -l 8192 -P 1 ; sleep 2; done

Performance with the nRF52840 is poor: around 1MBit/s (but that's the case for ECM as well)

JayToltTech, 2025-02-14:
Can we close this issue now @rgrr that ncm is part of the mainline?

jfischer-no, 2025-02-15:
Implemented in 8169ca2e085a101d090cde849d0b6c7a6fffb47f

closed 2025-02-15 by jfischer-no