zephyrproject-rtos/zephyr #71473: test: sample.smp.pi: ls1046ardb/ls1046a/smp/4cores build failure

hakehuang, 2024-04-15:
**Describe the bug**
build failure on sample.smp.pi with  ls1046ardb/ls1046a/smp/4cores

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - v3.6.0-2514-g4a30aab2aefa
 - What have you tried to diagnose or workaround this issue?
 - git bisect found it is after below commit

```
commit 0be0d2175bdbb660f510a6056794569b2527d348
Author: Jamie McCrae <jamie.mccrae@nordicsemi.no>
Date:   Thu Mar 21 09:56:27 2024 +0000

    cmake: modules: extensions: Revert using common board files
    
    This change reverts a change that was introduced with hwmv2 which
    allowed for using common board names for overlays, given the board
    target ``nrf9160dk@0.7.0/nrf9160/ns``.

```
 

**To Reproduce**
Steps to reproduce the behavior:
west build -b ls1046ardb/ls1046a/smp/4cores . -p always samples/arch/smp/pi

**Expected behavior**
build pass

**Impact**
SMP power feature

**Logs and console output**

```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/ls1046ardb_ls1046a_smp_4cores/samples/arch/smp/pi/sample.smp.pi/zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.16.5-1/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm64/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/ls1046ardb_ls1046a_smp_4cores/samples/arch/smp/pi/sample.smp.pi/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm64/core/libarch__arm64__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/toolchains/zephyr-sdk-0.16.5-1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.2.0"  -L/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/ls1046ardb_ls1046a_smp_4cores/samples/arch/smp/pi/sample.smp.pi/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--fatal-warnings  -mcpu=cortex-a72  -mabi=lp64  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_DOUBLE_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/ls1046ardb_ls1046a_smp_4cores/samples/arch/smp/pi/sample.smp.pi/zephyr && /usr/bin/cmake -E true
/opt/toolchains/zephyr-sdk-0.16.5-1/aarch64-zephyr-elf/bin/../lib/gcc/aarch64-zephyr-elf/12.2.0/../../../../aarch64-zephyr-elf/bin/ld.bfd: zephyr/arch/arch/arm64/core/libarch__arm64__core.a(smp.c.obj): in function `arch_cpu_start':
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/arch/arm64/core/smp.c:107: undefined reference to `pm_cpu_on'
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/arch/arm64/core/smp.c:107:(.text.arch_cpu_start+0xe4): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `pm_cpu_on'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-2514-g4a30aab2aefa

JiafeiPan, 2024-04-16:
As Hake mentioned in this ticket, the issue is caused by this commit
https://github.com/zephyrproject-rtos/zephyr/commit/0be0d2175bdbb660f510a6056794569b2527d348
,  it revert some configure and dts overlay introduced in hardware model v2,  in my original PR to covert ls1046a to hwmv2 (
https://github.com/zephyrproject-rtos/zephyr/pull/69416), I got the comments from @nordicjm  that:
 
And the way the hwmv2 configuration works is the base configuration will be applied first, that is nxp_ls1046ardb_defconfig, then nxp_ls1046ardb_ls1046a_defconfig will be applied (does not exist so can be ignored) then nxp_ls1046ardb_ls1046a_smp_defconfig will be applied, then finally nxp_ls1046ardb_ls1046a_smp_4cores_defconfig, so anything in the base files does not need to be set again in the files overlayed on top (note: defconfig files only work like this, only one .dts file is used)
 
but currently with commit 0be0d2175bdbb660f510a6056794569b2527d348, it seems there will be no overlay between these defconfig files,  for example, for the board "ls1046ardb/ls1046a/smp/4cores", it only use ls1046ardb_ls1046a_smp_4cores_defconfig, but not ls1046ardb_ls1046a_defconfig + ls1046ardb_ls1046a_smp_defconfig + ls1046ardb_ls1046a_smp_4cores_defconfig, not sure whether my understand is correct?  @nordicjm could you please help on this? thanks.

nordicjm, 2024-04-16:
> but currently with commit https://github.com/zephyrproject-rtos/zephyr/commit/0be0d2175bdbb660f510a6056794569b2527d348, it seems there will be no overlay between these defconfig files, for example, for the board "ls1046ardb/ls1046a/smp/4cores", it only use ls1046ardb_ls1046a_smp_4cores_defconfig, but not ls1046ardb_ls1046a_defconfig + ls1046ardb_ls1046a_smp_defconfig + ls1046ardb_ls1046a_smp_4cores_defconfig, not sure whether my understand is correct? @nordicjm could you please help on this? thanks.

Correct, same with overlays

JiafeiPan, 2024-04-17:
> > but currently with commit [0be0d21](https://github.com/zephyrproject-rtos/zephyr/commit/0be0d2175bdbb660f510a6056794569b2527d348), it seems there will be no overlay between these defconfig files, for example, for the board "ls1046ardb/ls1046a/smp/4cores", it only use ls1046ardb_ls1046a_smp_4cores_defconfig, but not ls1046ardb_ls1046a_defconfig + ls1046ardb_ls1046a_smp_defconfig + ls1046ardb_ls1046a_smp_4cores_defconfig, not sure whether my understand is correct? @nordicjm could you please help on this? thanks.
> 
> Correct, same with overlays

Thanks for confirmation, have crated RP to fix this issue.

aescolar, 2024-04-23:
#71473 indicates it fixes this. If not please reopen

closed 2024-04-23 by aescolar