zephyrproject-rtos/zephyr #71474: Move GNSS includes from zephyr/drivers/gnss to include directory

jeremyherbert, 2024-04-15:
**Is your enhancement proposal related to a problem? Please describe.**
It currently doesn't seem possible to create an out-of-tree NMEA-based GNSS driver without reimplementing the whole NMEA part. There is some specific functionality related to NMEA parsing in:

- `zephyr/drivers/gnss/gnss_nmea0183.h`
- `zephyr/drivers/gnss/gnss_nmea0183_match.h`
- `zephyr/drivers/gnss/gnss_parse.h`

which is not available on the include path to an out-of-tree application.

**Describe the solution you'd like**
I'd like to be able to create an out-of-tree driver for a GNSS device that can utilise the existing NMEA code. One way to resolve this would be to expose these header files by moving them to the include path.

**Describe alternatives you've considered**
I could copy+paste the entire GNSS directory, but that just means that upgrading zephyr versions becomes more difficult as the code can become out of sync.

bjarki-trackunit, 2024-04-15:
You can add the headers by adding the include directory using the absolute path to drivers/gnss using CMake, then including them like "gnss_nmea0183.h" in your out of tree driver.

These headers are internal, and not expected to be exposed to the application. That's why they are not in include/zephyr :)

jeremyherbert, 2024-04-15:
Thanks, that works too!

closed 2024-04-15 by jeremyherbert