zephyrproject-rtos/zephyr #71475: Add antenna selection to GNSS API

jeremyherbert, 2024-04-15:
**Is your enhancement proposal related to a problem? Please describe.**
Some GNSS modules have the ability to switch between multiple antennas (internal vs external, etc). It would be good if the API supported a way to adjust and report back which antenna was used.

**Describe the solution you'd like**
Change the `gnss_info struct` to something like:

```
/** GNSS info data structure */
struct gnss_info {
	...
	/** The index of the antenna used */
	uint8_t antenna_index;
};
```

If a device only has a single antenna, `antenna_index` would always be set to 0. This data can be parsed by the GNSS callback.

Add a function like:

```
int gnss_set_antenna(const struct device *dev, uint8_t antenna_index);
```

To set which antenna is being used.

**Describe alternatives you've considered**
An out of tree driver for this specific radio, but this still would require changes to the zephyr code anyway.

**Additional context**
Example module is Quectel LC86G, which supports an internal antenna (mounted directly on the module) or an external antenna connected to a pin on the module. Setting the antenna type is done through a command like:

```
$PQTMCFGANTENNA,W,0,0*7E
```

And the radio sends out regular messages like the following which contain the antenna selection status:

```
$PQTMANTENNASTATUS,3,1,2,1*52
```

jeremyherbert, 2024-04-15:
I agree that it seems like a good fit for the devicetree, but unfortunately yes, it is something that I would need to adjust at runtime. In my case, the user can select which antenna they want to used based on the device location/installation method. It may also be the case that the device can try both antennae to see if one can get a better fix than the other.

bjarki-trackunit, 2024-04-15:
We will need to enumerate the antennas, binding index 0 to a specific antenna, like 0=internal, 1=external0, ...

a dts schema like
````````
```
```
gnss {
        antennas {
                internal@0 {
                };
                external@1 {
                };
        };
};
```
`
```````