zephyrproject-rtos/zephyr #71476: mgmt.mcumgr.fs.mgmt.hash: lpcxpresso55s69/lpc55s69/cpu1: build failure

hakehuang, 2024-04-15:
**Describe the bug**

<html>
<body>
<span><span class="ui-provider a b c d e f g h i j k l m n o p q r s t u v w x y z ab ac ae af ag ah ai aj ak" dir="ltr">
CASE_NAME | ZEPHYR_VERSION | RESULT | PLATFORM | TEST_PATH
-- | -- | -- | -- | --
mgmt.mcumgr.fs.mgmt.hash.supported.crc32 | v3.6.0-2514-g4a30aab2aefa | Build failure | lpcxpresso55s69/lpc55s69/cpu1 | tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported
mgmt.mcumgr.fs.mgmt.hash.supported.sha256 | v3.6.0-2514-g4a30aab2aefa | Build failure | lpcxpresso55s69/lpc55s69/cpu1 | tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported
mgmt.mcumgr.fs.mgmt.hash.supported.all | v3.6.0-2514-g4a30aab2aefa | Build failure | lpcxpresso55s69/lpc55s69/cpu1 | tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported

</span></span>
</body>
</html>

**To Reproduce**
Steps to reproduce the behavior:
twister -b lpcxpresso55s69/lpc55s69/cpu1 -T tests/subsys/mgmt/mcumgr -S fs_mgmt_hash_supported

**Expected behavior**
Build PASS

**Impact**
flash management

**Logs and console output**

```
[2/163] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/shared/disk/zephyr_project/zephyr_test/zephyr), build: v3.6.0-2519-gf7e49cd074e9
[112/163] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj
FAILED: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj 
ccache /home/ubuntu/zephyr-sdk/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCPU_LPC55S69JBD100_cm33_core1 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DZCBOR_ASSERTS -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/build/zephyr/include/generated -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/soc/nxp/lpc -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/soc/nxp/lpc/lpc55xxx/. -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/mgmt/mcumgr/util/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/mgmt/mcumgr/transport/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/testsuite/include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/subsys/testsuite/ztest/include -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/cmsis/CMSIS/Core/Include -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/modules/cmsis/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69 -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/devices/LPC55S69/drivers -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_gpio -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/pint -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/inputmux -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/iap1 -I/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_iocon -I/home/shared/disk/zephyr_project/zephyr_test/zephyr/modules/hal_nxp/. -I/home/shared/disk/zephyr_project/zephyr_test/modules/lib/zcbor/include -isystem /home/shared/disk/zephyr_project/zephyr_test/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33+nodsp -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/ubuntu/zephyr-sdk/zephyr-sdk-0.16.4/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/shared/disk/zephyr_project/zephyr_test/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/shared/disk/zephyr_project/zephyr_test=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj -MF zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj -c /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/soc_flash_mcux.c
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/soc_flash_mcux.c:41:2: error: #error No matching compatible for soc_flash_mcux.c
   41 | #error No matching compatible for soc_flash_mcux.c
      |  ^~~~~
/home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/flash/soc_flash_mcux.c:47:10: fatal error: fsl_flash.h: No such file or directory
   47 | #include "fsl_flash.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[121/163] Building C object modules/zcbor/CMakeFiles/modules__zcbor.dir/home/shared/disk/zephyr_project/zephyr_test/modules/lib/zcbor/src/zcbor_decode.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/ubuntu/.local/bin/cmake --build /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/build
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-2514-g4a30aab2aefa

dleach02, 2024-04-23:
@hakehuang can you bisect this?

hakehuang, 2024-04-24:
> @hakehuang can you bisect this?

it fails since the HWMv2 added.

decsny, 2024-04-25:
seems related to https://github.com/zephyrproject-rtos/zephyr/issues/69944

hakehuang, 2024-05-20:
no such issue in test 34c84eccec0508b16f5001b20fa369f387d856df

closed 2024-05-20 by hakehuang