zephyrproject-rtos/zephyr #71479: Add return code 2 in documentation for k_work_reschedule_for_queue

marijnruyts, 2024-04-15:
**Describe the bug**

A return code of 2 will be returned for k_work_reschedule_for_queue (and k_work_schedule) whenever the work item is busy.
See static int submit_to_queue_locked(struct k_work *work, struct k_work_q **queuep) => K_WORK_RUNNING_BIT.
This should be documented

**Environment**

* Page: `kernel/services/threads/workqueue`
* Version: main
* SHA-1: 38635f889979030328ff3b35c8d25d2990e9819e

henrikbrixandersen, 2024-04-21:
@marijnruyts Could you submit a pull request adding this to the API documentation?

wkhadgar, 2024-04-23:
I've noticed the same issue, thought the missing `@retval` it's from `k_work_schedule_for_queue()`, also made a PR #71840 fixing that.

PR #71840 (2024-06-14): Update k_work_schedule_for_queue() docstring missing retval.
This commit updates the missing `@retval` from the underlying call of `submit_to_queue_locked()` when the work was running and has been queued to the queue that was running it.

Fixes #71479
closed 2024-06-14 by nashif