zephyrproject-rtos/zephyr #71522: doc: Documentation issue in 'samples/basic/blinky/README'

majidBahmanigbt, 2024-04-15:
**Blinky for IMX8MP DSP HiFi4: No SOURCES given to Zephyr library**

I want to compile blinky for IMX8MP DSP. I added led node in zephyr/zephyr/boards/nxp/imx8mp_evk/imx8mp_evk_mimx8ml8_adsp.dts :
```
	aliases {
		led0 = &myled0;
	};

	leds {
		compatible = "gpio-leds";
		myled0: led_0 {
			gpios = <&gpio3 16 GPIO_ACTIVE_LOW>;
				};
		status = "okay";
	};
```
Then I added gpio3 node in zephyr/zephyr/dts/xtensa/nxp/nxp_imx8m.dtsi:
```
gpio3: gpio@30220000 {
			compatible = "nxp,imx-gpio";
			reg = <0x30220000 DT_SIZE_K(64)>;
			rdc = <(RDC_DOMAIN_PERM(A7_DOMAIN_ID,\
					       RDC_DOMAIN_PERM_RW)|\
			       RDC_DOMAIN_PERM(M7_DOMAIN_ID,\
					       RDC_DOMAIN_PERM_RW))>;
			gpio-controller;
			#gpio-cells = <2>;
			status = "okay";
		};
```

However, I am getting this warning:
```
CMake Warning at /home/efacec/nxp/Repositories/DSP_Zephyr/zephyr/zephyr/CMakeLists.txt:865 (message):
  No SOURCES given to Zephyr library: drivers__gpio
  Excluding target from build.
```
And the compile failed with the following Error:

```
/home/efacec/zephyr-sdk-0.16.5-1/xtensa-nxp_imx8m_adsp_zephyr-elf/bin/../lib/gcc/xtensa-nxp_imx8m_adsp_zephyr-elf/12.2.0/../../../../xtensa-nxp_imx8m_adsp_zephyr-elf/bin/ld.bfd: app/libapp.a(main.c.obj):(.literal.main+0x8): undefined reference to `__device_dts_ord_37'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/efacec/nxp/Repositories/DSP_Zephyr/zephyr/build
```
According to <build>/zephyr/include/generated/devicetree_generated.h is related to gpio3

Could you please tell me how it is possible to solve this problem?

* Page: `samples/basic/blinky/README`
* Version: main
* SHA-1: b22bb172f28c7fb388d7703e43b5e24d2308a84f

kartben, 2024-04-15:
I think it is as simple as GPIO support not being there for the Xtensa ADSP core. 
@dbaluta and @iuliana-prodan can probably comment further.

dbaluta, 2024-04-15:
as @kartben said I think no one bothered with GPIO support for Xtensa  ADSP. 

So, first step would be to understand which driver needs to be pulled  in.

Maybe 

```
--- a/soc/nxp/imx/imx8m/Kconfig
+++ b/soc/nxp/imx/imx8m/Kconfig
@@ -51,6 +51,7 @@ config SOC_MIMX8ML8_ADSP
        select HAS_MCUX_CCM if CLOCK_CONTROL
        select HAS_MCUX_IOMUXC if PINCTRL
        select PINCTRL_IMX if HAS_MCUX_IOMUXC
+       select HAS_MCUX_IGPIO
        select CPU_HAS_DCACHE
```

Will have a look once we can get some free time. We are now travelling for EOSS.

majidBahmanigbt, 2024-04-16:
Hi,
I didn't know where I should have enabled the GPIO for DSP! It works now.
Thank you