zephyrproject-rtos/zephyr #71525: Accept() error in secure sockets.

Nulllix, 2024-04-15:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/66405

<div type='discussions-op-text'>

<sup>Originally posted by **Nulllix** December 11, 2023</sup>
## Introduction
Hey, everybody!
I have an nRF52840-DK debug board on which I am trying to run [BSD Secure Sockets](https://docs.zephyrproject.org/latest/connectivity/networking/api/sockets.html#secure-sockets) on up the [IPSP](https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html).
I used the [echo server (advanced) example](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_server/README.html#sockets-echo-server) as a basis.
My code on which the problem is stably reproduced [is here](https://github.com/Nulllix/test_project).

## Problem
I have written a [small python script](https://github.com/Nulllix/test_project/blob/main/utils/test_conn.py) that I'm trying to use to connect to my server. When I try to connect to nRF52840 with it, HardFault occurs in `accept(...)` function:
[image]
I have enabled additional logs inside the bluetooth stack and as you can see, at some point the pointer to `user_data` takes an invalid value.

Callstack which leads to HardFault:
[image]

Using a debugger I was able to get the full callstask which causes this error (inside `bt_conn_send_cb(...)` the functions you see in the screenshot above are called):
<details>
  <summary>BIG SCREENSHOTS WITH CALLSTACK</summary>
 [image] 
 [image] 
</details>

The full log with the error and callstack in text form, can be [found here](https://github.com/Nulllix/test_project/tree/main/logs).

## How to reproduce?
1. Clone [my repository](https://github.com/Nulllix/test_project/tree/main). 
2. Build and load firmware to nRF52840 from it.
3. Connect BLE adapter to PC (i use TL-LINK UB400)
4. Use script [board_connect.sh](https://github.com/Nulllix/test_project/blob/main/utils/board_connect.sh) to activate the 6lowpan and connect it to the board.
5. Use python script [test_conn.py](https://github.com/Nulllix/test_project/blob/main/utils/test_conn.py) to connect to the sockets.

Has anyone ever encountered anything like this?</div>

jukkar, 2024-04-15:
Note that IPSP support was just recently removed from upstream Zephyr. There were lot of issues with it, not sure if they were related to the issue you are seeing.
See for example #68629 and #68627 for more details.

Nulllix, 2024-04-16:
@jukkar Thanks for the info.

closed 2024-04-16 by Nulllix