zephyrproject-rtos/zephyr #71533: stm32wb55 cannot start coredump when the system HaltFault occurs.

lnhao, 2024-04-16:
**Describe the bug**
When enabling coredump feature with flash backend, on coredump, an error while erasing on stm32wb55 flash occurs.
I tracked the code and checked that the flash_area_erase() of the following code is returned -5.
```
static void coredump_flash_backend_start(void)
{
	const struct device *flash_dev;
	size_t offset, header_size;
	int ret;

	ret = partition_open();

	if (ret == 0) {
		/* Erase whole flash partition */
		ret = flash_area_erase(backend_ctx.flash_area, 0,
				       backend_ctx.flash_area->fa_size);
	}
```
output log:
```
err/[ms: 27050] (coredump_flash_backend_start:433)Cannot start coredump!
err: system halt, reason = 36, Unknown error.
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? STM32WB55

**To Reproduce**
Steps to reproduce the behavior:
1. Enable coredump config in prj.conf
```
CONFIG_DEBUG_COREDUMP=y
CONFIG_DEBUG_COREDUMP_BACKEND_FLASH_PARTITION=y
CONFIG_DEBUG_COREDUMP_BACKEND_LOGGING=n
CONFIG_DEBUG_COREDUMP_MEMORY_DUMP_MIN=y
```
2. In the board.dts file, add flash partition for coredump
```
coredump_partition: partition@f2000 {
	label = "coredump-partition";
	reg = <0x000f2000 DT_SIZE_K(4)>;
};
```
5. west build -b board -p
6. west flash / west debug
7. See error

**Expected behavior**
Coredump shall be written into flash.

**Impact**
Our product has malfunctioned at the customer's end, and I would like to obtain more fault information through coredump and save it to internal Flash.

**Environment (please complete the following information):**
 - OS:  Ubuntu
 - Toolchain: zephyr-sdk-0.16.0
 - Zephyr version: v3.3.0 / v3.6.0

Laczen, 2024-04-16:
@lnhao, it might be that you cannot use the flash interface during coredump for the stm32wb55. As a work around you could setup the flash simulator on a memory region, this would allow reading the coredump from this memory region.

erwango, 2024-04-17:
@lnhao Whatever you do, flash partition 0xDA000 is reserver for BLE stack (for 3.6.0), you cannot write there from whole zephyr side.
Please have a try using coredump partition within this boundary

lnhao, 2024-04-18:
> @lnhao, it might be that you cannot use the flash interface during coredump for the stm32wb55. As a work around you could setup the flash simulator on a memory region, this would allow reading the coredump from this memory region.

Hi @Laczen, If the memory region is used to store coredump information, it may be lost because entering the Haltfalt system is likely to cause a reboot. After the system starts, the memory will be initialized.

lnhao, 2024-04-18:
> @lnhao Whatever you do, flash partition 0xDA000 is reserver for BLE stack (for 3.6.0), you cannot write there from whole zephyr side. Please have a try using coredump partition within this boundary

Hi @erwango, we did not use the Bluetooth stack. And erase the Bluetooth stack. We also store other user data in the 0xDA000 address area of Flash, which can be stored normally. In Zephyr 3.6.0, we attempted to store coredump information in stm32wba52 and returned ERR -5 In Zephyr 3.3.0, we attempted to store coredump information on stm32wb55 and returned ERR -5.

Laczen, 2024-04-18:
> Hi @Laczen, If the memory region is used to store coredump information, it may be lost because entering the Haltfalt system is likely to cause a reboot. After the system starts, the memory will be [...]

@lnhao in my experience the memory is not always cleared after a reboot. Esspecially not if a memory region is reserved for it. A different use case can be found in https://github.com/zephyrproject-rtos/zephyr/discussions/54166.

lnhao, 2024-04-22:
> @lnhao in my experience the memory is not always cleared after a reboot. Esspecially not if a memory region is reserved for it. A different use case can be found in #54166.

 Thanks @Laczen , this is great, I will try this method.

closed 2024-09-07 by github-actions[bot]