zephyrproject-rtos/zephyr #71545: mec172xevb_assy6906: FATAL ERROR 4: Kernel panic

KamilxPaszkiet, 2024-04-16:
**Describe the bug**
List of failing tests:
```tests/benchmarks/sys_kernel/benchmark.kernel.core
tests/drivers/can/shell/drivers.can.shell
tests/drivers/eeprom/shell/drivers.eeprom.shell
tests/drivers/ipm/drivers.ipc.mailbox
tests/kernel/early_sleep/kernel.common.sleep
tests/kernel/early_sleep/kernel.common.sleep.minimallibc
tests/kernel/interrupt/arch.shared_interrupt
tests/kernel/interrupt/arch.shared_interrupt.lto
tests/kernel/workq/work_queue/kernel.workqueue
tests/subsys/logging/log_backend_init/logging.backend.initialization
tests/subsys/logging/log_cache/logging.cache
tests/subsys/logging/log_output/logging.output
tests/subsys/logging/log_output/logging.output.thread_id
tests/subsys/logging/log_output/logging.output.ts64
tests/subsys/logging/log_timestamp/logging.output.custom_timestamp
tests/subsys/logging/log_timestamp/logging.output.default_timestamp
```
I did a bisection, the problem started with the commit: [f11027d](https://github.com/zephyrproject-rtos/zephyr/commit/f11027df8061932ddf1209f985a67164d140f303)

**To Reproduce**
`twister --hardware-map <path_to>/mec172x.map --device-testing -T <test> -x=CONFIG_BOOT_DELAY=500 -v`

**Expected behavior**
Pass

**Logs and console output**
sample log:
```
*** Booting Zephyr OS build v3.6.0-2554-g2c8ea07b3498 (delayed boot 500ms) ***
Running TESTSUITE workqueue_delayed
===================================================================
ASSERTION FAIL [!z_is_thread_state_set(_kernel.cpus[0].current, ((1UL << (4))))] @ WEST_TOPDIR/os.rtos.zephyr.zephyr/kernel/sched.c:1170
E: ***** HARD FAULT *****
E: Fault escalation (see below)
E: ARCH_EXCEPT with reason 4
E: r0/a1: 0x00000004 r1/a2: 0x00000492 r2/a3: 0x00118848
E: r3/a4: 0x00000004 r12/ip: 0x00118044 r14/lr: 0x000c5345
E: xpsr: 0x21000000
E: Faulting instruction address (r15/pc): 0x000c6e2c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x118848 (main)
E: Halting system
```

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK v0.16.5-1
 - v3.6.0-2554-g2c8ea07b3498

ithinuel, 2024-04-26:
Are you happy closing this issue with the fix that was merged?

KamilxPaszkiet, 2024-04-29:
Problem fixed