zephyrproject-rtos/zephyr #71559: samples: drivers: mbox: sample.drivers.mbox.real_hw fails

katgiadla, 2024-04-16:
**Describe the bug**
The sample `sample.drivers.mbox.real_hw` fails.

Observed for `nrf5340dk/nrf5340/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk/nrf5340/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/drivers/mbox -p nrf5340dk/nrf5340/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash`
4. See console output with errorr

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (oPing (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Pon channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on channel 1)
Ping (on el 1)
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: `v3.6.0-2554-g2c8ea07b3498`

henrikbrixandersen, 2024-04-16:
@katgiadla When mass-reporting bugs like this, please add the relevant labels (`platform: nRF`). Thanks.

anangl, 2024-04-17:
Fixed by #71593.

closed 2024-04-17 by anangl