zephyrproject-rtos/zephyr #71565: board: particle argon - nRF52840 : Not booting

WilliamGFish, 2024-04-16:
**Describe the bug**
After downloading the latest changes in zephyr applications that previously worked on the Particle Argon now will not boot.

I have compiled other applications and they still work other devices

**Expected behavior**
device booting correctly 

**Impact**
Personally showstopper...

**Logs and console output**
No output to capture

**Environment (please complete the following information):**
 - OS: Windows 10
-- CMake version: 3.25.0-rc2
-- Zephyr version: 3.6.99
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: particle_argon, qualifiers: nrf52840
-- Found toolchain: gnuarmemb (C:\gnu_arm_embedded_10)

henrikbrixandersen, 2024-04-16:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

anangl, 2024-04-17:
@WilliamGFish Could you check if #71593 fixed this issue?

WilliamGFish, 2024-04-17:
All working as before

closed 2024-04-17 by WilliamGFish

oscar-lauth, 2024-09-29:
How were you able to flash ZephyrOS programs onto the particle argon, removing DeviceOS? Very curious as I am trying to find a way to flash my particle argon using ZephyrOS.

WilliamGFish, 2024-09-29:
Look up the device in documentation and follow the flashing instructions. There esp32 firmware will not work without being manually updated.