zephyrproject-rtos/zephyr #71582: doc: Documentation issue in 'samples/subsys/fs/fs_sample/README'

ConnorRigby, 2024-04-16:
**Describe the bug**

Running the commands exactly as they are described in the fs_sample guide, i get a compiler error:

```
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/connorrigby/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF52840_XXAA -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/include -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/build/zephyr/include/generated -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/soc/nordic -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/soc/nordic/nrf52/. -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/soc/nordic/common/. -I/home/connorrigby/workspace/connorrigby/my-workspace/example-application/include -I/home/connorrigby/workspace/connorrigby/my-workspace/modules/hal/cmsis/CMSIS/Core/Include -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/modules/cmsis/. -I/home/connorrigby/workspace/connorrigby/my-workspace/modules/hal/nordic/nrfx -I/home/connorrigby/workspace/connorrigby/my-workspace/modules/hal/nordic/nrfx/drivers/include -I/home/connorrigby/workspace/connorrigby/my-workspace/modules/hal/nordic/nrfx/mdk -I/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/modules/hal_nordic/nrfx/. -isystem /home/connorrigby/workspace/connorrigby/my-workspace/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /home/connorrigby/workspace/connorrigby/my-workspace/zephyr/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/connorrigby/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/connorrigby/workspace/connorrigby/my-workspace/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/samples/subsys/fs/fs_sample=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/connorrigby/workspace/connorrigby/my-workspace/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/connorrigby/workspace/connorrigby/my-workspace=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/connorrigby/workspace/connorrigby/my-workspace/zephyr/samples/subsys/fs/fs_sample/src/main.c
/home/connorrigby/workspace/connorrigby/my-workspace/zephyr/samples/subsys/fs/fs_sample/src/main.c:19:10: fatal error: ff.h: No such file or directory
   19 | #include <ff.h>
```

**Environment**

* Page: `samples/subsys/fs/fs_sample/README`
* Version: main
* SHA-1: 8dc3f856229ce083c956aa301c31a23e65bd8cd8

nordicjm, 2024-04-17:
You didn't though, because the documentation for zephyr clearly states to use `west init` on https://docs.zephyrproject.org/latest/develop/getting_started/index.html#get-zephyr-and-install-python-dependencies so if you don't have the fatfs module then you did not follow these instructions

nordicjm, 2024-04-17:
> ![image](https://private-user-images.githubusercontent.com/21150830/323255469-7b9e206a-1062-43b2-924f-ad7c81fdd2a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjQyMjAsIm5iZiI6MTcxMzM2MzkyMCwicGF0aCI6Ii8yMTE1MDgzMC8zMjMyNTU0NjktN2I5ZTIwNmEtMTA2Mi00M2IyLTkyNGYtYWQ3YzgxZmRkMmE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDE0MjUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhNzFhYjc2ZWJjYzg4YTgxMDUwZjI4MWM2MTU0YTQ3NTYzZmJkODllZWFlNDA3YjdjN2UyY2VhYWYxMDYzNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-ZxPFrVdPL_1eLLE_z_wj-RKwhkpFM_gqWf8r4f-pEI) ![image](https://private-user-images.githubusercontent.com/21150830/323255725-a62b089c-40a1-4cba-a80f-af0c3ee6593b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjQyMjAsIm5iZiI6MTcxMzM2MzkyMCwicGF0aCI6Ii8yMTE1MDgzMC8zMjMyNTU3MjUtYTYyYjA4OWMtNDBhMS00Y2JhLWE4MGYtYWYwYzNlZTY1OTNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDE0MjUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1MmZkMDFjMWYyZDljMGQ0NDkzZDYzMzc0ODlhOTUwOGI3NjlhMGM3NDViOWYxNjg5NjEyNjI5ODhhZWNjMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hQ-O2oshgHNqUrWenGu5UXcn-hfjZ1F6cuGxJpnCjXA)
> 
> the fatfs module is listed in both the root of `zephyr/modules` and in my workspace's `modules` folder

If you've added then it would be, it is not in the default allow list for example application https://github.com/zephyrproject-rtos/example-application/blob/main/west.yml example application is exactly as the name implies, it demonstrates how to make a workspace-based zephyr project with out of tree boards and drivers, it is not meant nor should it be used to run samples inside the zephyr directory, as all the modules and most of the hals are not in the allow list.

aescolar, 2024-04-23:
@ConnorRigby this seems like a duplicate of https://github.com/zephyrproject-rtos/zephyr/discussions/71826 , so I will close it as such. Please reopen if you disagree.

closed 2024-04-23 by aescolar

AndreyDodonov-EH, 2024-08-09:
I had the same issue, maybe somebody finds this helpful:
if you are working with your own app and just sometimes switching to samples, make sure that your  `west update` works with the right YAML where `- fatfs` is included.

E.g. I added it to my app explicitly:

```
  projects:
    - name: zephyr
      path: zephyr
      remote: zephyrproject-rtos
      revision: v3.7.0
      # explicitly add Zephyr-specific West extensions
      west-commands: scripts/west-commands.yml
      import:
        name-allowlist:
          - fatfs
          ```