zephyrproject-rtos/zephyr #71584: llext roadmap [WIP]

pillo79, 2024-04-16:
After a chat with @teburd @edersondisouza @selescop, wishlist in no particular order:

* unified architecture relocation 
* unified linking path
* kobject instantiation from the llext
* XIP
* security signing
* versioning / ABI for syscalls 
* versioning for exported symbols, maybe subsys level?
* module information / init / done function
* ask for capabilities -> get assigned permissions
* limit extension to userspace  
* filter syscalls

teburd, 2024-04-19:
After a discussion at EOSS, XIP could be implemented by providing a function that applies all relocations assuming given target addresses for storing the ELF sections. I thought this was an incredible idea and I’m hopeful someone will contribute it.

teburd, 2024-08-21:
I'd add supporting additional architectures to the list...

* x86
* riscv
* arc