zephyrproject-rtos/zephyr #71586: Clock output on Raspberry Pi Pico GPIOs

68ec020, 2024-04-16:
I use a ENC424J600 ethernet controller connected to a RPi Pico. The Pico also should generate the 25 MHz clock for the ENC424J600. In Zephyr v3.5.0 I used a PRE_KERNEL_1  SYS_INIT and the RPi Pico SDK to initialise the clock output:

`clock_gpio_init(21, CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_PLL_SYS, 5);`

In Zephyr v3.6.0 this is no longer working, because we have a real clock control now. This disables GPOUT0 during initialisation. And so far I didn't manage to enable the pin again in the device tree. I managed to set the pinctrl to the right function:

```
&pinctrl {
    clocks_default: clocks_default {
        group1 {
            pinmux = <GPOUT0_P21>;
        };
    };
};
```

This is working, but the ENABLE bit in CLK_GPOUT0_CTRL is still 0. I tried to "use" the clk_gpout0 in gpio0 and set the frequency, but no luck so far:

```
&clk_gpout0 {
    clock-frequency = < 25000000 >;
};

&gpio0 {
    clocks = <&clk_gpout0>;
    clock-names = "clk_gpout0";
};
```

Any help is appreciated.

Regards,
Jan

yonsch, 2024-04-24:
cc @soburi

soburi, 2024-04-25:
Hi, @68ec020 

Could you try `clock_control_on` the GPOUT0 before the ENC424J600 driver initializing?
such like...

```
const struct device *clk_dev = DEVICE_DT_GET(DT_NODELABEL(clocks));
clock_control_on(clk_dev, RPI_PICO_CLKID_CLK_GPOUT0);
```

68ec020, 2025-07-23:
This worked in v4.1, but stopped working in v4.2.

I configured the ENC424J600 as "deferred-init" in the device tree. Then I enable the clock as you described and then call device_init for the ENC424J600 device.

The problem with v4.2 is that parts of the initialization is already done before I can enable the clock. "enc424j600_iface_init‎" in the driver is called when the network is initialized, even if the device is configured as deferred-init. Because the device is not initialized, "enc424j600_init" hasn't run yet and so it didn't read the MAC address from the chip. So, "enc424j600_iface_init‎" calls "net_if_set_link_addr" with null bytes (00:00:00:00:00:00) and ethernet doesn't work.

At the moment I see no other way to "deferred-init" the device, if network initialization requires the device to be initialized at startup.