zephyrproject-rtos/zephyr #71589: zperf example does not work with CONFIG_USB_DEVICE_STACK_NEXT=y (nRF52840)

rgrr, 2024-04-17:
**Notes**
Trying to create an NCM driver for the new USB stack, using zperf with ECM as a starting point, see #71451 

**Describe the bug**
Using the zperf example and selecting the configuration with CONFIG_USB_DEVICE_STACK_NEXT=y.  west command line:

    west build --build-dir _build . --pristine --board nrf52840dk_nrf52840 --no-sysbuild -- -DNCS_TOOLCHAIN_VERSION=NONE -DBOARD_ROOT=. -DCACHED_CONF_FILE=prj.conf -DEXTRA_CONF_FILE=overlay-usbd_next_ecm.conf -DDTC_OVERLAY_FILE=usbd_next_ecm.overlay

After flashing the target gives the following log output:

    [07:31:33.906677 4.616036] [00:00:00.375,213] <inf> cdc_ecm: USB configuration is not enabled
    [07:31:33.908408 0.001732] *** Booting nRF Connect SDK v3.5.99-ncs1 ***
    [07:31:33.909375 0.000968] [00:00:00.375,762] <inf> net_config: Initializing network
    [07:31:33.910767 0.001392] [00:00:00.375,762] <inf> net_config: Waiting interface 1 (0x20000fc4) to be up...
    [07:32:04.217535 30.306766] [00:00:30.376,464] <inf> net_config: IPv4 address: 192.0.2.1
    [07:32:04.219149 0.001616] [00:00:30.376,617] <err> net_if: Cannot join all nodes address ff02::1 (-115)
    [07:32:04.220967 0.001819] [00:00:30.376,708] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-115)
    [07:32:04.222995 0.002028] [00:00:30.376,770] <err> net_config: Timeout while waiting network interface
    [07:32:04.224818 0.001821] [00:00:30.376,800] <err> net_config: Network initialization failed (-115)

It seems that the interface is not coming up automatically.  BTW: to me the start sequence is not clear at all ;-)

USB is not starting at all, meaning that the host computer does not see a (new) USB device.  (before you ask: with the old USB stack everything is working fine).

**To Reproduce**
see above

**Expected behavior**
I'ld expect that USB starts up, the interface is becoming available and the target device provides some IP functionality.

**Impact**
annoying...

**Logs and console output**
see above, log output with more debug looks like this:

    [07:58:39.485060 233.332755] [00:00:00.375,976] <inf> udc_nrf: Preinit
    [07:58:39.486638 0.001581] [00:00:00.376,220] <dbg> cdc_ecm: usbd_cdc_ecm_preinit: CDC ECM device initialized
    [07:58:39.489086 0.002447] [00:00:00.376,373] <dbg> usbd_core: usbd_pre_init: Available USB class nodes:
    [07:58:39.491341 0.002256] [00:00:00.376,403] <dbg> usbd_core: usbd_pre_init:   0x200010f4, name cdc_ecm_0
    [07:58:39.493618 0.002277] [00:00:00.376,892] <dbg> cdc_ecm: cdc_ecm_iface_init: CDC ECM interface initialized
    [07:58:39.496037 0.002418] [00:00:00.377,258] <dbg> cdc_ecm: cdc_ecm_iface_start: Start interface 0x20000fc4
    [07:58:39.498418 0.002382] [00:00:00.377,258] <inf> cdc_ecm: USB configuration is not enabled
    [07:58:39.500394 0.001975] *** Booting nRF Connect SDK v3.5.99-ncs1 ***
    [07:58:39.501460 0.001067] [00:00:00.377,746] <inf> net_config: Initializing network
    [07:58:39.503279 0.001818] [00:00:00.377,746] <inf> net_config: Waiting interface 1 (0x20000fc4) to be up...
    [07:59:10.477751 30.974470] [00:00:30.378,448] <inf> net_config: IPv4 address: 192.0.2.1
    [07:59:10.479521 0.001774] [00:00:30.378,601] <err> net_if: Cannot join all nodes address ff02::1 (-115)
    [07:59:10.481432 0.001910] [00:00:30.378,692] <err> net_if: Cannot join solicit node address ff02::1:ff00:1 (-115)
    [07:59:10.483506 0.002074] [00:00:30.378,753] <err> net_config: Timeout while waiting network interface
    [07:59:10.485389 0.001884] [00:00:30.378,784] <err> net_config: Network initialization failed (-115)

**Environment (please complete the following information):**
 - Linux, Debian testing
 -- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /mnt/d/u/src/ncs/zperf_checko
-- CMake version: 3.21.0
-- Found Python3: /home/hardy/ncs/toolchains/2be090971e/usr/local/bin/python3.9 (found suitable version "3.9.18", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/hardy/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/hardy/ncs/v2.6.0/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
-- Found host-tools: zephyr 0.16.5 (/home/hardy/ncs/toolchains/2be090971e/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.5 (/home/hardy/ncs/toolchains/2be090971e/opt/zephyr-sdk)

**Additional context**
set `CONFIG_NET_SOCKETS_POLL_MAX=10` to make network startup.  Otherwise log shows

    [07:49:46.671932 0.001648] *** Booting nRF Connect SDK v3.5.99-ncs1 ***
    [07:49:46.672882 0.000950] [00:00:00.374,908] <inf> net_config: Initializing network
    [07:49:46.674242 0.001360] [00:00:00.374,908] <inf> net_config: Waiting interface 1 (0x20000fc4) to be up...
    [07:49:46.921153 0.246910] [00:00:00.376,190] <err> net_sock_svc: You have 9 services to monitor but 5 poll entries configured.
    [07:49:46.923561 0.002410] [00:00:00.376,251] <err> net_sock_svc: Please increase value of CONFIG_NET_SOCKETS_POLL_MAX to at least 9

jfischer-no, 2024-04-18:
It is not broken, I use it regularly. Not initializing it in the main sample was intentional at the time. The shell is enabled in the zperf sample, what is missing in your build command is -DCONFIG_USBD_SHELL=y, see `git log samples/net/zperf/usbd_next_ecm.overlay`

rgrr, 2024-04-18:
Thanks for your help!

Added some code to main.c:

```C
#if defined(CONFIG_USB_DEVICE_STACK_NEXT)
    #include <zephyr/usb/usbd.h>

    USBD_CONFIGURATION_DEFINE(config_foo, USB_SCD_SELF_POWERED, 200);
	USBD_DEVICE_DEFINE(sh_uds_ctx, DEVICE_DT_GET(DT_NODELABEL(zephyr_udc0)),
		  			   0x2fe3, 0xffff);

	USBD_DESC_LANG_DEFINE(lang);
	USBD_DESC_MANUFACTURER_DEFINE(mfr, "ZEPHYR");
	USBD_DESC_PRODUCT_DEFINE(product, "Zephyr USBD foobaz");
	USBD_DESC_SERIAL_NUMBER_DEFINE(sn, "0123456789ABCDEF");

	static struct usbd_contex *my_uds_ctx = &sh_uds_ctx;
#endif

...

int main(void)
{
...
#if defined(CONFIG_USB_DEVICE_STACK_NEXT)
	{
		int err;

		err = usbd_add_configuration(my_uds_ctx, &config_foo);
		printk("usbd_add_configuration: %d\n", err);

		err = usbd_register_class(my_uds_ctx, "cdc_ecm_0", 1);
		printk("usbd_register_class: %d\n", err);

		err = usbd_add_descriptor(my_uds_ctx, &lang);
		err |= usbd_add_descriptor(my_uds_ctx, &mfr);
		err |= usbd_add_descriptor(my_uds_ctx, &product);
		err |= usbd_add_descriptor(my_uds_ctx, &sn);
		printk("usbd_add_descriptor: %d\n", err);

		err = usbd_init(my_uds_ctx);
		printk("usbd_init: %d\n", err);
		err = usbd_enable(my_uds_ctx);
		printk("usbd_enable: %d\n", err);
	}
#endif
```

USB is now starting up (with a delay of 30s because it is still waiting for the IF to come up), the network interface also shows up after that and pinging works.
iperf unfortunately does not.

Have to look tomorrow deeper into this.

rgrr, 2024-04-18:
gosh... I think I'm too blonde today...

Forgot to insert my other code to start the server automatically.  Looks like:

```C
    {
        static const struct zperf_download_params param = { .port = 5001 };

        zperf_tcp_download( &param, my_zperf_callback, NULL);
        zperf_udp_download( &param, my_zperf_callback, NULL);   // just with IPv4?  And no callback?
    }
```

So it seems to work as expected.  Thanks for support and if you want to, I can create a PR.

jfischer-no, 2024-04-18:
That is a bit to much https://github.com/zephyrproject-rtos/zephyr/issues/71589#issuecomment-2064095539. We have common initialization code for the new USB samples, take a look at the samples/bluetooth/hci_usb. It should look something like this:

```diff
diff --git a/samples/net/zperf/CMakeLists.txt b/samples/net/zperf/CMakeLists.txt
index c4552f97141..dd1a09bbb22 100644
--- a/samples/net/zperf/CMakeLists.txt
+++ b/samples/net/zperf/CMakeLists.txt
@@ -22,3 +22,7 @@ if (CONFIG_NET_L2_ETHERNET)
     LOCATION "${CONFIG_NET_SAMPLE_CODE_RAM_NAME}_TEXT" NOKEEP)
 endif()
 endif()
+
+if (CONFIG_USB_DEVICE_STACK_NEXT)
+  include(${ZEPHYR_BASE}/samples/subsys/usb/common/common.cmake)
+endif()
diff --git a/samples/net/zperf/Kconfig b/samples/net/zperf/Kconfig
index a56cd6b9416..54ab59ca312 100644
--- a/samples/net/zperf/Kconfig
+++ b/samples/net/zperf/Kconfig
@@ -20,3 +20,5 @@ config NET_SAMPLE_CODE_RAM_NAME
          Region to relocate networking code to
 
 endif # NET_SAMPLE_CODE_RELOCATE
+
+source "samples/subsys/usb/common/Kconfig.sample_usbd"
diff --git a/samples/net/zperf/src/main.c b/samples/net/zperf/src/main.c
index 8c9d6a5c64c..e81a490ae66 100644
--- a/samples/net/zperf/src/main.c
+++ b/samples/net/zperf/src/main.c
@@ -9,17 +9,40 @@
  * @brief Zperf sample.
  */
 #include <zephyr/usb/usb_device.h>
+#include <zephyr/usb/usbd.h>
 #include <zephyr/net/net_config.h>
 
 #ifdef CONFIG_NET_LOOPBACK_SIMULATE_PACKET_DROP
 #include <zephyr/net/loopback.h>
 #endif
+
+#if defined(CONFIG_USB_DEVICE_STACK_NEXT)
+#include <sample_usbd.h>
+
+static int enable_usb_device_next(void)
+{
+       struct usbd_contex *sample_usbd = sample_usbd_init_device(NULL);
+
+       if (sample_usbd == NULL) {
+               printk("Failed to initialize USB device");
+               return -ENODEV;
+       }
+
+       return usbd_enable(sample_usbd);
+}
+#endif /* CONFIG_USB_DEVICE_STACK_NEXT */
+
 int main(void)
 {
-#if defined(CONFIG_USB_DEVICE_STACK)
+#if defined(CONFIG_USB_DEVICE_STACK) || defined(CONFIG_USB_DEVICE_STACK_NEXT)
        int ret;
 
+#if defined(CONFIG_USB_DEVICE_STACK)
        ret = usb_enable(NULL);
+#endif
+#if defined(CONFIG_USB_DEVICE_STACK_NEXT)
+       ret = enable_usb_device_next();
+#endif
        if (ret != 0) {
                printk("usb enable error %d\n", ret);
        }
```

Whereby it is too much ifdefery, IS_ENABLED() could also work.

rgrr, 2024-04-19:
Seems that I have to integrate the current development branch.  Is there any doc about this?  (I'm actually using nRF Connect SDK)

tmon-nordic, 2024-04-23:
> Seems that I have to integrate the current development branch. Is there any doc about this? (I'm actually using nRF Connect SDK)

NRF Connect SDK development model has the concept of upmerge. Currently the HWMv2 is being upmerged to NCS (See https://github.com/nrfconnect/sdk-nrf/pull/14626 for more information). Considering how complex upmerge is, I believe it is better for you to simply wait until NCS has all the required Zephyr changes integrated. In the meantime, it is probably best to use upstream Zephyr directly (if you can get away without NCS specific features).

rgrr, 2024-04-23:
> > Seems that I have to integrate the current development branch. Is there any doc about this? (I'm actually using nRF Connect SDK)
> 
> ...it is probably best to use upstream Zephyr directly (if you can get away without NCS specific features).

It is much simpler: installed zephyr to ~/zephyrproject according to installation doc & the VSCode plugin automatically detects it and one can select "Zephyr v3.6.99".

Now your suggestion from above is compiling & working.

Question 1: ok if I extend the example so that a zperf_tcp_download() server is auto started?  This makes testing a lot easier.

Question 2: how stable is the current ECM driver?  Is it work in progress?

jfischer-no, 2024-05-21:
> Question 2: how stable is the current ECM driver? Is it work in progress?

Everything in subsys/usb/device_next is experimental. CDC ECM is not a work in progress.