zephyrproject-rtos/zephyr #71596: Arm: Hard fault on nRF5340 after #65071 for Bluetooth

Thalley, 2024-04-17:
**Describe the bug**
After https://github.com/zephyrproject-rtos/zephyr/pull/65071 got merged, running any Bluetooth application results in a hardfault on the nRF5340's application core. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash the [hci_ipc](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/hci_ipc) sample to the network core 
2. Build and flash the [peripheral](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/peripheral) sample to the application core
3. See error

**Expected behavior**
Expect samples to work

**Impact**
Blocks nearly all usage of the nRF5340 boards

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-2565-gb4eac67b8b2e ***
[00:00:00.423,309] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.423,309] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.423,309] <inf> fs_nvs: data wra: 0, 0
[00:00:00.424,194] <err> bt_hci_driver: Endpoint binding failed with -11
[00:00:00.424,224] <err> bt_hci_core: HCI driver open failed (-11)
Bluetooth init failed (err -11)
[00:00:00.424,255] <err> os: ***** HARD FAULT *****
[00:00:00.424,255] <err> os:   Fault escalation (see below)
[00:00:00.424,285] <err> os: ***** BUS FAULT *****
[00:00:00.424,285] <err> os:   Precise data bus error
[00:00:00.424,285] <err> os:   BFAR Address: 0x0
[00:00:00.424,316] <err> os: r0/a1:  0x50000000  r1/a2:  0x00003ff0  r2/a3:  0x00000000
[00:00:00.424,316] <err> os: r3/a4:  0x00000005 r12/ip:  0x0000000c r14/lr:  0x0000152f
[00:00:00.424,316] <err> os:  xpsr:  0x89000000
[00:00:00.424,316] <err> os: Faulting instruction address (r15/pc): 0x000015e4
[00:00:00.424,346] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:00.424,377] <err> os: Current thread: 0x200018b0 (main)
[00:00:00.535,034] <err> os: Halting system
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: b4eac67b8b2eee5381d58671fddb8ae9bf2a3989 but issue also appears with https://github.com/zephyrproject-rtos/zephyr/pull/71593

**Additional context**
Non Bluetooth samples are still working, such as the [hello_world](https://github.com/zephyrproject-rtos/zephyr) sample

Thalley, 2024-04-17:
From https://github.com/zephyrproject-rtos/zephyr/pull/71595 it appears that this also applies the nRF52840 board, which uses a different driver, so it's still related to Bluetooth, but not the driver (and affects multiple boards)

nordic-piks, 2024-04-17:
Non bluetooth are affected too e.g
https://github.com/zephyrproject-rtos/zephyr/pull/65071#issuecomment-2060727351

closed 2024-04-17 by aescolar

hassan20akram, 2025-10-28:
can anyone explain what happened here because im still having this error.