zephyrproject-rtos/zephyr #71616: Contributor Nomination - Łukasz Majewski (OA TC6 driver)

lmajewski, 2024-04-17:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Łukasz Majewski
1. lmajewski
1. Denx Software Engineering

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/65507
https://github.com/zephyrproject-rtos/zephyr/pull/63614
https://github.com/zephyrproject-rtos/zephyr/pull/40704
https://github.com/zephyrproject-rtos/zephyr/pull/33772
https://github.com/zephyrproject-rtos/zephyr/pull/33914
https://github.com/zephyrproject-rtos/zephyr/pull/27620

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

lmajewski, 2024-04-23:
@aescolar @MaureenHelm @carlescufi  Gentle ping on this issue.

nashif, 2024-04-24:
https://github.com/zephyrproject-rtos/zephyr/invitations