zephyrproject-rtos/zephyr #71631: Exporting SBOM fails

traluke, 2024-04-17:
I tried to export the SBOM according to the [documentation](https://docs.zephyrproject.org/latest/develop/west/zephyr-cmds.html#software-bill-of-materials-west-spdx), however, the last command fails with below error:

```
vscode:~$ west spdx -d FOO
parsing CMake Cache file
parsing CMake Codemodel files
setting up SPDX documents
walking through targets
Traceback (most recent call last):
  File "/home/vscode/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/vscode/.local/lib/python3.10/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 65, in do_run
    do_run_spdx(args)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 114, in do_run_spdx
    makeSPDX(cfg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/sbom.py", line 81, in makeSPDX
    retval = w.makeDocuments()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 92, in makeDocuments
    self.walkTargets()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 276, in walkTargets
    self.collectTargetDependencies(cfgTargets, cfgTarget, pkg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 471, in collectTargetDependencies
    rd.ownerFileAbspath = pkg.targetBuildFile.abspath
AttributeError: 'NoneType' object has no attribute 'abspath'
```

Found also that somebody else was struggling with it online: https://devzone.nordicsemi.com/f/nordic-q-a/91659/how-to-generate-sbom

Any suggestions?

traluke, 2024-04-18:
Hi @marc-hb,
Thank you very much for the fast response, highly appreciated.
Please find below the requested section from the verbose command. Note that I modified the path in order not to leak any information. 

```
  - initializing Package for target: drivers__i2c
  - adding File /home/foo/zephyr/drivers/i2c/libdrivers__i2c.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/i2c/libdrivers__i2c.a
  - collecting source files and adding to pending queue
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_common.c
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_nrfx_twim.c
  - collecting target dependencies for drivers__i2c
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for zephyr_generated_headers
    - adding pending relationship for mbedcrypto
    - adding pending relationship for mbedcrypto_base
    - adding pending relationship for pw_build.empty
    - adding pending relationship for pw_log.config
    - adding pending relationship for pw_log.facade
    - adding pending relationship for pw_log_zephyr
    - adding pending relationship for pw_polyfill
    - adding pending relationship for pw_preprocessor
    - adding pending relationship for tfm
  - initializing Package for target: drivers__interrupt_controller
  - adding File /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
  - target drivers__interrupt_controller lists build artifact /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a but file not found after build; skipping
  - collecting target dependencies for drivers__interrupt_controller
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for mbedcrypto
Traceback (most recent call last):
  File "/home/vscode/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/vscode/.local/lib/python3.10/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 65, in do_run
    do_run_spdx(args)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 114, in do_run_spdx
    makeSPDX(cfg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/sbom.py", line 81, in makeSPDX
    retval = w.makeDocuments()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 92, in makeDocuments
    self.walkTargets()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 276, in walkTargets
    self.collectTargetDependencies(cfgTargets, cfgTarget, pkg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 471, in collectTargetDependencies
    rd.ownerFileAbspath = pkg.targetBuildFile.abspath
AttributeError: 'NoneType' object has no attribute 'abspath'
```

kartben, 2024-04-18:
> Can you please run `west -v -v -v -v spdx -d FOO`, redirect the logs and search for any such message in them: `but file not found after build`
> 
> See commit [c2ae5b3](https://github.com/zephyrproject-rtos/zephyr/commit/c2ae5b3bbf1af57303e8fef4ae1cef57237a9ff1) why.

@traluke it looks like you're running a version of Zephyr from *before* this fix mentioned by @marc-hb was introduced, based on the line numbers in your error log.

marc-hb, 2024-04-18:
Thanks! Can you please:

- Make sure the current Zephyr main branch still reproduces. If yes, share the recent Zephyr commit that reproduces... ok now I see that @kartben just beat me to it!

By transfering this bug from the "west" repo I made you skip all these "mandatory" questions but please take a look at them now:
https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

traluke, 2024-04-19:
@kartben, @marc-hb
Thank you for your support. We are using `sdk-nrf v2.5.2` which is the second latest release. It pulls in `sdk-zephyr v3.4.99-ncs1-2` if I am not mistaken. The fix that you mentioned is included already.

Please find the requested template filled below:

---
name: SPDX Command Fails
about: Software bill of materials generation fails on 
title: 'Software bill of materials generation fails'
labels: bug
assignees: ''

---

**If the bug is for a project fork (such as NCS) specific feature, please open an issue in the fork project
instead.**

Sorry, to late for that. Should I create a new issue in the NCS project, 
or would you like to follow up with the issue here?

**Describe the bug**
SPDX Document creation fails.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? -> nrf7002dk_nrf5340_cpuapp_ns
 - What have you tried to diagnose or workaround this issue? -> no

**To Reproduce**
Steps to reproduce the behavior:
1. west build -d foo
2. west build application/ -b nrf7002dk_nrf5340_cpuapp_ns -d foo
3. west spdx -d foo
See Error

**Expected behavior**
SBOM documents in SPDX format (app.spdx, zephyr.spdx, build.spdx) to be created.

**Impact**
Not able in integrate SBOM into our dependency tracking system.

**Logs and console output**
Output of `west -v -v -v -v spdx -d foo`
```
 - initializing Package for target: drivers__i2c
  - adding File /home/foo/zephyr/drivers/i2c/libdrivers__i2c.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/i2c/libdrivers__i2c.a
  - collecting source files and adding to pending queue
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_common.c
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_nrfx_twim.c
  - collecting target dependencies for drivers__i2c
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for zephyr_generated_headers
    - adding pending relationship for mbedcrypto
    - adding pending relationship for mbedcrypto_base
    - adding pending relationship for pw_build.empty
    - adding pending relationship for pw_log.config
    - adding pending relationship for pw_log.facade
    - adding pending relationship for pw_log_zephyr
    - adding pending relationship for pw_polyfill
    - adding pending relationship for pw_preprocessor
    - adding pending relationship for tfm
  - initializing Package for target: drivers__interrupt_controller
  - adding File /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
  - target drivers__interrupt_controller lists build artifact /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a but file not found after build; skipping
  - collecting target dependencies for drivers__interrupt_controller
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for mbedcrypto
Traceback (most recent call last):
  File "/home/vscode/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/vscode/.local/lib/python3.10/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 65, in do_run
    do_run_spdx(args)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 114, in do_run_spdx
    makeSPDX(cfg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/sbom.py", line 81, in makeSPDX
    retval = w.makeDocuments()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 92, in makeDocuments
    self.walkTargets()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 276, in walkTargets
    self.collectTargetDependencies(cfgTargets, cfgTarget, pkg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 471, in collectTargetDependencies
    rd.ownerFileAbspath = pkg.targetBuildFile.abspath
AttributeError: 'NoneType' object has no attribute 'abspath'
```

**Environment (please complete the following information):**
 - OS: Linux 6a95583e2309 5.15.133.1-microsoft-standard-WSL2
 - Toolchain: sdk-nrf
 - Version: v2.5.2

marc-hb, 2024-04-19:
> Sorry, to late for that. Should I create a new issue in the NCS project, or would you like to follow up with the issue here?

Can you try to reproduce "upstream" = without NCS/NRF?

traluke, 2024-04-23:
> Can you try to reproduce "upstream" = without NCS/NRF?

Unfortunately we have our dev environment tightly integrated with NCF/NRF. However, I was quickly able to test it with `nrf connect 2.6.0` which uses `zephyr v3.5.99-ncs1`. I can confirm that the issue is still present there. Is this information helpful as well?

kartben, 2024-04-23:
> Unfortunately we have our dev environment tightly integrated with NCF/NRF. However, I was quickly able to test it with `nrf connect 2.6.0` which uses `zephyr v3.5.99-ncs1`. I can confirm that the [...]

That's weird. Could you post the error log for that one too?

traluke, 2024-04-23:
Thank you for your help!

I copied the same snippet as before for the `west -v -v -v -v spdx -d foo` command.

```
- initializing Package for target: drivers__i2c
  - adding File /home/foo/zephyr/drivers/i2c/libdrivers__i2c.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/i2c/libdrivers__i2c.a
  - collecting source files and adding to pending queue
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_common.c
    - add pending source file and relationship for /home/vscode/workspaces/external_dependencies/zephyr/drivers/i2c/i2c_nrfx_twim.c
  - collecting target dependencies for drivers__i2c
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for zephyr_generated_headers
    - adding pending relationship for mbedcrypto
    - adding pending relationship for mbedcrypto_base
    - adding pending relationship for pw_build.empty
    - adding pending relationship for pw_log.config
    - adding pending relationship for pw_log.facade
    - adding pending relationship for pw_log_zephyr
    - adding pending relationship for pw_polyfill
    - adding pending relationship for pw_preprocessor
    - adding pending relationship for tfm
  - initializing Package for target: drivers__interrupt_controller
  - adding File /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
    - relativeBaseDir: /home/foo
    - artifacts[0]: zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a
  - target drivers__interrupt_controller lists build artifact /home/foo/zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a but file not found after build; skipping
  - collecting target dependencies for drivers__interrupt_controller
    - adding pending relationship for app_version_h
    - adding pending relationship for syscall_list_h_target
    - adding pending relationship for driver_validation_h_target
    - adding pending relationship for kobj_types_h_target
    - adding pending relationship for mbedcrypto
Traceback (most recent call last):
  File "/home/vscode/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/home/vscode/.local/lib/python3.10/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/vscode/.local/lib/python3.10/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 65, in do_run
    do_run_spdx(args)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/spdx.py", line 114, in do_run_spdx
    makeSPDX(cfg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/sbom.py", line 81, in makeSPDX
    retval = w.makeDocuments()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 92, in makeDocuments
    self.walkTargets()
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 276, in walkTargets
    self.collectTargetDependencies(cfgTargets, cfgTarget, pkg)
  File "/home/vscode/workspaces/external_dependencies/zephyr/scripts/west_commands/zspdx/walker.py", line 471, in collectTargetDependencies
    rd.ownerFileAbspath = pkg.targetBuildFile.abspath
AttributeError: 'NoneType' object has no attribute 'abspath'
```

kartben, 2024-04-23:
OK so it looks like this was fixed in #66182 (and not the commit mentioned earlier), which unfortunately is part of the 3.6.0 release, and introduced after NCS' v.3.5.9-ncs1. Not sure there is a good solution for you until NCS updates their Zephyr version, which it looks like they might do in a future 2.6.x release looking at their current repo.

nashif, 2024-04-23:
this is not an issue in upstream zephyr, closing.

closed 2024-04-23 by nashif

sweng5080, 2025-03-25:
The problem still exists in the upstream zephyr code. Here is the required sequence of events:

1. The build configuration is such that an empty zephyr module library is included that has no contents (i.e., built with empty_file.c)
2. addBuildFile() is then called with an artifactPath that does not exists and takes the early exist returning None on line 450 without ever assigning the pkg.targetBuildFile = bf on line 468
3. collectTargetDependencies() is called on line 416 with the pkg as the 3rd argument and pkg.targetBuildFile is None still
4. collectTargetDependencies() loops through the target dependencies and eventually gets to assigning the relationship data between the build files on lines 608-617
5. On line 611 inside of collectTargetDependencies() is executed and pkg.targetBuildFile is None so there is no abspath member
6. The exception with the NoneType object is thrown by the python interpreter

The simple solution is to avoid adding the relationship data between build files here, something like this:

`diff --git a/scripts/west_commands/zspdx/walker.py b/scripts/west_commands/zspdx/walker.py`
`index 196c68228fe..7752074c782 100644`
`--- a/scripts/west_commands/zspdx/walker.py`
`+++ b/scripts/west_commands/zspdx/walker.py`
`@@ -605,6 +605,9 @@ class Walker:`
`             if depAbspath == "":`
`                 continue`
`            ` 
`+            if pkg.targetBuildFile is None:`
`+                log.wrn(f'skipping STATIC_LINK for empty {pkg.cfg.name} target build file')`
`+                continue`
`             # create relationship data between build files`
`             rd = RelationshipData()`
`             rd.ownerType = RelationshipDataElementType.FILENAME`