zephyrproject-rtos/zephyr #71633: Xtensa: dup_table duplicates the pinned PTEVADDR entry

ldmstm, 2024-04-18:
The page table duplication routine in the current Xtensa implementation [dup_table](https://github.com/zephyrproject-rtos/zephyr/blob/d0be2010c31fed8ade258dae0417efdc02eca235/arch/xtensa/core/ptables.c#L728) only skips copying and allocating L2 page tables when it detects a PTE as "illegal" (attributes 12 or 14), however it should also skip this when the index is the same as the one for the kernel ptables pinned virtual address, ie the entry that maps `CONFIG_XTENSA_MMU_PTEVADDR`.

Doing this copy wastes an entire 4k of memory and could potentially be a security issue, as it's copying the L1 table as an L2 page table.

**Expected behavior**
The code should entirely skip the entry that matches `CONFIG_XTENSA_MMU_PTEVADDR` and mark it as unmapped when duplicating the table.

**Impact**
Wasted memory, potential security issue.

**Additional context**
Xtensa platform with MMU and userspace enabled

ceolin, 2024-04-23:
Not clear what is the problem here, we indeed map the physical memory for all page tables, but we don't map the virtual address for the page table [CONFIG_XTENSA_MMU_PTEVADDR + 4MB], we just pin the l1 page for this area and the auto refill does the rest.

ldmstm, 2024-04-24:
In the dup_tables routine it believes this entry you place at XTENSA_MMU_L1_POS(CONFIG_XTENSA_MMU_PTEVADDR) during kernel startup _is_ a valid L1 page table entry to a different L2 page table, when in reality it's just itself.

When duplicating it, this causes memory to be mapped at CONFIG_XTENSA_MMU_PTEVADDR for other memory domains, when in reality we only want the mapping at the 4MB * user_asid offset

ceolin, 2024-04-24:
> XTENSA_MMU_L1_POS

I think I got what you mean, it is confuse because we don't map the page itself until it is used, so if I understood correctly, we can do:
```
		if (is_pte_illegal(source_table[i]) || (i == XTENSA_MMU_L1_POS(XTENSA_MMU_PTEVADDR))) {
			dst_table[i] = XTENSA_MMU_PTE_ILLEGAL;
			continue;
		}
```
I can't see a security issue because it is mapped as using kernel ring and the whole page table (physical memory) is already mapped, but indeed we can save memory doing it :)

ldmstm, 2024-04-24:
Looks perfect to me, a whole 4k saved per memory domain.

Thanks!

closed 2024-04-27 by jhedberg