zephyrproject-rtos/zephyr #71652: west boards can not find custom SoC

foxriver1025, 2024-04-18:
https://github.com/zephyrproject-rtos/zephyr/blob/1c3edca8d89c7f66d2c0c0f5bec631517a6fe468/scripts/west_commands/boards.py#L77C8-L77C39

west boards
ERROR: SoC 'm88star5' is not found, please ensure that the SoC exists and that soc-root containing 'm88star5' has been correctly defined.
Out of tree boards and SoC, The boards command was not properly using the zephyr_module functionality to obtain the soc roots of all modules.

nordicjm, 2024-04-18:
There is no information here to go on, please provide **full** reproduction steps, ideally a full repo with your setup, or the issue cannot be investigated, this also includes the **full mandatory** information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

henrikbrixandersen, 2024-04-21:
Please provide missing information for reproducing/analyzing this issue and reopen.

closed 2024-04-21 by henrikbrixandersen