zephyrproject-rtos/zephyr #71698: Zephyr doesn't boot on STM32F4 (and probably other ARMv7-M cores)

duda-patryk, 2024-04-19:
**Describe the bug**
Zephyr can't boot on STM32F4 after #65071.

**To Reproduce**
Build Zephyr and run on STM32F4

**Expected behavior**
It boots

**Impact**
Very high. Zephyr doesn't work.

**Logs and console output**
No logs are printed.

**Environment (please complete the following information):**
N/A

**Additional context**
Reverting https://github.com/zephyrproject-rtos/zephyr/commit/f11027df8061932ddf1209f985a67164d140f303 seems to fix the problem.

I compared assembly code and looks like commit  above removed `cpsie if` instruction from arch_switch_to_main_thread().
For me, the following change fixes problem (ARMv7-M only!):
```
diff --git a/arch/arm/core/cortex_m/thread.c b/arch/arm/core/cortex_m/thread.c
index f9a030675b5..87900df9091 100644
--- a/arch/arm/core/cortex_m/thread.c
+++ b/arch/arm/core/cortex_m/thread.c
@@ -576,6 +576,7 @@ void arch_switch_to_main_thread(struct k_thread *main_thread, char *stack_ptr,
        "mov   r4,  %0\n"       /* force _main to be stored in a register */
        "msr   PSP, %1\n"       /* __set_PSP(stack_ptr) */
 
+       "cpsie if\n"            /* __enable_irq(); __enable_fault_irq() */
        "mov   r0,  #0\n"       /* arch_irq_unlock(0) */
        "ldr   r3, =arch_irq_unlock_outlined\n"
        "blx   r3\n
```

nordicjm, 2024-04-19:
Did you try with https://github.com/zephyrproject-rtos/zephyr/pull/71593 ?

duda-patryk, 2024-04-19:
> Did you try with #71593 ?

Yes

I realized that it fails to boot on warm reboots (cold reboots are fine).

RichardSWheatley, 2024-04-19:
I am having a similar Issue and it is due to this commit:

> Commit: f11027df8061932ddf1209f985a67164d140f303 [f11027d]
> Parents: 72312feead
> Author: Wilfried Chauveau <wilfried.chauveau@arm.com>
> Date: Friday, October 6, 2023 10:45:48 PM
> Committer: Carles Cufí
> Commit Date: Monday, April 15, 2024 9:09:28 AM
> arch: arm: cortex_m: Use outlined arch_irq_(un)lock in assembly
> 
> Asm is notoriously harder to maintain than C and requires core specific
> adaptation which impairs even more the readability of the code.
> 
> This change reduces the need for core specific conditional compilation.
> 
> Signed-off-by: Wilfried Chauveau <wilfried.chauveau@arm.com>

ithinuel, 2024-04-19:
It appears that I have missed something important around the (seemingly inconsistent) use of BASEPRI & cpsie/d… 🤔

Interrestingly, this isn’t caught by any of the tests I ran. I would like to fix the issue but also eventually add some tests to capture future regressions.
Could you please detail the process you’re following to reproduce the issue?
Do you use an in-tree example/test? How do you load/reload/reset?

soburi, 2024-04-21:
I also encountered a similar phenomenon with arduino_uno_r4_minima (Renesas RA4M1, Cortex-M4/ARMv7-M).
Similar to this issue, I confirmed that reverting https://github.com/zephyrproject-rtos/zephyr/commit/f11027df8061932ddf1209f985a67164d140f303 works correctly.
It is also working fine with @duda-patryk patch.

ithinuel, 2024-04-22:
Thank you for letting me know. I am currently running tests on a nucleo_f401re but can’t seem to be able to reproduce the issue.

Along with the fix, I’d like to understand & document the why is this extra cpsie required.

ithinuel, 2024-04-22:
I am running Zephyr’s tests & samples.
I get some false positives (ie failing test because of twister missing the start of the log) but they match before & after the changes to `arch/arm`.

Would you be able to provide a sample project that exhibits this issue?

duda-patryk, 2024-04-22:
In my case, I'm jumping to Zephyr reset vector from another code (chain-loading) with interrupts disabled (`cpsid i`).

To reproduce issue without chain-loading you can use following code:
```
void z_arm_reset(void);

void test(void)
{
        __disable_irq();
        z_arm_reset();
}
```

ithinuel, 2024-04-22:
So I edited `samples/hello_world/src/main.c` such that it becomes:
```c
#include "zephyr/kernel.h"
#include <stdio.h>

extern void z_arm_reset(void);

int main(void) {
    printf("Hello World! %s\n", CONFIG_BOARD_TARGET);

    k_sleep(Z_TIMEOUT_MS(2000));

    __disable_irq();
    z_arm_reset();

    /* should not be reached */
    printf("Reboot failed\n");
    k_sleep(Z_FOREVER);

    return 0;
}
```

And this indeeds behaves incorrectly as the sleep is not stopping the thread for the requested amount of time.

However, `z_arm_reset` is not a proper way to reset the system, `sys_reboot(_)` from `zephyr/sys/reboot.h` is. Calling `sys_reboot(SYS_REBOOT_WARM)` or `sys_reboot(SYS_REBOOT_COLD)` does work as expected.

With reguards to chain loading, I also tried MCUBoot with the helloworld (using `sys_reboot(SYS_REBOOT_WARM)` and `sys_reboot(SYS_REBOOT_COLD)`) and it all seems to work as expected too.

For the record, here’s how I built & flashed them:
```shell
west build --sysbuild -p -b nucleo_f401re samples/hello_world -DSB_BOOTLOADER=MCUBOOT=y
pyocd load build/mcuboot/zephyr/zephyr.elf
pyocd load build/hello_world/zephyr/zephyr.signed.hex
```

It might be worth noting that the documentation for `BOOTLOADER_MCUBOOT` [states](https://github.com/zephyrproject-rtos/zephyr/blob/37e3449a393710df6339bd97cfc73679c63546ca/modules/Kconfig.mcuboot#L31-L34) that:

> By default, this option instructs Zephyr to initialize the core architecture HW registers during boot, when this is supported by the application. This removes the need by MCUboot to reset the core registers' state itself.

Could this be something useful for your usecase?

duda-patryk, 2024-04-23:
MCUBoot works because they use `irq_lock()` to disable interrupts (faults are enabled) https://github.com/mcu-tools/mcuboot/blob/d2e69bf720ec4e5825f1bd7be441a9e3dedb5533/boot/zephyr/main.c#L205

@yperess do you remember why we disable interrupts using `cpsid i`?

ithinuel, 2024-04-23:
[`CONFIG_MCUBOOT_CLEANUP_ARM_CORE`](https://github.com/mcu-tools/mcuboot/blob/d2e69bf720ec4e5825f1bd7be441a9e3dedb5533/boot/zephyr/Kconfig#L151-L165) defaults to `y`.

When the cleanup is enabled, [`do_boot()`](https://github.com/mcu-tools/mcuboot/blob/d2e69bf720ec4e5825f1bd7be441a9e3dedb5533/boot/zephyr/main.c#L180C5-L206) calls [`cleanup_arm_nvic()`](https://github.com/mcu-tools/mcuboot/blob/d2e69bf720ec4e5825f1bd7be441a9e3dedb5533/boot/zephyr/arm_cleanup.c#L14-L27) which disables interrupts too 🤔.

duda-patryk, 2024-04-23:
> When the cleanup is enabled, [do_boot()](https://github.com/mcu-tools/mcuboot/blob/d2e69bf720ec4e5825f1bd7be441a9e3dedb5533/boot/zephyr/main.c#L180C5-L206) calls [...]

`__disable_irq()` is called indeed. @ithinuel I assume that it works for you :sweat_smile:

ithinuel, 2024-04-23:
Well, I cannot reproduce the issue but from the bug report what's missing is enabling the IRQ so if anything it only confuses me more.

ithinuel, 2024-04-26:
Are you happy closing this issue with the fix that was merged?

closed 2024-04-30 by aescolar