zephyrproject-rtos/zephyr #71743: Hard fault (reason=25) when the thread is waiting for the message queue

lnhao, 2024-04-22:
**Describe the bug**
The recurrence of this bug is irregular, and I don't know when it will occur. Some devices have been running for more than 4 days and are still functioning normally. But some devices may experience this after running for more than a day. The function entry I checked during debugging was generated from the **k_msgq_get()** function. There is very little other useful information. It's not like a thread stack overflow.

 - What target platform are you using?  -> STM32WB55
 - What have you tried to diagnose or workaround this issue?  -> west debug / codedump.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a stm32wb55 board to PC. 
2. cd ./zephyrproject/app_stm32wb55/application/app
3. west build -b board -p
4. west flash
5. west debug

**Expected behavior**
I don't think the program should enter the Hard Fault handler. 

**Impact**
The customer's device has also encountered this bug.

**Logs and console output**
```
(gdb) info stack

#0  arch_system_halt (reason=25) at /home/alexliao/git/zephyrproject/app_stm32wb55/application/app/src/main.c:65

#1  0x080367ec in k_sys_fatal_error_handler (reason=reason@entry=25, esf=esf@entry=0x200101c0 <z_interrupt_stacks+2048>) at /home/alexliao/git/zephyrproject/zephyr/kernel/fatal.c:46

#2  0x0803681a in z_fatal_error (reason=reason@entry=25, esf=esf@entry=0x200101c0 <z_interrupt_stacks+2048>) at /home/alexliao/git/zephyrproject/zephyr/kernel/fatal.c:131

#3  0x0803347e in z_arm_fatal_error (reason=reason@entry=25, esf=esf@entry=0x200101c0 <z_interrupt_stacks+2048>) at /home/alexliao/git/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:63

#4  0x08017b74 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>)

    at /home/alexliao/git/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1138

#5  0x08017c58 in z_arm_usage_fault () at /home/alexliao/git/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102

#6  <signal handler called>

#7  arch_swap (key=0) at /home/alexliao/git/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53

#8  0x08026afa in z_swap_irqlock (key=<optimized out>) at /home/alexliao/git/zephyrproject/zephyr/kernel/include/kswap.h:185

#9  0x080263de in z_impl_k_msgq_get (msgq=msgq@entry=0x20003078 <sys_event>, data=data@entry=0x2000f954 <app_main_thread_stack+5076>, timeout=...)

    at /home/alexliao/git/zephyrproject/zephyr/kernel/msg_q.c:260

#10 0x08015dca in k_msgq_get (timeout=..., data=0x2000f954 <app_main_thread_stack+5076>, msgq=0x20003078 <sys_event>)

    at /home/alexliao/git/zephyrproject/app_stm32wb55/application/app/build/zephyr/include/generated/syscalls/kernel.h:1176

#11 sys_event_message_receive (rmsg=rmsg@entry=0x2000f98c <app_main_thread_stack+5132>, msg_len=msg_len@entry=0x2000f98a <app_main_thread_stack+5130>, timeout=...)

    at /home/alexliao/git/zephyrproject/app_stm32wb55/application/app/src/msg.c:167

#12 0x080139d4 in app_main_thread () at /home/alexliao/git/zephyrproject/app_stm32wb55/application/app/src/main_thread.c:599

#13 0x08032c90 in z_thread_entry (entry=0x8013889 <app_main_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/alexliao/git/zephyrproject/zephyr/lib/os/thread_entry.c:36

#14 0x08032c90 in z_thread_entry (entry=0x8013889 <app_main_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/alexliao/git/zephyrproject/zephyr/lib/os/thread_entry.c:36

#15 0x94fb2920 in ?? ()

Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

**Environment (please complete the following information):**
- > OS: Ubuntu
- > Toolchain: zephyr-sdk-0.16.0
- > Zephyr version: v3.3.0

aescolar, 2024-04-23:
@lnhao without a reproduction case it may be quite difficult for anybody to guess what the problem is.
Are you able to provide one?

peter-mitsis, 2024-04-24:
Based on the limited information so far, there is not much that I can add yet. That being said, it looks like exception (which according to the 'reason' in the debug dump is a "precise data bus error") occurred at line 53 of swap.c in arch_swap(). This corresponds to ..

`return _current->arch.swap_return_value;`

Not being an ARM expert, I am uncertain what causes this error. However, given the code location I am suspicious of the `_current` pointer. Perhaps this pointer became corrupt in some way?

One other thing of note as far as execution flow goes, slightly earlier in arch_swap(), we generated a PENDSV interrupt (necessary for scheduling). If I remember things correctly, PENDSV is the lowest priority interrupt, which makes it ideal for scheduling. And in arch_swap(), it does not get serviced until the interrupts are unlocked via `irq_unlock(0);`.

Given the location, I do wonder if this is reproducible on other cortex-m platforms as this code is pretty foundational to it.

lnhao, 2024-04-25:
@aescolar @peter-mitsis thank you all. This problem is difficult to reproduce. Some devices have not been replicated yet. Some devices can be replicated after running for 24 hours. I cannot provide a sample at the moment; it will take some more time.

peter-mitsis, 2024-05-07:
@lnhao - Figured I would check in and see if there is any additional information that can be added to this issue.

teburd, 2024-06-04:
Maybe related #70077

ihtamnetu, 2024-06-26:
[image]

what happens if you read and then write smth with hal i2c driver in interrupt mode?

mathieuchopstm, 2024-07-03:
> Based on the limited information so far, there is not much that I can add yet. That being said, it looks like exception (which according to the 'reason' in the debug dump is a "precise data bus [...]

This rings a bell. I don't remember exact details but I have encountered a similar issue recently, although in a completely different context. The core problem was a stack overflow that corrupted `_kernel.cpus[0].idle_thread`. When the kernel attempted to context switch to idle thread, because no other thread was schedulable, it would panic in a similar fashion.

I'm not sure that this is the issue here, but it could be worth investigating:
1. Is `CONFIG_HW_STACK_PROTECTION` enabled in the application?
If no, you may be encountering the same issue as I was - try increasing stack size.
(seeing `app_main_thread_stack+5076` in log, I doubt this is the problem though)
2. During a `west debug` session, place a watchpoint (i.e., `w <symbol>`) on `_kernel.cpus[0].idle_thread`.
It should be hit only once in `z_init_cpu` - anything else is most likely corruption.
Another candidate for watchpoints is `_kernel.cpus[0].current`, but this one may legitimately be written during application lifetime. Printing its value could be enough to reveal if it is wrong, though.
3. From the call stack, I am deducing the call flow leading to the panic is `z_impl_k_msgq_get` -> (`z_reschedule`/`z_pend_curr`) -> `z_swap` -> `z_swap_irqlock` -> `arch_swap` - breakpoints placed in some of these functions might also help finding the issue

closed 2024-09-16 by github-actions[bot]

daleonpz, 2025-04-01:
I see the error always when I flash mcuboot

```
west build -b nucleo_wb55rg  -s bootloader/mcuboot/boot/zephyr -d build-mcuboot
west flash -d build-mcuboot
```

here is the trace

```
Program received signal SIGINT, Interrupt.
arch_system_halt (reason=25) at /home/dnl/Documents/git/nix_zephyr/zephyr/kernel/fatal.c:30
30              for (;;) {
(gdb) bt
#0  arch_system_halt (reason=25) at /home/dnl/Documents/git/nix_zephyr/zephyr/kernel/fatal.c:30
#1  0x080071be in k_sys_fatal_error_handler (reason=<optimized out>, esf=<optimized out>) at /home/dnl/Documents/git/nix_zephyr/zephyr/kernel/fatal.c:44
#2  0x08004154 in z_fatal_error (reason=<optimized out>, esf=<optimized out>) at /home/dnl/Documents/git/nix_zephyr/zephyr/kernel/fatal.c:119
#3  0x08005546 in z_arm_fatal_error (reason=reason@entry=25, esf=esf@entry=0x20003288 <z_interrupt_stacks+2056>) at /home/dnl/Documents/git/nix_zephyr/zephyr/arch/arm/core/fatal.c:86
#4  0x08001bc2 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>)
    at /home/dnl/Documents/git/nix_zephyr/zephyr/arch/arm/core/cortex_m/fault.c:1080
#5  0x08001c8c in z_arm_usage_fault () at /home/dnl/Documents/git/nix_zephyr/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  memcpy (d=d@entry=0x20005b78 <z_main_stack+10040>, s=s@entry=0x80d7ff0, n=n@entry=16) at /home/dnl/Documents/git/nix_zephyr/zephyr/lib/libc/minimal/source/string/string.c:319
#8  0x080058e8 in flash_stm32_read (dev=<optimized out>, offset=884720, data=0x20005b78 <z_main_stack+10040>, len=16)
    at /home/dnl/Documents/git/nix_zephyr/zephyr/drivers/flash/flash_stm32.c:154
#9  0x08003f98 in boot_read_swap_state (fap=0x8007870 <default_flash_map+48>, state=state@entry=0x20005ba8 <z_main_stack+10088>)
    at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/bootutil_public.c:252
#10 0x08001032 in swap_status_source (state=state@entry=0x20000fa4 <boot_data>) at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/swap_scratch.c:364
#11 0x08005250 in swap_read_status (state=state@entry=0x20000fa4 <boot_data>, bs=bs@entry=0x20005bf0 <z_main_stack+10160>)
    at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/swap_misc.c:161
#12 0x08000c44 in boot_prepare_image_for_update (bs=0x20005bf0 <z_main_stack+10160>, state=0x20000fa4 <boot_data>)
    at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/loader.c:2181
#13 context_boot_go (state=state@entry=0x20000fa4 <boot_data>, rsp=rsp@entry=0x20005c34 <z_main_stack+10228>)
    at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/loader.c:2495
#14 0x08000e2c in boot_go (rsp=rsp@entry=0x20005c34 <z_main_stack+10228>) at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/bootutil/src/loader.c:3196
#15 0x08000426 in main () at /home/dnl/Documents/git/nix_zephyr/bootloader/mcuboot/boot/zephyr/main.c:531
```

Zephyr: 1bdffce3b296f74d466a62869e335c87e5f6cfae 
Zephyr SDK: 0.17.0
West version: v1.3.0