zephyrproject-rtos/zephyr #71744: samples/net/sockets/dumb_http_server: mimxrt1015_evk: build failure

hakehuang, 2024-04-22:
**Describe the bug**
samples/net/sockets/dumb_http_server: mimxrt1015_evk: build failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1015_evk
 - What have you tried to diagnose or workaround this issue?
 - NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
c4766f9572180e1a07d3ef3bb499a2bc48d1cee9 is the first bad commit
commit c4766f9572180e1a07d3ef3bb499a2bc48d1cee9
Author: Martin Åberg <martin.aberg@gaisler.com>
Date:   Wed Apr 17 13:01:25 2024 +0200

    scripts/checkpatch: Improve blank line check after declarations
    
    This adds volatile type qualifier to the struct declaration matching
    when looking for blank line after declarations.
    
    Before this commit, this structure was not accepted by chheckpatch:
      struct cfg {
             struct gpio_driver_config common;
             volatile struct grgpio_regs *regs;
             int interrupt;
      };
    
    checkpatch.pl generated the following warning:
      -:158: WARNING:LINE_SPACING: Missing a blank line after declarations
      #158: FILE: drivers/gpio/gpio_grgpio2.c:27:
      +     struct gpio_driver_config common;
      +     volatile struct grgpio_regs *regs;
    
    With this commit, the warning is no longer generated.
    
    Signed-off-by: Martin Åberg <martin.aberg@gaisler.com>

 scripts/checkpatch.pl | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

**To Reproduce**
Steps to reproduce the behavior:
west build -b mimxrt1015_evk samples/net/sockets/dumb_http_server

**Expected behavior**
build PASS

**Impact**
net functions

**Logs and console output**

```
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/soc/nxp/imxrt/imxrt10xx/soc.c: In function 'clock_init':
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/soc/nxp/imxrt/imxrt10xx/soc.c:216:39: error: 'kIOMUXC_GPR_ENET1TxClkOutputDir' undeclared (first use in this function); did you mean 'kIOMUXC_GPR_SAI1MClkOutputDir'?
  216 |         IOMUXC_EnableMode(IOMUXC_GPR, kIOMUXC_GPR_ENET1TxClkOutputDir, true);
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       kIOMUXC_GPR_SAI1MClkOutputDir
/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/soc/nxp/imxrt/imxrt10xx/soc.c:216:39: note: each undeclared identifier is reported only once for each function it appears in
ninja: build stopped: subcommand failed.

INFO    - /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1015_evk/samples/net/sockets/dumb_http_server/sample.net.sockets.dumb_http_server.netusb/build.log
INFO    - 782 test scenarios (782 test instances) selected, 675 configurations skipped (616 by static filter, 59 at runtime).
INFO    - 105 of 782 test configurations passed (98.13%), 0 failed, 2 errored, 675 skipped with 0 warnings in 209.65 seconds
INFO    - 0 test configurations executed on platforms, 107 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples/net/sockets/dumb_http_server/sample.net.sockets.dumb_http_server.netusb on mimxrt1015_evk error (Build failure)
INFO    - 2) samples/net/sockets/dumb_http_server/sample.net.sockets.dumb_http_server.netusb_zeroconf on mimxrt1015_evk error (Build failure

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.6.0-2754-g0eb4dc53525d

decsny, 2024-04-24:
Hi, I can not reproduce this build failure on the current main or the commit listed as the first bad commit, so I will close this for now, feel free to reopen if you still see the failure, it seems like it may have been a mistake, I dont know.

hakehuang, 2024-04-24:
@decsny , it still fails at my CI. and I can also reproduce it locally with the same failure log.

decsny, 2024-04-24:
> @decsny , it still fails at my CI. and I can also reproduce it locally with the same failure log. the bisect result is not correct, as the case is unbuild able in former version with RAM overlfow error

is it with the command you put above? because I tried it and I really dont get any error

hakehuang, 2024-04-24:
> > @decsny , it still fails at my CI. and I can also reproduce it locally with the same failure log. the bisect result is not correct, as the case is unbuild able in former version with RAM overlfow error
> 
> @hakehuang is it with the command you put above? because I tried it and I really dont get any error

strange, now it is ok with v3.6.0-2909-ga33f9d25f0f. let close it.

hakehuang, 2024-04-25:
@decsny I get the right command line to reproduce this error
```
west build -b mimxrt1015_evk samples/net/sockets/dumb_http_server -- -DOVERLAY_CONFIG="overlay-netusb.conf;overlay-zeroconf.conf"
```

decsny, 2024-04-25:
> @decsny I get the right command line to reproduce this error
> 
> ```
> west build -b mimxrt1015_evk samples/net/sockets/dumb_http_server -- -DOVERLAY_CONFIG="overlay-netusb.conf;overlay-zeroconf.conf"
> ```
> 
> and this is a bug in the zephyr/soc/nxp/imxrt/imxrt10xx/soc.c as there is no kIOMUXC_GPR_ENET1TxClkOutputDir defined for 101x series platform, see below
> 
> ```
> ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/modules/hal/nxp/mcux$ grep -Rn kIOMUXC_GPR_ENET1TxClkOutputDir .
> ./mcux-sdk/devices/MIMXRT1064/drivers/fsl_iomuxc.h:1219:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET1_TX_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1021/drivers/fsl_iomuxc.h:808:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET_REF_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1041/drivers/fsl_iomuxc.h:955:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET1_TX_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1042/drivers/fsl_iomuxc.h:955:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET1_TX_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1061/drivers/fsl_iomuxc.h:1235:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET1_TX_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1024/drivers/fsl_iomuxc.h:754:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET_REF_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1051/drivers/fsl_iomuxc.h:1041:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET_REF_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1062/drivers/fsl_iomuxc.h:1235:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET1_TX_CLK_DIR_MASK,
> ./mcux-sdk/devices/MIMXRT1052/drivers/fsl_iomuxc.h:1041:    kIOMUXC_GPR_ENET1TxClkOutputDir       = IOMUXC_GPR_GPR1_ENET_REF_CLK_DIR_MASK,
> ```

okay , I think this PR from a few days ago will fix this: https://github.com/zephyrproject-rtos/zephyr/pull/71844

PR #71844 (2024-04-26): dts: rt1015: Remove ethernet node
There is not an ethernet on RT1015.

Fixes #71744
closed 2024-04-26 by fabiobaltieri