zephyrproject-rtos/zephyr #71749: Add modem shell support to the modem cellular driver

niclas-ats, 2024-04-22:
**Is your enhancement proposal related to a problem? Please describe.**
Some of the modem drivers support the interaction with the device at runtime using the modem shell. The generic modem cellular driver currently lacks this feature.

**Describe the solution you'd like**
Register a modem context and implement the neccessary api within the modem cellular driver so that the modem shell commands (list, send, ...) work at runtime. 

**Describe alternatives you've considered**
None.

**Additional context**
After looking into the code of the modem subsystem and drivers I have some questions:

1. The modem shell needs either a "modem context" or a "modem receiver". Which one is recommended? Is the "modem context" the successor of the "modem receiver"?
2. What would be the best way to implement the neccessary changes? Do we have to rewrite the communication logic to make it thread safe or can we use the modem_pipe as another backend?

If I understand more about the underlying architecture I could work on implementing this.

I'm mentioning @bjarki-trackunit as the main developer of the modem cellular driver. Maybe you can point me to the right direction ;)

bjarki-andreasen, 2024-11-05:
Hi, the `modem_cellular.c` supports the `at_shell.c` selected with `CONFIG_MODEM_AT_SHELL=y` :) Its used like
```
at AT
OK
```
first at being the shell command, second at being the command to send, and an optional third command being an expected response:
```
at AT OK
OK
```

bjarki-andreasen, 2024-11-05:
I see this comment is quite old, the AT shell was added a few months back, utilizing a newer feature `modem_pipelink` to connect a shell module (or anything really) to a free CMUX channel :)

closed 2024-11-05 by bjarki-andreasen