zephyrproject-rtos/zephyr #71754: Bluetooth: Support for "Object Changed Characteristic" in Object Transfer Service

sjanc, 2024-04-22:
**Is your enhancement proposal related to a problem? Please describe.**
Add support for "Object Changed Characteristic" in Object Transfer Service . This is optional feature.

**Describe alternatives you've considered**
Disable feature in launch Studio project.

**Additional context**
This is currently affecting following qualification test cases:
OTS/SR/SGGIT/CHA/BV-15-C
OTS/SR/CON/BV-03-C
OTS/SR/OC/BV-01-C

jhedberg, 2024-04-22:
@sjanc are you planning to implement this yourself, or do you know that someone else is planning to do it? That would be useful information to determine who is the most suitable assignee for this issue.

sjanc, 2024-04-22:
I'm not going to work on that yet (at least not in near future),  this is more of a triage on failing OTS qualification test cases.

As mentioned in enhancement description, alternative is simply to disable optional feature in test plan.

sjanc, 2024-04-25:
there are more discrepancies in what implementation support and what is currently enabled in our test plan,   I'll update test plan to match this and than close this issue.

sjanc, 2024-04-25:
actually, it looks like OTS support "Object Changed Characteristic" is needed due to MCP mandatory requirements...