zephyrproject-rtos/zephyr #71761: The west boards command fails when adding boards and socs in out-of-tree mode

foxriver1025, 2024-04-22:
**Describe the bug**
The west boards command fails when adding boards and socs in out-of-tree mode.
This error can be fixed by modifying the zephyr/scripts/west_commands/boards.py file.
In function 'do_run' add this codes:
```
for module in zephyr_module.parse_modules(ZEPHYR_BASE, self.manifest):
    soc_root = module.meta.get('build', {}).get('settings', {}).get('soc_root')
    if soc_root is not None:
        args.soc_roots.append(Path(module.project) / soc_root)
```
So I think the west boards command did not take into account that soc_root needs to be found in zephyr_module, and board_root has already done this.

**To Reproduce**
My workspce is as follows:
```
m88star5_work
├── m88star5-sdk
│   ├── app
│   ├── boards
│   │     └── montage-tech
│   │		└── qemu_tps
│   │			├── board.cmake
│   │			├── board.yml
│   │			├── doc
│   │			│   └── index.rst
│   │			├── Kconfig
│   │			├── Kconfig.defconfig
│   │			├── Kconfig.qemu_tps
│   │			├── qemu_tps_defconfig
│   │			├── qemu_tps.dts
│   │			└── qemu_tps.yaml
│   ├── CMakeLists.txt
│   ├── drivers
│   ├── dts
│   ├── include
│   ├── Kconfig    
│   ├── README.md
│   ├── soc
│   │	└── montage-tech
│   │		├── CMakeLists.txt
│   │		├── Kconfig
│   │		├── Kconfig.defconfig
│   │		├── Kconfig.soc
│   │		├── m88star5
│   │		│   ├── CMakeLists.txt
│   │		│   ├── common_linker
│   │		│   │   └── init.ld
│   │		│   ├── Kconfig
│   │		│   ├── Kconfig.defconfig
│   │		│   ├── Kconfig.defconfig.m88star5
│   │		│   ├── Kconfig.soc
│   │		│   ├── l2_cache.c
│   │		│   ├── linker.ld
│   │		│   ├── pma.c
│   │		│   ├── soc_context.h
│   │		│   ├── soc_irq.S
│   │		│   ├── soc_offsets.h
│   │		│   ├── soc_v5.h
│   │		│   └── start.S
│   │		└── soc.yml
│   ├── west.yml
│   └── zephyr
└── zephyr
```
I add a board named 'qemu_tps' and a soc named 'm88star5'.
The code in boards/montage-tech/qemu_tps/board.yml is as follows:
```
board:
  name: qemu_tps
  vendor: qemu
  socs:
    - name: m88star5
```
And the code in soc/montage-tech/soc.yml is as follows:
```
family:
- name: m88star5-family
  series:
  - name: m88star5
    socs:
    - name: m88star5
```
Then I used the west boards command and the following error occurred：
`ERROR: SoC 'm88star5' is not found, please ensure that the SoC exists and that soc-root containing 'm88star5' has been correctly defined.`

**Expected behavior**
The west boards command can list custom boards named 'qemu_tps'.

**Impact**
I cannot use the west boards command to list custom boards

**Logs and console output**
`ERROR: SoC 'm88star5' is not found, please ensure that the SoC exists and that soc-root containing 'm88star5' has been correctly defined.`

**Environment (please complete the following information):**
 - OS: (Ubuntu 22.04.3 LTS)
 - Toolchain (Zephyr SDK 0.16.3)
 - Commit SHA lastest

nordicjm, 2024-04-23:
Where is your `zephyr/module.yml` file that has e.g.
```
build:
  settings:
    board_root: .
    soc_root: .
```
?

foxriver1025, 2024-04-23:
@nordicjm The file is in _m88star5-sdk/zephyr/module.yml_, as follows:
```
name: m88star5-sdk
build:
  kconfig: Kconfig
  cmake: .
  settings:
    board_root: .
    dts_root: .
    soc_root: .
    #module_ext_root: .
samples:
  - samples
tests:
  - tests
boards:
  - boards
```

rexut, 2024-04-29:
The same behaviour with twister! `west twister --jobs 4 --verbose --inline-logs --testsuite-root zephyr/tests/kernel/mutex` exit with `ERROR: SoC 'stm32f777xx' is not found, please ensure that the SoC exists and that soc-root containing 'stm32f777xx' has been correctly defined.`

This looks like a similar wrong implementation as for out-of-tree `arch_roots` in `scripts/pylib/twister/twisterlib/testplan.py`.

https://github.com/zephyrproject-rtos/zephyr/blob/5033399322fe10745a2e37db4c7c31ecd9241e74/scripts/pylib/twister/twisterlib/testplan.py#L406-L410

1. in L406: `soc_roots` hard set to `[Path(ZEPHYR_BASE), Path(ZEPHYR_BASE) / 'subsys' / 'testsuite']` – **can't see any link to the 3rd party entry `soc_root` in one or multiple `<out-of-tree-folder>/zephyr/module.yml` files !!!**
2. in L410: `soc_roots` goes unchanged to the board find algo `list_boards.find_v2_boards()`

In both cases, arch and soc, only `ZEPHYR_BASE` will be respected – BUT NEVER anything from the well documented Zephyr Module files, or I'm wrong here?

rexut, 2024-04-29:
I found the same code sequences in:

https://github.com/zephyrproject-rtos/zephyr/blob/5033399322fe10745a2e37db4c7c31ecd9241e74/doc/_extensions/zephyr/kconfig/__init__.py#L116-L118

@nashif This all leads to a very important question for me and especially for contributors, maintainers and the TSC: **How reliable are the statements in the Zephyr documentation?** –> "Developing with Zephyr" –> "Modules (External projects)" –> "Build settings" (https://docs.zephyrproject.org/latest/develop/modules.html#build-settings)

`arch_root`, `soc_root`, `board_root`, `dts_root`, `…` and so on have to be interpreted (taken) exactly from all the multiple `module.yml` files. The sum off all that files is THE Build Settings Database, nothing else, for compilation, testing and documentation – DO NOT SET HARD TO `${ZEPHYR_BASE}` – that is a nightmare for years and with any new feature we start this kind of issue reports again and again.

I think, the maintainer needs a compliant check to hit out any code, that use hard setups with `${ZEPHYR_BASE}` (mostly hidden deep in Python scripts).

nashif, 2024-06-04:
@nordicjm  any update on this issue?

nordicjm, 2024-06-05:
@tejlmand

deyro, 2024-07-01:
Hi,

I am having a similar problem. In my case my SOC (stm32h753) is supported in zephyr but my board isn't. So, I just have a boards directory in my application. I had my out-of-tree board working fine with zephyr-v3.6.0 but today I updated to zephyr-v3.7.0-rc2 and the build stopped with the following error.

```
/home/acpuvm001/zephyrproject_main/zephyr/scripts/kconfig/kconfig.py: Kconfig.zephyr:27: '/home/acpuvm001/zephyr_workspace/trials/cmx655_sample/build/Kconfig/soc/Kconfig.defconfig' not found (in 'source "$(KCONFIG_BINARY_DIR)/soc/Kconfig.defconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is set to '/home/acpuvm001/zephyrproject_main/zephyr'). Also note that unset environment variables expand to the empty string.
```

Below is my application structure:
```
my_sample
	|- boards
	|	|- arm
	|		|- e206
	|			|- board.cmake
	|			|- e206_defconfig
	|			|- e206.dts
	|			|- e206.yaml
	|			|- Kconfig.board
	|			|- Kconfig.defconfig
	|			|- doc
	|			|- support
	|
	|- dts
	|	|- e206
	|		|- many dtsi files
	|
	|- modules
	|	|- zephyr
	|	|	|- module.yml
	|	|- custom drivers
	|
	|- src
	|
	|- CMakelist.txt
	|- Kconfig
	|- prj.conf
```

Contents of the main CMakeLists.txt file
```
cmake_minimum_required(VERSION 3.20.0)

# Add our modules directory to Zephyr Modules
list(APPEND ZEPHYR_EXTRA_MODULES ${CMAKE_CURRENT_SOURCE_DIR}/modules)

set(PROJ_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/)
set(DTS_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/dts)
set(BOARD_ROOT ${CMAKE_CURRENT_SOURCE_DIR})

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(cmx655_sample)

add_subdirectory(src)
```

What changes do I need to do to make it work?

Thanks

nordicjm, 2024-07-01:
You need to transition your board to hwmv2: https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#hw-model-v2

deyro, 2024-07-01:
@nordicjm Thank you! It worked.

nashif, 2024-08-16:
Reopening fort the twister part of the bug report.

PR #75371 (2024-07-05): scripts: west_commands: Fix west boards for SoC and arch in modules
Fixes #71761

The `west boards` command parses extra BOARD_ROOTs from Zephyr modules, so that the boards defined in those modules are automatically listed.

In HWMv2, OOT boards can be described in terms of OOT SoCs, which means that extra SOC_ROOTs must also be provided. Otherwise, an error message will be displayed when attempting to list all boards. Therefore, every Zephyr module SOC_ROOT should be included as well.

In HWMv1 (deprecated), OOT boards can be defined in terms of OOT archs, but module ARCH_ROOTs had never been [... cut]
commit 7a10c7df (2024-07-05): scripts: west_commands: Fix west boards for SoC and arch in modules

Fixes #71761

The `west boards` command parses extra BOARD_ROOTs from Zephyr modules,
so that the boards defined in those modules are automatically listed.

In HWMv2, OOT boards can be described in terms of OOT SoCs, which means
that extra SOC_ROOTs must also be provided. Otherwise, an error message
will be displayed when attempting to list all boards. Therefore, every
Zephyr module SOC_ROOT should be included as well.

In HWMv1 (deprecated), OOT boards can be defined in terms of OOT archs,
but module ARCH_ROOTs had never bee [... cut]
PR #77163 (2024-08-30): twister: setup soc/arch roots based on module settings
Get data for various roots from modules and do not hardcode location
when retrieving list of boards in testplan.

Fixes #71761
commit d145db01 (2024-08-30): twister: setup soc/arch roots based on module settings

Get data for various roots from modules and do not hardcode location
when retrieving list of boards in testplan.

Fixes #71761
closed 2024-08-30 by nashif