zephyrproject-rtos/zephyr #71777: sensors: Fully hardware handled triggers

teburd, 2024-04-22:
**Is your enhancement proposal related to a problem? Please describe.**
Some hardware allows for triggering DMA transfers from things such as GPIO triggers perhaps, where a GPIO software interrupt handler may not be needed at all to read data from a sensor fifo. This is useful in cases where it is costly to jump to an interrupt handler (xip from slow flash) or when cycles spent swapping into/out of ISR handling only to program the DMA/Bus controller could be better spent elsewhere.

**Describe the solution you'd like**
A platform agnostic way of allowing for this functionality without massively disrupting the sensor API or drivers.

Today most sensor drivers themselves directly connect and implement a GPIO ISR to then generically do bus transaction work. It would require programming, indirectly, bus transactions tied to an interrupt trigger in a descriptive format rather than by programming this in C.

E.g. describe that given a GPIO trigger I want to do a SPI transceive.

**Additional context**
It was noted by @PetervdPerk-NXP at EOSS2024 this would be very useful in the context of i.MX RT parts running from external flash (and likely others such as rp2040) where jumping into an ISR is costly.

@yperess @MaureenHelm might also be interested in the above.

RichardSWheatley, 2024-05-09:
@teburd Silabs has cascading triggers that require setup only and no code after that.
Is this what you are speaking of? For example:  GPIO input triggers timer which then triggers GPIO output.

This would be odd to do in Zephyr in any current driver context and I believe if you can describe a way to do this in DT and driver, that would be helpful.

teburd, 2024-05-09:
> @teburd Silabs has cascading triggers that require setup only and no code after that. Is this what you are speaking of? For example: GPIO input triggers timer which then triggers GPIO output.

Partially yes, but in this case another bus master (DMA)for doing SPI/I2C transfers...

e.g. when configuring a sensor device in zephyr to read when DRDY/FIFO_WM/etc occur (via GPIO interrupt) have the DMA then trigger an i2c/spi bus transfer.

This is hard in Zephyr for a few reasons...

1. Exclusive access to the i2c/spi bus isn't guaranteed
2. We don't have a generic platform agnostic API for doing this sort of thing

I think we mostly have 2 partially covered through the updated sensor API which lets you describe triggers that signal data collection should occur. The second half of that is a way to have the bus driver set this up for you. Do that maybe one way would be having an RTIO OP code be... a interrupt trigger await OP that links to a spi transaction/i2c transaction.

This still doesn't solve issue 1 though, bus exclusive access. Unless you say that waiting on an interrupt implies exclusive access until that interrupt occurs.

> 
> I believe this would be odd to do in Zephyr in any current driver context and if you can describe a way to do this in DT and driver, that would be helpful. I can provide code and context for that in Silabs. Or I can provide a Silabs contact for you to work with on a sample/test.

teburd, 2025-03-18:
I realized recently that exclusive access to the bus could be done by having the bus controller lend out a exclusive to use handle (struct rtio_iodev) that would take all precedence over the common one if checked out.

This might fix issue 1 with this idea but would require dedicated access to the bus. That's arguably the right thing to do anyways given that a seperate bus master would be manipulating the peripheral block (e.g. DMA).