zephyrproject-rtos/zephyr #71790: Need help to add mb1835b DSI display for board stm32u5a9j_dk and Introduce STM32U5 DSI host driver

jn-coueron-muxen, 2024-04-22:
**The display of STM32U5A9J-DK Discovery kit is not yet supported on zephyr.**
Today It's not possible to use the display because the dts and the display driver doesnt exists on zephyr

**I wish to be able to use the STM32U5A9J-DK Discovery kit with the sample lvgl for example**

**The alternative is to use the stm32Cube**
I have done a small project with stm32cube to check the display working, but i love zephyr to much to use stm32cube for my project.

**Additional context**
I'm trying to add the support of the STM32U5A9J-DK Discovery kit's display.
My work is avaible here : https://github.com/jn-coueron-muxen/zephyr/tree/zephyr-ltdc-dsi-stm32u5a9j-dk

**No build Error, No execution Error**
But the screen is still black.
The only thing visible is the activation of the backlight :

https://github.com/zephyrproject-rtos/zephyr/assets/163851055/f924ad7d-e86a-4493-ae06-80cfc37e57e0

**Ressources used**
The hx8379 driver code is based on the code of [hx8394](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/display/display_hx8394.c) and the code from [stm32u5x9j-dk-bsp](https://github.com/STMicroelectronics/stm32u5x9j-dk-bsp/blob/main/stm32u5x9j_discovery_lcd.c) from @STMicroelectronics

The display controller is a hx8379 from Himax. (proof [here](https://github.com/STMicroelectronics/stm32u5x9j-dk-bsp/blob/48797784d762244a243d4aaf5fd961cd28695f2c/stm32u5x9j_discovery_lcd.c#L1364))
The datasheet of the hx8379 to name and comment the cmd sended to the screen.
The reference manual of the stm32u5 ([rm0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u575585-armbased-32bit-mcus-stmicroelectronics.pdf))
The work done in zephyr for the stm32h7.

**Need help**
I don't known if the problem is on the host part or if it's my display driver is not working.
What test can i done to validate one part or an other ?
I need help and advices, maybe from @dariis and @FRASTM who's worked on the stm32h747i_disco or anyone (@ABOSTM, @erwango, @danieldegrasse, @marcusfolkesson) want to help my adding this display on zephyr.

**My Log Trace**

```
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config: DISPLAY: pix 20833 kHz, lane 62500 kHz
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config: HAL_DSI_Init setup:
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   AutomaticClockLaneControl 0x0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   TXEscapeCkdiv 4
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   NumberOfLanes 1
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   PLLNDIV 125
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   PLLIDF 4
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   PLLODF 2
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config: HAL_DSI_ConfigVideoMode setup:
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VirtualChannelID 0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   ColorCoding 0x5
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LooselyPacked 0x0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   Mode 0x2
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   PacketSize 480
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   NumberOfChunks 0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   NullPacketSize 4095
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   HSPolarity 0x0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VSPolarity 0x0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   DEPolarity 0x0
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   HorizontalSyncActive 6
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   HorizontalBackPorch 3
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   HorizontalLine 1452
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VerticalSyncActive 1
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VerticalBackPorch 12
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VerticalFrontPorch 50
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   VerticalActive 481
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPCommandEnable 0x8000
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPLargestPacketSize 64
[00:00:00.134,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPVACTLargestPacketSize 0
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPHorizontalFrontPorchEnable 0x2000
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPHorizontalBackPorchEnable 0x1000
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPVerticalActiveEnable 0x800
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPVerticalFrontPorchEnable 0x400
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPVerticalBackPorchEnable 0x200
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   LPVerticalSyncActiveEnable 0x100
[00:00:00.135,000] <dbg> dsi_stm32: mipi_dsi_stm32_log_config:   FrameBTAAcknowledgeEnable 0x4000
[... 50 lines cut ...]
[00:00:00.141,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len  2
                                    bd 02                                            |..               
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len 43
                                    c1 00 09 0f 18 21 2a 34  3c 45 4c 56 5e 66 6e 76 |.....!*4 <ELV^fnv
                                    7e 87 8e 95 9d a6 af b7  bd c5 ce d5 df e7 ee f4 |~....... ........
                                    fa ff 0c 31 83 3c 5b 56  1e 5a ff                |...1.<[V .Z.     
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len  2
                                    bd 00                                            |..               
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len  2
                                    bd 01                                            |..               
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len 43
                                    c1 00 08 0f 16 1f 28 31  39 41 48 51 59 60 68 70 |......(1 9AHQY`hp
                                    78 7f 87 8d 94 9c a3 ab  b3 b9 c1 c8 d0 d8 e0 e8 |x....... ........
                                    ee f5 3b 1a b6 a0 07 45  c5 37 00                |..;....E .7.     
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len  2
                                    bd 00                                            |..               
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x00, len 44
                                    c1 01 00 07 0f 16 1f 27  30 38 40 47 4e 56 5d 65 |.......' 08@GNV]e
                                    6d 74 7d 84 8a 90 99 a1  a9 b0 b6 bd c4 cd d4 dd |mt}..... ........
                                    e5 ec f3 36 07 1c c0 1b  01 f1 34 00             |...6.... ..4.    
[00:00:00.151,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x11, len  0
[00:00:00.271,000] <dbg> dsi_stm32: mipi_dsi_stm32_transfer: TX: ch   0, reg 0x29, len  0
*** Booting Zephyr OS build 1a35bc844862 ***

```

ajarmouni-st, 2024-05-07:
Hi @jn-coueron-muxen, there is this small quirk in case you are not aware of it

"The microcontroller software system must invert DSI.CK_N and DSI.CK_P lines together to align with DSI_V3 add-on boards. The DSI clock differential signals (_N and _P) are inverted on CN2 for layout reasons."

section 8.1 https://www.st.com/resource/en/user_manual/um2967-discovery-kits-with-stm32u5x9nj-mcus-stmicroelectronics.pdf

erwango, 2025-08-19:
Linked PR: https://github.com/zephyrproject-rtos/zephyr/pull/94433
Thanks @CharlesDias !

closed 2025-10-31 by JarmouniA

canpool, 2025-12-29:
My STM32U5A9J-DK board configuration is as follows:
1. stm32u5a9j_dk.conf
```
CONFIG_LV_COLOR_DEPTH_16=y
CONFIG_LV_Z_BITS_PER_PIXEL=16
```
2. stm32u5a9j_dk.overlay
```
#include <st/u5/stm32u5a9njhxq-pinctrl.dtsi>
#include <zephyr/dt-bindings/display/panel.h>
#include <zephyr/dt-bindings/mipi_dsi/mipi_dsi.h>

/ {
	chosen {
		zephyr,display = &ltdc;
	};
};

&ltdc {
	status = "okay";

	bl-ctrl-gpios = <&dsi_lcd_qsh_030 53 GPIO_ACTIVE_HIGH>;

	width = <480>;
	height = <480>;
	pixel-format = <PANEL_PIXEL_FORMAT_RGB_565>;

	def-back-color-red = <0x00>;
	def-back-color-green = <0x00>;
	def-back-color-blue = <0x00>;

	display-timings {
		compatible = "zephyr,panel-timing";

		de-active = <0>;
		pixelclk-active = <0>;
		hsync-active = <1>;
		vsync-active = <1>;
		hsync-len = <2>;
		vsync-len = <1>;
		hback-porch = <1>;
		vback-porch = <12>;
		hfront-porch = <1>;
		vfront-porch = <50>;
	};
};

/* as pixelclk */
&pll3 {
	status = "okay";
	div-m = <4>;
	mul-n = <125>;
	div-p = <8>;
	div-q = <8>;
	div-r = <24>;
	clocks = <&clk_hse>;
};

&mipi_dsi {
	status = "okay";

	host-timeouts = <1 0 0 0 0 0 0 0 0>;
	phy-timings = <11 40 12 23 0 7>;

	/*
	 * Required D-PHY properties from driver:
	 * These are checked in st_stm32u5_mipi_dsi.yaml
	 */
	phy-freq-range = <8>;           /* DSI_DPHY_FRANGE_450MHZ_510MHZ */
	phy-low-power-offset = <0>;     /* PHY_LP_OFFSSET_0_CLKP */

	/*
	 * Required PLL properties from driver:
	 * These are specific to STM32U5
	 */
	pll-vco-range = <1>;            /* DSI_DPHY_VCO_FRANGE_800MHZ_1GHZ */
	pll-charge-pump = <0>;          /* DSI_PLL_CHARGE_PUMP_2000HZ_4400HZ */
	pll-tuning = <0>;               /* DSI_PLL_LOOP_FILTER_2000HZ_4400HZ */

	/* DSI PLL configuration */
	pll-ndiv = <125>;
	pll-idf = <4>; 					/* DSI_PLL_IN_DIV4 */
	pll-odf = <2>;					/* DSI_PLL_OUT_DIV2 */

	/* Signal polarity - driver reads these */
	hs-active-high;
	vs-active-high;
	de-active-high;
	largest-packet-size = <0>;

	hx8379c_panel: panel@0 {
		compatible = "himax,hx8379c";
		reg = <0x0>;

		/* Display dimensions - required by driver */
		width = <480>;
		height = <480>;

		/* Control pins */
		reset-gpios = <&dsi_lcd_qsh_030 57 GPIO_ACTIVE_LOW>;

		/* DSI interface parameters */
		data-lanes = <2>;
		pixel-format = <MIPI_DSI_PIXFMT_RGB565>;

		/* Display timing - required by driver */
		display-timings {
			compatible = "zephyr,panel-timing";

			de-active = <0>;
			pixelclk-active = <0>;
			hsync-active = <1>;
			vsync-active = <1>;
			hsync-len = <2>;
			vsync-len = <1>;
			hback-porch = <1>;
			vback-porch = <12>;
			hfront-porch = <1>;
			vfront-porch = <50>;
		};

		status = "okay";
	};
};
```
I also encountered the problem that the screen kept going black.

Moreover, when I adapted the stm32u5a9j_dk to the st_lcd_dsi_mb1835 shield, it still went black.

@jn-coueron-muxen @CharlesDias 

"The microcontroller software system must invert DSI.CK_N and DSI.CK_P lines together to align with DSI_V3 add-on boards. The DSI clock differential signals (_N and _P) are inverted on CN2 for layout reasons."
@ajarmouni-st What should I do？

JarmouniA, 2026-01-02:
> The active signal for my DSI reset pin is configured as low:
```
reset-gpios = <&dsi_lcd_qsh_030 57 GPIO_ACTIVE_LOW>;
```
> but the driver code for the HX8379C requires the active signal to be high

Then the description in the binding shall be corrected
https://github.com/zephyrproject-rtos/zephyr/blob/0601335d85ed22439671a25a23b8d550fb4893fd/dts/bindings/display/himax%2Chx8379c.yaml#L22

CharlesDias, 2026-01-02:
Hi @JarmouniA,

The correct DSI reset configuration is GPIO_ACTIVE_HIGH. Thus, the description in the binding file is not correct. My bad! :(

canpool, 2026-01-03:
``` 
west build -p -b stm32u5a9j_dk --shield st_lcd_dsi_mb1835 samples/drivers/display/
```
@CharlesDias I have already tested it.

[image]

But, my zephyr version is v4.3.0, and I downloaded zephyr too slowly.

https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/shields/st_lcd_dsi_mb1835/st_lcd_dsi_mb1835.overlay#L118

I have initially supported touch input in my project ( https://github.com/canpool/pony_zephyr/commit/f3e5f206ad28579c11cfb82954e0a36621ee6c4e ), and I have tested and found that st1633i is incompatible with cf1133. I think my st1633i ( https://github.com/canpool/pony_zephyr/commit/12c61a6f2ebbebc3bfdc82532640d877b96cb5de ) is not perfect, but I don't have a datasheet.

CharlesDias, 2026-01-03:
Hi @canpool, thank you for the information.

This is strange because the display shield used in the both boards should be the same. Is this understanding correct @erwango and @JarmouniA?