zephyrproject-rtos/zephyr #71797: tests/ztest/base/testing.ztest.base.verbose_1 fails on qemu_arc/qemu_arc_em

aescolar, 2024-04-23:
**Describe the bug**
tests/ztest/base/testing.ztest.base.verbose_1 fails on qemu_arc/qemu_arc_em
fails.
Twister reports an "FAILED : unexpected eof"

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p qemu_arc/qemu_arc_em -s tests/ztest/base/testing.ztest.base.verbose_1 -T tests/ztest/base
4. See error

**Expected behavior**
No failure

**Impact**
Failure in CI

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/8793149990/job/24130590820#step:11:1636

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 22.04)
 - Toolchain : Zephyr SDK
 - Commit SHA or Version used: Main as of now (1ed452b295bedcd7d00bcf0f3c8a0c5a13651faf)

**Additional context**
Seems to have been failing for a while, even if main's CI runs were green (also present in e3e009a1990027f7592eb6c24b4d605e32ae70de )

nashif, 2024-04-23:
looks like there is a fault:

```
DEBUG   - QEMU (3609742): ===================================================================
DEBUG   - QEMU (3609742): RunID: e4ffaaec0ec40846fe29e04863adc890
DEBUG   - QEMU (3609742): PROJECT EXECUTION SUCCESSFUL
DEBUG   - QEMU (3609742): E: ***** Exception vector: 0x6, cause code: 0x0, parameter 0x4
DEBUG   - QEMU (3609742): E: Address 0x8040020c
DEBUG   - QEMU (3609742): E: EV_ProtV
DEBUG   - QEMU (3609742): E: Instruction fetch violation (MPU)
DEBUG   - QEMU (3609742): E:  r0: 0x8040020c  r1: 0x1  r2: 0x8040020c  r3: 0x8040020c
DEBUG   - QEMU (3609742): E:  r4: 0x0  r5: 0x1  r6: 0x0  r7: 0x1
DEBUG   - QEMU (3609742): E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0xa
DEBUG   - QEMU (3609742): E: r12: 0xffffffff r13: 0x8040020c  pc: 0x8040020c
DEBUG   - QEMU (3609742): E:  blink: 0x800006f2 status32: 0x8000201c
DEBUG   - QEMU (3609742): E: lp_end: 0x0 lp_start: 0x0 lp_count: 0x0
DEBUG   - QEMU (3609742): E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - QEMU (3609742): E: Current thread: 0x80400190 (idle)
DEBUG   - QEMU (3609742): E: Halting system
DEBUG   - QEMU (3609742) complete (passed) after 2.059034824371338 seconds
```

nashif, 2024-04-24:
bisected to 4593f0d71c906803c3bd4ccb707e1c41109c153c, which is different that the report above, interesting. This is just core movement, no functional change.

PR #71843 (2024-04-24): kernel: increase main stack size for ztest
stack has been on the edge when using ztest, increase to 1024.

Fixes #71797
commit 297ad518 (2024-04-24): kernel: increase main stack size for ztest on ARC

stack has been on the edge when using ztest, use 1024 for ARC as well.

Fixes #71797
closed 2024-04-24 by aescolar