zephyrproject-rtos/zephyr #71838: Error Cloning Repository during Zephyr Setup on Linux

Hafsa-shoaib989, 2024-04-23:
**GitHub Discussion:**
In GitHub Discussions, I found a similar issue, but they pertain to Windows environments. Despite following the steps outlined there, I still encounter the same error. You can find the discussion at: [](https://github.com/zephyrproject-rtos/zephyr/discussions/67654)

**Description:**
I followed all instructions and steps mentioned in the [](https://docs.zephyrproject.org/latest/develop/getting_started/index.html), but when attempting to initialize Zephyr in a Linux environment using the west init command, an error occurs during the cloning of the manifest repository from [](https://github.com/zephyrproject-rtos/zephyr).

**To Reproduce:**
- Execute west init ~/zephyrproject in the terminal.
- Observe the error during the cloning process.

**Expected behavior:**
The manifest repository should be cloned successfully without encountering any errors.

**Logs and console output:**
Upon running the command **west init ~/zephyrproject**, the following error occurs:

(.venv) masfiyan@masfiyan-HP-EliteBook-840-G2:~$ west init ~/zephyrproject
=== Initializing in /home/masfiyan/zephyrproject
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr
Cloning into '/home/masfiyan/zephyrproject/.west/manifest-tmp'...
remote: Enumerating objects: 1031744, done.
remote: Counting objects: 100% (735/735), done.
remote: Compressing objects: 100% (457/457), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
FATAL ERROR: command exited with status 128: git clone https://github.com/zephyrproject-rtos/zephyr /home/masfiyan/zephyrproject/.west/manifest-tmp

**Environment:**
OS: Linux

ericlinsechs, 2024-04-24:
Hi @Hafsa-shoaib989 ,

Try to use ssh instead of https for cloning:
`west init -m git@github.com:zephyrproject-rtos/zephyr ~/zephyrproject`

For generating and adding a new ssh key, please refer [Generating a new SSH key and adding it to the ssh-agent](https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent?platform=linux).

nordicjm, 2024-04-24:
Try the tip above, this is not a zephyr problem but a problem with your connection and/or ISP.

closed 2024-04-24 by nordicjm

hariprasath1163, 2024-09-26:
> I'm getting this error west init === Initializing in /home/ee213032/zephyrproject --- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr Cloning into '/home/ee213032/zephyrproject/.west/manifest-tmp'... error: RPC failed; HTTP 403 curl 22 The requested URL returned error: 403 fatal: expected 'packfile' FATAL ERROR: command exited with status 128: git clone https://github.com/zephyrproject-rtos/zephyr /home/ee213032/zephyrproject/.west/manifest-tmp

- In your workspace, you might added `.west` in *.gitignore* which cause this error. Remove the `.west` from the *.gitignore* and try it.

ahsanu123, 2024-12-15:
get this error either when use ssh or https, 

then i found solution with _west clone option_  `-o`, try to clone in `depth=1` 
```shell
west init ~/zephyr-sdk -o=--depth=1
```

Vishnu-Opsera, 2025-03-10:
@ahsanu123 did this resolve the issue for you? or you have any other suggestions?

ahsanu123, 2025-03-12:
@Vishnu-Opsera I Believe it solve my issues at that time (bit forgot now been 3 month already)