zephyrproject-rtos/zephyr #71858: drivers: uart: uart_pm: drivers.uart.pm.* fail on the nrf52840dk/nrf52840 and nrf54l15pdk/nrf54l15/cpuapp

katgiadla, 2024-04-24:
**Describe the bug**
The tests:
- drivers.uart.pm`
- `drivers.uart.pm.async`
- `drivers.uart.pm.int_driven`

fail on the `nrf52840dk/nrf52840` and `nrf54l15pdk/nrf54l15/cpuapp`.

Additionally, the tests:

- `drivers.uart.pm.enhanced_poll`
- `drivers.uart.pm.int_driven.enhanced_poll`
- `drivers.uart.pm.no_rxpin`
- `drivers.uart.pm.async.enhanced_poll`

fail on the `nrf52840dk/nrf52840`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk/nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/uart/uart_pm -p nrf52840dk/nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
In file included from ../../../../../../../zephyr/include/zephyr/sw_isr_table.h:18,
                 from ../../../../../../../zephyr/include/zephyr/arch/arm/irq.h:19,
                 from ../../../../../../../zephyr/include/zephyr/arch/arm/arch.h:27,
                 from ../../../../../../../zephyr/include/zephyr/arch/cpu.h:19,
                 from ../../../../../../../zephyr/include/zephyr/kernel_includes.h:36,
                 from ../../../../../../../zephyr/include/zephyr/kernel.h:17,
                 from ../../../../../../../zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from ../../../../../../../zephyr/arch/arm/core/offsets/offsets.c:9:
../../../../../../../zephyr/include/zephyr/device.h:785:15: error: 'device_init' declared 'static' but never defined [-Werror=unused-function]
  785 | __syscall int device_init(const struct device *dev);
      |               ^~~~~~~~~~~
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDk 0.15.6
 - Commit SHA or Version used: `v3.6.0-2817-gdbf62a5a2a84`

nordic-krch, 2024-04-24:
it seems that it's due to errors when doing an incremental build (#64729).

katgiadla, 2024-04-24:
Related on #64729

closed 2024-04-24 by katgiadla