zephyrproject-rtos/zephyr #71888: STM32: ethernet: Unexpected stackoverflow error after flashing.

lucasAbadFr, 2024-04-24:
**Describe the bug**

In the context of developing a complex networking application (not a sample), an unexpected stack overflow error occurs after flashing the application. Surprisingly, this error occurs even before the main function or any other application-related code starts execution. To troubleshoot, I attempted a simpler approach by compiling Zephyr with the same configuration but for the `hello_world` sample.

- **Platform:** Nucleo H563ZI
    - **eth port:** Connected to my host machine with a Ethernet cable.
- **Diagnostic:** I attached a debugger and found that the last calls before the error were in `/home/user/zephyrproject/zephyr/drivers/ethernet/eth_stm32_hal.c` file in `rx_thread`. I will give more details below.
- **Regression:** I'm quite new to Zephyr, so I can't tell if this is a regression. I'm using the latest SDK version, `zephyr-sdk-0.16.5-1` and `Zephyr OS build v3.6.0-2908-g6a80a52b6cbe`.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `hello_world` sample:
2. Edit the `proj.conf` file to include the following configuration:
    ```conf
    # Log config
    CONFIG_PRINTK=y
    CONFIG_LOG=y
    CONFIG_LOG_MODE_IMMEDIATE=y
    CONFIG_NET_LOG=y

    CONFIG_MAIN_STACK_SIZE=131072
    CONFIG_REQUIRES_FULL_LIBC=y

    # Networking config
    CONFIG_NETWORKING=y
    CONFIG_NET_IPV4=y
    CONFIG_NET_IPV6=y
    CONFIG_NET_TCP=y
    CONFIG_NET_SOCKETS=y
    CONFIG_POSIX_API=n

    # Stack conf
    CONFIG_NO_OPTIMIZATIONS=y
    CONFIG_STACK_SENTINEL=y
    CONFIG_HW_STACK_PROTECTION=y
    CONFIG_STACK_CANARIES=y
    # CONFIG_ISR_STACK_SIZE=4096

    # Network driver config
    CONFIG_TEST_RANDOM_GENERATOR=y

    # Network address config
    CONFIG_NET_CONFIG_SETTINGS=y
    CONFIG_NET_CONFIG_NEED_IPV4=y
    CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
    CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
    CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"

    # CONFIG_DNS_RESOLVER=y
    # CONFIG_DNS_SERVER_IP_ADDRESSES=y
    # CONFIG_DNS_SERVER1="192.0.2.2"

    # Config init stack
    # CONFIG_INIT_STACKS=y 
    # CONFIG_NET_PKT_RX_COUNT=200
    # CONFIG_NET_PKT_TX_COUNT=200
    # CONFIG_NET_BUF_RX_COUNT=200
    # CONFIG_NET_BUF_TX_COUNT=200

    # Debug
    CONFIG_DEBUG=y
    ```
    > 📄 **Notes:** The main stack size is 128KB but can be more the error still occurs. Globaly the others configuration comes from other networking samples.
3. Build the sample 
    ```bash
    west build -b nucleo_h563zi -p always samples/hello_world/
    ```
4. Flash the sample
    ```bash
    west flash
    ```
5. Observe the error with a serial monitor or a debugger.
    ```bash
    minicom -D /dev/ttyACM0
    ```
**Expected behavior**
I expect the application to run without any error and the Zephyr OS to boot successfully.

**Impact**
This issue is blocking me from developing my application. I can't move forward without understanding the root cause of this error.

**Logs and console output**
1. **Stack overflow error:**
```bash
*** Booting Zephyr OS build v3.6.0-2908-g6a80a52b6cbe ***
[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.012,000] <inf> net_config: Waiting interface 1 (0x200013d8) to be up...
[00:00:01.628,000] <err> os: ***** USAGE FAULT *****
[00:00:01.633,000] <err> os:   Stack overflow (context area not valid)
[00:00:01.640,000] <err> os: r0/a1:  0x080588e6  r1/a2:  0x01000000  r2/a3:  0x20027498
[00:00:01.649,000] <err> os: r3/a4:  0x08058bdb r12/ip:  0x0802e509 r14/lr:  0x20001918
[00:00:01.658,000] <err> os:  xpsr:  0xe8ec8c00
[00:00:01.663,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:01.671,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:01.679,000] <err> os: Current thread: 0x20001d20 (unknown)
[00:00:01.685,000] <err> os: Halting system
```

2. **Bus fault error:** that happened when I tried to debug the error:
```bash	
*** Booting Zephyr OS build v3.6.0-2886-geddb3822d2aa ***
[00:00:00.006,000] <inf> net_config: Initializing network
[00:00:00.012,000] <inf> net_config: Waiting interface 1 (0x20002b74) to be up...
[00:00:00.001,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.009,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.018,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.026,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.034,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.043,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.051,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.059,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.068,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.076,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.084,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.093,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.101,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.109,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.118,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.126,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.134,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.143,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.151,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.159,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.168,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.176,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.184,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.193,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.201,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.210,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.218,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.226,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.235,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:00.243,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
*** Booting Zephyr OS build v3.6.0-2886-geddb3822d2aa ***
[00:00:00.257,000] <inf> net_config: Initializing network
[00:00:00.263,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.272,000] <err> os: ***** BUS FAULT *****
[00:00:00.277,000] <err> os:   Precise data bus error
[00:00:00.283,000] <err> os:   BFAR Address: 0x0
[00:00:00.288,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x20029158  r2/a3:  0x00000001
[00:00:00.297,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000110 r14/lr:  0x08019c81
[00:00:00.306,000] <err> os:  xpsr:  0x61000000
[00:00:00.311,000] <err> os: Faulting instruction address (r15/pc): 0x08019c28
[00:00:00.319,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:00.327,000] <err> os: Current thread: 0x20002be0 (unknown)
[00:00:00.334,000] <err> os: Halting system
```
> 📄 **Notes:** Strangely, unable to reproduce at will, happened around 5 times.

**Environment (please complete the following information):**
 - OS: Linux (WSL2 Ubuntu 22.04)
 - Toolchain: 
    - sdk: zephyr-sdk-0.16.5-1
    - CMake version: 3.29.2
    - zephyr version: 3.6.99
    - Commit: latest 

**Additional context**

* I observed that If I used the debugger to pass step by step, the error does not occur instantaneaously, but instead I can see a bunch of this logs:
    ```bash
    [00:00:00.009,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
    ```
    If I continue to pass, the error occurs.   
    > 📄 **Notes:** I got `errno`  set to `ENETDOWN`, but the interface should be up.
* I was not able to find the exact place were the error occurs. But the code seem to be stuck in an infinite loop here:

https://github.com/zephyrproject-rtos/zephyr/blob/a33f9d25f0f0af70a9310faac81e9098d933e68a/drivers/ethernet/eth_stm32_hal.c#L831-L844

* I also tried various configuration flags that are commented in the `proj.conf` file, but the error still persists.

I'm not sure what is causing the error, and how to investigate further as I'm quite new to Zephyr. In fact It may be related to a configuration issue.

Any help would be appreciated. Thanks.

erwango, 2024-04-30:
I'd suggest to build a supported net sample (`samples/net/dns_resolve`), see if problem occurs and start from there.

erwango, 2024-06-17:
@lucasAbadFr Is this problem still valid or did you manage to find the issue in your sample ?

lucasAbadFr, 2024-06-19:
Hello @erwango.
I managed to make my application work.
Unfortunately, I don't have time to further investigate why building the network stack lead to such a behaviour.
I will close the issue if you consider the problem as solved.

erwango, 2024-06-19:
> Hello @erwango.
I managed to make my application work.
Unfortunately, I don't have time to further investigate why building the network stack lead to such a behaviour.
I will close the issue if you consider the problem as solved.

Well indeed, if this was only seen in your initial configuration, I think we should close as we don't have other reports in the same direction, so likely a configuration issue in your end.

closed 2024-06-19 by erwango