zephyrproject-rtos/zephyr #71891: New experimental USB device support broken for rpi_pico_w?

SamsonBox, 2024-04-24:
**Describe the bug**
I try to use the new USB device support for the rpi_pico_w, but I alway get the following link error:

undefined reference to `__device_dts_ord_62'
A look to the devicetree_generated.h shows, that no struct device is created for the /soc/usbd@50100000.

I see the same issue when i try to build the "samples\subsys\usb\shell" or the "samples\subsys\usb\cdc_acm" (in the later case using the C:\Users\pvdho\Documents\Programme\zephyrproject\zephyr\samples\subsys\usb\cdc_acm\usbd_next_prj.conf)

Do I miss something?

**To Reproduce**
checkout commit 4b276c496201c27abfde4cd8cff2b3d56c8a08bf. 
west  build -p always -b rpi_pico_w samples\subsys\usb\shell

**Expected behavior**
Complete linked binary

**Logs and console output**
:/progra~1/zephyr/zephyr~1.5-1/arm-ze~1/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/subsys/usb/device_next/libsubsys__usb__device_next.a(usbd_shell.c.obj):(._usbd_contex.static.sh_uds_ctx_+0x18): undefined reference to `__device_dts_ord_62'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\....\Documents\Programme\zephyrproject\zephyr\build'

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Commit 4b276c496201c27abfde4cd8cff2b3d56c8a08bf

jfischer-no, 2024-05-21:
The rpi_pico USB device controller driver is not ported to the new device support.

closed 2024-05-21 by jfischer-no

SamsonBox, 2024-05-23:
@jfischer-no Is there any documentation which devices are ported? Is there a miration guide to the new device support?