zephyrproject-rtos/zephyr #71922: doc: Documentation issue in 'boards/espressif/esp32_devkitc_wroom/doc/index'

kalyanboddula, 2024-04-25:
**Describe the bug**
(.venv) C:\Users\KALYAN\zephyrproject\zephyr>west espressif monitor
Serial port COM9
Connecting...
COM9 failed to connect: Write timeout
Serial port COM6
Connecting...............
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Traceback (most recent call last):
  File "C:\Users\KALYAN\zephyrproject\zephyr\..\modules\hal\espressif\tools\idf_monitor.py", line 34, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
Traceback (most recent call last):
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 165, in do_run
    self.monitor(module_path, args)
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 184, in monitor
    cmd_exec(("python.exe", monitor_path, "-p", esp_port,
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 40, in cmd_exec
    return subprocess.check_call(cmd, cwd=cwd, shell=shell)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '('python.exe', WindowsPath('C:/Users/KALYAN/zephyrproject/zephyr/../modules/hal/espressif/tools/idf_monitor.py'), '-p', 'COM6', '-b', '115200', WindowsPath('C:/Users/KALYAN/zephyrproject/zephyr/build/zephyr/zephyr.elf'), '--eol', 'CRLF')' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\KALYAN\zephyrproject\.venv\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 244, in run
    self.run_command(argv, early_args)
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 543, in run_command
    f'{quote_sh_list(cpe.cmd)}', fatal=True)
       ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\util.py", line 42, in quote_sh_list
    return ' '.join(shlex.quote(s) for s in cmd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\util.py", line 42, in <genexpr>
    return ' '.join(shlex.quote(s) for s in cmd)
                    ^^^^^^^^^^^^^^
  File "C:\Python311\Lib\shlex.py", line 329, in quote
    if _find_unsafe(s) is None:
       ^^^^^^^^^^^^^^^
TypeError: expected string or bytes-like object, got 'WindowsPath'

**Environment**

* Page: `boards/espressif/esp32_devkitc_wroom/doc/index`
* Version: main
* SHA-1: c0866030ebc9c0588eb349de34eb96af5317161e

I am facing this issue trying to serial monitor the hello_world code,facing no issue on flashing but monitor doesn't works,
can anyone suggest me resolve this issue.

nordicjm, 2024-04-25:
Did you follow the setup instructions as per https://docs.zephyrproject.org/latest/develop/getting_started/index.html#get-zephyr-and-install-python-dependencies ?

kalyanboddula, 2024-04-25:
C:\Windows\system32>cd %HOMEPATH%

C:\Users\KALYAN>python -m venv zephyrproject\.venv

C:\Users\KALYAN>zephyrproject\.venv\Scripts\activate.bat

(.venv) C:\Users\KALYAN>pip install west
Requirement already satisfied: west in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (1.2.0)
Requirement already satisfied: colorama in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from west) (0.4.6)
Requirement already satisfied: PyYAML>=5.1 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from west) (6.0.1)
Requirement already satisfied: pykwalify in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from west) (1.8.0)
Requirement already satisfied: setuptools in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from west) (65.5.0)
Requirement already satisfied: packaging in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from west) (24.0)
Requirement already satisfied: docopt>=0.6.2 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from pykwalify->west) (0.6.2)
Requirement already satisfied: python-dateutil>=2.8.0 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from pykwalify->west) (2.9.0.post0)
Requirement already satisfied: ruamel.yaml>=0.16.0 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from pykwalify->west) (0.18.6)
Requirement already satisfied: six>=1.5 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from python-dateutil>=2.8.0->pykwalify->west) (1.16.0)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from ruamel.yaml>=0.16.0->pykwalify->west) (0.2.8)

(.venv) C:\Users\KALYAN>west init zephyrproject
FATAL ERROR: already initialized in C:\Users\KALYAN, aborting.
  Hint: if you do not want a workspace there,
  remove this directory and re-run this command:

  C:\Users\KALYAN\.west

(.venv) C:\Users\KALYAN>cd zephyrproject

(.venv) C:\Users\KALYAN\zephyrproject>west update
=== updating acpica (modules/lib/acpica):
HEAD is now at da5f2721e Merge pull request #905 from najumon1980/acpi_timer
=== updating cmsis (modules/hal/cmsis):
HEAD is now at 4b96cbb doc: Update documentations for CMSIS 5.9.0
=== updating cmsis-dsp (modules/lib/cmsis-dsp):
HEAD is now at 6489e771 do not generate warnings when scalar functions are used with MVE
=== updating cmsis-nn (modules/lib/cmsis-nn):
HEAD is now at 0c8669d Merge pull request #1 from XenuIsWatching/zephyr
=== updating edtt (tools/edtt):
HEAD is now at 64e5105 edttt_bsim: Avoid infinite exception loop
=== updating fatfs (modules/fs/fatfs):
HEAD is now at 427159b zephyr: fix ffconf.h override use of STRINGIZE
=== updating hal_altera (modules/hal/altera):
HEAD is now at 0d225dd zephyr: update include paths to use <zephyr/...>
=== updating hal_ambiq (modules/hal/ambiq):
HEAD is now at ff4ca35 Stimer: remove disabling stmer's COMPARE interrupts in delta_set() api
=== updating hal_atmel (modules/hal/atmel):
HEAD is now at d6221e7 asf: CMakeLists.txt: Use new family prefix
=== updating hal_espressif (modules/hal/espressif):
HEAD is now at e705cc16c4 uart: add call to retrieve port_num by its register address
=== updating hal_ethos_u (modules/hal/ethos_u):
HEAD is now at 90ada2e Clarify README regarding CPU choice
=== updating hal_gigadevice (modules/hal/gigadevice):
HEAD is now at 2994b7d README.md: add pllmf and pack exception for gd32a50x
=== updating hal_infineon (modules/hal/infineon):
HEAD is now at b1a4723 hal: infineon: Update for hwmv2
=== updating hal_intel (modules/hal/intel):
HEAD is now at 7b4c256 iut: add -Werror to make all warnings into errors
=== updating hal_microchip (modules/hal/microchip):
HEAD is now at 68575aa CMakeLists: use MEC15XX for the series option
=== updating hal_nordic (modules/hal/nordic):
HEAD is now at 13ac55b nrfx: Update to version 3.4.0
=== updating hal_nuvoton (modules/hal/nuvoton):
HEAD is now at cf24f9e hal: nuvoton: support external dts or *.h of hal-M2L31X
=== updating hal_nxp (modules/hal/nxp):
HEAD is now at 3c4f74b4 devices: MCXN947: Update system file to remove SDK sepecific defines
=== updating hal_openisa (modules/hal/openisa):
HEAD is now at eabd530 fix-double-promotion in fsl_xcvr_trim
=== updating hal_quicklogic (modules/hal/quicklogic):
HEAD is now at b3a66fe Remove references to FPGA IP core drivers
=== updating hal_renesas (modules/hal/renesas):
[... 144 lines cut ...]
Requirement already satisfied: cryptography>=2.4.2 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from imgtool>=2.0.0->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-extras.txt (line 22)) (42.0.5)
Requirement already satisfied: grpcio>=1.62.1 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from grpcio-tools>=1.47.0->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-extras.txt (line 25)) (1.62.1)
Requirement already satisfied: pynacl>=1.4.0 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from PyGithub->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-extras.txt (line 29)) (1.5.0)
Requirement already satisfied: pyjwt>=2.4.0 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from pyjwt[crypto]>=2.4.0->PyGithub->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-extras.txt (line 29)) (2.8.0)
Requirement already satisfied: Deprecated in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from PyGithub->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-extras.txt (line 29)) (1.2.14)
[... cut ...]
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from ruamel.yaml>=0.16.0->pykwalify->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-base.txt (line 14)) (0.2.8)
Requirement already satisfied: pycparser in c:\users\kalyan\zephyrproject\.venv\lib\site-packages (from cffi->cmsis-pack-manager<1.0,>=0.5.2->pyocd>=0.35.0->-r \Users\KALYAN\zephyrproject\zephyr\scripts\requirements-run-test.txt (line 6)) (2.21)

followed setup instructions .All Requirements are satisfied but still facing an issue,you can see it above..

kalyanboddula, 2024-04-25:
(.venv) C:\Users\KALYAN\zephyrproject\zephyr>west flash
-- west flash: rebuilding
[1/1] C:\Windows\system32\cmd.exe /C "cd /D C:\Users\KALYA...Users/KALYAN/zephyrproject/zephyr/build/zephyr/zephyr.elf"
-- west flash: using runner esp32
-- runners.esp32: reset after flashing requested
-- runners.esp32: Flashing esp32 chip on None (921600bps)
esptool.py v4.5
Found 5 serial ports
Serial port COM9
Connecting...
COM9 failed to connect: Write timeout
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:33:07:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00033fff...
Flash params set to 0x0220
Wrote 32768 bytes at 0x00001000 in 0.7 seconds (383.4 kbit/s)...
Hash of data verified.
Wrote 16384 bytes at 0x00008000 in 0.3 seconds (489.4 kbit/s)...
Hash of data verified.
Wrote 147456 bytes at 0x00010000 in 2.1 seconds (571.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

(.venv) C:\Users\KALYAN\zephyrproject\zephyr>west espressif monitor
Serial port COM9
Connecting...
COM9 failed to connect: Write timeout
Serial port COM6
Connecting............
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Traceback (most recent call last):
  File "C:\Users\KALYAN\zephyrproject\zephyr\..\modules\hal\espressif\tools\idf_monitor.py", line 34, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
Traceback (most recent call last):
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 165, in do_run
    self.monitor(module_path, args)
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 184, in monitor
    cmd_exec(("python.exe", monitor_path, "-p", esp_port,
  File "C:\Users\KALYAN\zephyrproject\modules\hal\espressif\west\tools.py", line 40, in cmd_exec
    return subprocess.check_call(cmd, cwd=cwd, shell=shell)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '('python.exe', WindowsPath('C:/Users/KALYAN/zephyrproject/zephyr/../modules/hal/espressif/tools/idf_monitor.py'), '-p', 'COM6', '-b', '115200', WindowsPath('C:/Users/KALYAN/zephyrproject/zephyr/build/zephyr/zephyr.elf'), '--eol', 'CRLF')' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\KALYAN\zephyrproject\.venv\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 244, in run
    self.run_command(argv, early_args)
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 543, in run_command
    f'{quote_sh_list(cpe.cmd)}', fatal=True)
       ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\util.py", line 42, in quote_sh_list
    return ' '.join(shlex.quote(s) for s in cmd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KALYAN\zephyrproject\.venv\Lib\site-packages\west\util.py", line 42, in <genexpr>
    return ' '.join(shlex.quote(s) for s in cmd)
                    ^^^^^^^^^^^^^^
  File "C:\Python311\Lib\shlex.py", line 329, in quote
    if _find_unsafe(s) is None:
       ^^^^^^^^^^^^^^^
TypeError: expected string or bytes-like object, got 'WindowsPath'

Thank you @nordicjm for your reply,Tried again after following the setup instructions but still facing the problem.

nordicjm, 2024-04-25:
The error doesn't really make sense because serial is pyserial.

sylvioalves, 2024-04-25:
@kalyanboddula, can you try applying this change in your environment? The error might be Windows related: https://github.com/zephyrproject-rtos/hal_espressif/pull/277

kalyanboddula, 2024-04-26:
@sylvioalves i tried applying the changes you have suggested in my environment and changed by window path to string by doing changes in my 

west/tools.py

These are the changes I have done

```
def cmd_exec(cmd, cwd=None, shell=True):
    return subprocess.check_call(str(cmd), cwd=str(cwd), shell=shell)
```

By changing these windowpath to string i am still getting this error,I am not understanding the issue of it showing the error like this:

(.venv) C:\Users\KALYAN\zephyrproject\zephyr>west espressif monitor
Serial port COM9
Connecting...
COM9 failed to connect: Write timeout
Serial port COM6
Connecting..........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
'-p' was unexpected at this time.
FATAL ERROR: command exited with status 1: '(' ''"'"'' p y t h o n . e x e ''"'"'' , ' ' W i n d o w s P a t h '(' ''"'"'' C : / U s e r s / K A L Y A N / z e p h y r p r o j e c t / z e p h y r / . . / m o d u l e s / h a l / e s p r e s s i f / t o o l s / i d f _ m o n i t o r . p y ''"'"'' ')' , ' ' ''"'"'' - p ''"'"'' , ' ' ''"'"'' C O M 6 ''"'"'' , ' ' ''"'"'' - b ''"'"'' , ' ' ''"'"'' 1 1 5 2 0 0 ''"'"'' , ' ' W i n d o w s P a t h '(' ''"'"'' C : / U s e r s / K A L Y A N / z e p h y r p r o j e c t / z e p h y r / b u i l d / z e p h y r / z e p h y r . e l f ''"'"'' ')' , ' ' ''"'"'' - - e o l ''"'"'' , ' ' ''"'"'' C R L F ''"'"'' ')'

ajf58, 2024-04-29:
It looks like you've made some progress by taking the changes in the PR @kalyanboddula , but now have a slightly different issue. Is that correct?

sylvioalves, 2024-07-10:
Re-tested and confirmed it is working. Open again if there is new issue.

closed 2024-07-10 by sylvioalves