zephyrproject-rtos/zephyr #71932: TBS: interface function parameter to specify targeted TBS/GTBS instance

lukas-fwdev, 2024-04-25:
**Is your enhancement proposal related to a problem? Please describe.**
In TBS bluetooth specification it is stated that: "GTBS and each individual TBS have their own Call Index number space. ..."
The current implementation has only a global call index space. 
Also the interface functions are only parametrized with the call index and it should be possible to specify also the service instance to be addressed. 

**Describe the solution you'd like**
Every TBS instance and GTBS have their own call index number space.
Interface functions which don't already have the bearer_index parameter will take this additionally and then perform their functionality on the requested service instance (and the specified call).

**Describe alternatives you've considered**
An alternative whould be to expose the existing static functions inst_index, inst_lookup_index, lookup_call_in_inst, lookup_call, ... to be able to deal with the instances and call indexes in custom implementations. But it will make things more complicated and less according to specification I think.

**Additional context**
Could be done maybe together with #41738

Thalley, 2024-07-17:
While it is true that 

> GTBS and each individual TBS have their own Call Index number space. There is no requirement that the Call Index that is exposed by TBS matches the Call Index that is exposed by GTBS (even if the same call is exposed by TBS and GTBS).

is stated in the spec, it is up to implementation to determine how to handle this. 
We have, for simplicity, a shared call index space for all bearers. This means that we always have a 1:1 between GTBS call index and a TBS call index (unless the call exists only on the GTBS bearer). 
This has the advantage that we can also lookup a specific instance based on the call index. 

The only time where this won't work, is if the device wants to support more than 254 calls in total, which I don't think is going to be an issue for Zephyr. 

I guess what this PR is suggesting is that for any API that today takes a call index like 
```c
int bt_tbs_accept(uint8_t call_index);
```
also gets a TBS instance identifier. 

@lukas-fwdev given the above reasoning for the design, do you still request this enhancement? Given each TBS instance their own number space adds additional code and complexity with the only advantage that we can in theory support 200 calls on bearer 1 and another 200 calls on bearer 2, as today we are limited to 254 calls across all bearers.

lukas-fwdev, 2024-07-22:
Hi @Thalley 
I understand the simplification and I think it is a good and practical solution.
And yes, you are right. To have the TBS instance as second parameter would be very helpful in my eyes.

Thalley, 2024-07-29:
Just to be sure: You still request the addition of the TBS bearer to the callbacks and functions, even if the call ID is unique across all bearers?

lukas-fwdev, 2024-08-06:
I still think it is valuable to have the bearer in the callbacks but it is not a blocking request as the call IDs are unique across the bearers.

Thalley, 2024-08-13:
> I still think it is valuable to have the bearer in the callbacks but it is not a blocking request as the call IDs are unique across the bearers.

Will keep this open then :) I am currently refactoring the CCP/TBS API, so might add it as part of that