zephyrproject-rtos/zephyr #71970: Unable to build lz4 compression sample

afeinman-snap, 2024-04-26:
**Describe the bug**
Building lz4 sample seems broken after v3.4.0 regardless of platform
Build complains of missing lz4.h
```
/home/user/zephyrproject/zephyr/samples/compression/lz4/src/main.c:11:10: fatal error: lz4.h: No such file or directory
   11 | #include "lz4.h"
      |          ^~~~~~~
compilation terminated.
```
Git bisect point to the commit f30f08a3ce75ae7d7384a7608b7e3dcef804661a where submodules were made optional

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr/samples/compression/lz4
2. west build -b qemu_x86 --pristine
3. See error

Additionally the cmake phase prints:
```
warning: LZ4 (defined at modules/lz4/Kconfig:7) was assigned the value 'y' but got the value 'n'.
Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_LZ4 and/or look up LZ4 in the
```
This warning is pretty much useless. What is dependency 0? How to deal with it?
In menuconfig interface it is impossible to enable LZ4 option either

**Expected behavior**
If the optional module needs to be somehow enabled, a message should be displayed in kconfig phase (probably by config script)

**Impact**
So far I could not find a way to build this sample in v3.5.0+. Nothing that I could see in the documentation either.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.5.1
 - v3.5.0

kartben, 2024-04-26:
LZ4 is indeed an optional module now, you need to do the following to enable it:

```
west config manifest.project-filter -- +lz4
west update
```

Would you be able to please send a PR to update the README along the same lines as e.g. https://docs.zephyrproject.org/latest/samples/modules/tflite-micro/hello_world/README.html#building-and-running ?

afeinman-snap, 2024-04-26:
> Would you be able to please send a PR to update the README

Yes, will do. Thank you for the hint

parthitce, 2024-08-08:
> This method of installing the optional module did not work for me. After I rerun `west update` the module is not pulled and I still cannot run the example. I could instead download the module by passing the project filter as an argument to `west update` Regardless, even if the module is downloaded, I still get the Kconfig error and the module is not added to the build. Am I missing some steps?

lz4 is moved to optional module, you can try pulling all the optional modules using
```
west config manifest.group-filter -- +optional
west update
```

parthitce, 2024-08-13:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/76184.

closed 2025-02-28 by kartben