zephyrproject-rtos/zephyr #71983: tests: driver: drivers.uart.async_api.lpuart: mimxrt1010_evk: hardfault

hakehuang, 2024-04-26:
This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
tests/drivers/uart/uart_async_api meet hardfault

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1010_evk
 - What have you tried to diagnose or workaround this issue?
 - NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:

west twister -p mimxrt1010_evk -T tests/drivers/uart/uart_async_api

**Expected behavior**

PASS

**Impact**

DMA issue iwth lpuart in this platform. (other RT series is ok including RT1015)

**Logs and console output**

```
*** Booting Zephyr OS build v3.6.0-3046-g78150c8e1b2c ***
Running TESTSUITE uart_async_chain_read
===================================================================
E: No buffers to release from RX DMA!
E: No buffers to release from RX DMA!
START - test_chained_read
E: 0x60011a50 used before initialization
E: syscall z_vrfy_uart_rx_enable failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x3f750880
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20002840 (test_chained_read)
E: Halting system

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.6.0-3046-g78150c8e1b2c

hakehuang, 2024-04-26:
this is a long missing issue, I misjudge rt1015 and rt1010 use the same ip, and rt1015 is ok which misleading me as my board settings issue. but after some analysis, this is a dma issue.

decsny, 2024-06-04:
I confirmed that it fails on all the other RT10xx boards as well.

dleach02, 2024-08-15:
ping @EmilioCBen

Dat-NguyenDuy, 2024-09-25:
Hi @hakehuang, i have fixed an DMA issue on imxrt 1010 recently via https://github.com/zephyrproject-rtos/zephyr/pull/77140.
Could you pls run the test against latest main? I expect that it should work

hakehuang, 2024-09-25:
> Could you pls run the test against latest main? I expect that it should work

@Dat-NguyenDuy I tested it is PASS on all RT10xx series.

closed 2024-09-25 by hakehuang