zephyrproject-rtos/zephyr #72017: Bluetooth: (conn.c) ACL buffers can "steal" ISO TX contexts

jori-nordic, 2024-04-26:
Reporting just so I don't forget about it.

Both `iso_tx[]` and `conn_tx[]` arrays end up in the `free_tx` fifo.
This fifo is then used indiscriminately by `conn_tx_alloc()` to allocate a TX context for a given buffer. That buffer can be for an ACL or ISO link.

This means an ACL link can starve an ISO link.

# Env

Zephyr @ https://github.com/zephyrproject-rtos/zephyr/blob/78150c8e1b2c07537ba35a05d9428983f1968f99/subsys/bluetooth/host/conn.c#L427-L449

Thalley, 2024-07-03:
I guess the solution here is to have individual pools for ACL and ISO? 
What about BREDR? L2CAP?

jori-nordic, 2024-07-03:
we already have the individual pools. I guess the solution is pretty simple, also have individual fifos, by adding a `conn-type` param to `conn_tx_alloc()`. I have not dived too deep on this one yet.

Thalley, 2024-07-03:
> we already have the individual pools. I guess the solution is pretty simple, also have individual fifos, by adding a `conn-type` param to `conn_tx_alloc()`. I have not dived too deep on this one yet.

Right, I'm sure that's also what I meant :P 

But I still assume that the issue between ISO and ACL also exist for ACL and SCO/BR, although much less apparent?

jori-nordic, 2024-07-04:
It's possible, yes. Although I'm not sure how to test SCO.

PR #89258 (2025-05-12): Bluetooth: Host: ACL/ISO buffers and bt_conn_tx objects mismatch
This PR addresses an issue of mismatch between ACL/ISO buffers reported by Controller and `bt_conn_tx` objects allocated by Host.

After zephyrproject-rtos#72090, each packet to be sent (whether ACL or ISO data) has a corresponding `bt_conn_tx` object, regardless of whether a callback is used.

This means that:
1. number of packets Host can send to Controller is limited by the smaller of two values: ACL/ISO packets Controller can receive, and the number of `bt_conn_tx` objects allocated by Host.
2. `CONFIG_BT_CONN_TX_MAX` is supe [... cut]
closed 2025-05-12 by henrikbrixandersen