zephyrproject-rtos/zephyr #72018: Serial: Bluetooth: Increase Throughput by tracking run-time ATT MTU Size

ubieda, 2024-04-26:
**Is your enhancement proposal related to a problem? Please describe.**
The UART over Bluetooth LE (NUS) currently fragments TX data in chunks of 20-bytes per Bluetooth notification. This is done since the maximum payload we can put through a single packet, assuming minimum ATT MTU size (MTU - 3). This, however, represents a throughput bottleneck and limits its usability (Simple but limited).

**Describe the solution you'd like**
Split data in chunks of (min_current_mtu - 3), where the `min_current_mtu` is the minimum MTU of all the peers currently connected. 
 e.g:
``` c
		chunk_size = get_max_payload_packet_size();
		len = ring_buf_get_claim(dev_data->uart.tx_ringbuf, &data, chunk_size);
		if (len > 0) {
			err = bt_nus_inst_send(NULL, dev_data->bt.inst, data, len);
			if (err) {
				LOG_ERR("Failed to send data over BT: %d", err);
			}
		}
```

**Describe alternatives you've considered**
No additional alternatives.

**Additional context**
To achieve so, `uart_bt.c` needs to have run-time awareness of all connected peers and their MTU size at all times.

ubieda, 2024-04-26:
@jori-nordic @alwa-nordic @jhedberg LMK if you have any inputs on this!

jhedberg, 2024-04-27:
@ubieda sounds like a good potential improvement to me

PR #77178 (2024-09-18): serial: bluetooth: Maximize notification payload size based on ATT MTU
### Description
This PR adds the UART BT driver the ability to maximize the payload notification size by keeping track of run-time ATT MTU of all connected peers; thus making it possible to optimize the data-throughput sent over Bluetooth.

Note that this assumes the Bluetooth connection is properly configured (with the corresponding connection parameters) to optimize performance. This is application-specific and shall be examined on each scenario for optimal performance.

Fixes: #72018

### Testing
#### Test - Verify MTU-s [... cut]
closed 2024-09-18 by nashif