zephyrproject-rtos/zephyr #72019: Bluetooth: UART BT using System Workqueue no longer compatible with Blocking BT TX calls

ubieda, 2024-04-26:
**Describe the bug**
UART Bluetooth driver currently sends messages by scheduling a delayed work-item in the system work-queue. This no longer works well with the Bluetooth stack, per https://github.com/zephyrproject-rtos/zephyr/pull/71666. As a result, samples sending more than a handful of bytes per second (e.g: Logging over UART BT, Shell) will no longer work properly.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
- Build Shell over Bluetooth sample: `west build -b nrf52840dk/nrf52840 samples/subsys/shell/shell_module -- -DOVERLAY_CONFIG='overlay-bt.conf' -DDTC_OVERLAY_FILE='bt.overlay'`
- Flash: `west flash --erase`
- Connect to the device using a NUS client (e.g: https://github.com/ubieda/etales-zephyr/blob/master/scripts/bluetoothcli)
- See logging errors pop-up endlessly.

**Expected behavior**
Shell output is serialized properly, no endless logging errors show up.

**Impact**
UART over Bluetooth LE is somewhat broken (not working with subsystems such as Logging, nor Shell).

**Logs and console output**
``` console
[00:00:07.101,684] <wrn> bt_conn: Unable to allocate buffer within timeout
[00:00:07.101,715] <err> bt_att: Unable to allocate buffer for op 0x1b
[00:00:07.101,715] <wrn> bt_gatt: No buffer available to send notification
[00:00:07.101,745] <wrn> uart_nus: Failed to send data over BT: -12
[00:00:07.101,928] <wrn> bt_conn: Unable to allocate buffer within timeout
[00:00:07.101,928] <err> bt_att: Unable to allocate buffer for op 0x1b
[00:00:07.101,928] <wrn> bt_gatt: No buffer available to send notification
[00:00:07.101,959] <wrn> uart_nus: Failed to send data over BT: -12
[00:00:07.102,050] <wrn> bt_conn: Unable to allocate buffer within timeout
[00:00:07.102,081] <err> bt_att: Unable to allocate buffer for op 0x1b
[00:00:07.102,081] <wrn> bt_gatt: No buffer available to send notification
[00:00:07.102,111] <wrn> uart_nus: Failed to send data over BT: -12
```

**Suggested Solution**
Have all UART NUS instances in `uart_bt.c` use a separate work-queue (instead of the system work-queue), which would allow blocking BT TX calls, and therefore restore expected behavior.

ubieda, 2024-04-26:
@jori-nordic @alwa-nordic LMK if you have inputs on this!

alwa-nordic, 2024-04-29:
A separate work queue for NUS is fine. Otherwise:

Quick fix: Use `bt_gatt_notify_cb` to flow control how fast notifications are fed into the host. If the call to `bt_gatt_notify_cb` fails while within the flow control limits and there are no outstanding notifications, schedule the next `bt_gatt_notify_cb` on a timeout.

Intermediate fix: To avoid the timeout-and-try-again, we need API to listen for available ATT buffers.

Ultimate fix: Of course, the new GATT API we are discussing could outsource this problem by borrowing memory from the calling application instead of internally allocating.

> The next one is, there's this thing we call [long_wq](https://github.com/jori-nordic/zephyr/blob/3a7d462daeecbf9cfc78c3ae9682b26d753d57a7/subsys/bluetooth/host/long_wq.h#L1) that we use for long-running tasks in the stack. It is pre-emptible so no worries of blocking other users there. @alwa-nordic maybe we could make it public?

I am against doing this. If we really want to share a work queue with the app, then we should do it the other way around: The app should allocate it and give us a reference. That way, we are free to remove the long work queue in the future.

PR #72121 (2024-05-13): drivers: serial: bt: Use dedicated Work-queue
### Description
Add dedicated NUS Work-queue since the System Work-queue no longer supports Blocking BT TX APIs (and for a good reason).
The default Priority is set to Main Thread and Stack-size 1KiB, but can be adjusted through Kconfig configuration.

Fixes #72019.
closed 2024-05-13 by nashif