zephyrproject-rtos/zephyr #72035: net: net_context: `zsock_connect` can error out due to duplicate ports

JordanYates, 2024-04-27:
**Describe the bug**

`check_used_port` can return `false` when the port is actually in use, leading to `find_available_port` returning an unavailable port.
This can be reliably triggered when `sys_rand32_get` is derived from the system clock, but can occur with a 1/2**16 chance in normal operation.

The initial `net_context_get` from `zsock_socket` assigns a random port to the net context:
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L502-L503

`net_context_connect` from `zsock_connect` then generates another random port inside `bind_default`:
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L656-L658
A duplicate port number at this stage is not detected because the context `sin_addr` pointer is currently NULL, which skips further checks here:
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L247

Inside `net_context_bind` (called from `bind_default`), the `sin_addr` pointer is finally set (using the first generated port):
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L884
And the second generated port is then checked again:
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L893-L898

At this point `check_used_port` returns an error, and `sock_connect` errors out.

**Expected behavior**
It should not be possible for duplicate port numbers to be generated inside `zsock_connect` that cause an error without the user being able to intervene.

**Environment (please complete the following information):**
 - Zephyr v3.6 (Cannot see any meaningful changes on main)

JordanYates, 2024-04-27:
I'm not sure whether the bug is in the implementation of `check_used_port` (i.e. the NULL check is not sufficient), or in one of the calling functions (context `sin_addr` pointer should be set earlier)

jukkar, 2024-05-10:
@JordanYates I am trying to replicate this but have so far failed. Do you have a setup / test app I could use?

What is baffling is that here in `check_used_port()` before the NULL check you mentioned,
https://github.com/zephyrproject-rtos/zephyr/blob/a495fd9475e324b453a02b543d917e0816cfce3a/subsys/net/ip/net_context.c#L173-L175
we check whether there is a local port (that is set by `net_context_get()`) already set. So I just wonder why we would get the duplicate port selected.

jukkar, 2024-05-10:
Are you using IPv4-to-IPv6 mapping  `CONFIG_NET_IPV4_MAPPING_TO_IPV6` as that could explain this issue?

JordanYates, 2024-05-10:
> @JordanYates I am trying to replicate this but have so far failed. Do you have a setup / test app I could use?
>
> What is baffling is that here in `check_used_port()` before the NULL check you [...]

I can reliably trigger this in my local code by changing this line:
https://github.com/zephyrproject-rtos/zephyr/blob/c059405da57b8b3b7bd22295c7f66ea3541cc55f/subsys/net/ip/net_context.c#L315
to this: 
```
local_port = k_uptime_get() / 2 | 0x8000;
```
Which in theory should just make the function loop for ~2 msec.

> Are you using IPv4-to-IPv6 mapping CONFIG_NET_IPV4_MAPPING_TO_IPV6 as that could explain this issue?

This option is not enabled (`CONFIG_IPV6` as a whole is disabled). I will try and generate a reproduction in NCS using a nRF7002DK.

PR #72595 (2024-05-14): net: context: Do not check our own ports
There is no need to check our own context when going through the used ports in the system. This prevents error when binding in some corner cases.

Fixes #72035
commit 1d7970be (2024-05-14): net: context: Do not check our own ports

There is no need to check our own context when going through
the used ports in the system. This prevents error when binding
in some corner cases.

Fixes #72035
closed 2024-05-14 by nashif