zephyrproject-rtos/zephyr #72045: ws2812-spi not sending correct data on sam0 platform

PeeJay, 2024-04-28:
**Describe the bug**
The ws2812-spi driver appears to have issues on the sam0 platform. I am using a custom board that is compatible with the Adafruit itsybitsy M4 express, and have connected a WS2812b strip to PA21, running the led_strip sample.
The issue is that the first LED doesn't do anything, but the remaining 5 do.

I've checked the timings with a logic analyser and they are correct, however I notice that the spi bus will return to high when idle. When using the FastLed library with an Arduino sketch the data line was low at idle. 
In the overlay for the ESP32 there is this...
```
* Workaround to support WS2812 driver */
	line-idle-low;
```
...in the SPI dts section, but no such option exists for sam0.

Here is the relevant part from the DTS
```
&sercom3 {
	status = "okay";
	compatible = "atmel,sam0-spi";
	dipo = <2>;
	dopo = <3>;

	#address-cells = <1>;
	#size-cells = <0>;

	pinctrl-0 = <&sercom3_spi_default>;
	pinctrl-names = "default";

	status_rgb_led: ws2812@0 {
		compatible = "worldsemi,ws2812-spi";
		reg = <0>;
		spi-max-frequency = <WS2812C_SPI_FREQ>;

		chain-length = <6>;
		color-mapping =	<LED_COLOR_ID_GREEN>,
						<LED_COLOR_ID_RED>,
						<LED_COLOR_ID_BLUE>;
		spi-one-frame = <WS2812C_ONE_FRAME>;
		spi-zero-frame = <WS2812C_ZERO_FRAME>;
		
	};
};
```

PeeJay, 2024-04-29:
Update: After experimenting with the SPI port there doesn't seem to be any way to get it to return to LOW when the transmission is finished. Switching the pin to GPIO has around a 200ns delay both ways so that's not an option either. I think it's simply not possible to use the sam0 SPI port. sam0-i2s does not exist (I wonder how similar the hardware is to sam-spi?) and ws2812-gpio is NRF only.

closed 2024-07-14 by github-actions[bot]

wookhyunshin, 2025-07-21:
I faced the exactly same issue, and I found temporary workaround. I use SAME51 and ws2812-spi.

I added some idle data at the beginning and end of the actual data transmission.
exsting `ws2812_reset_delay` has no effect, it may be removed.

My code is not yet ready to be published but this code works.

drivers\led_strip\ws2812_spi.c
`
static int ws2812_strip_update_rgb(const struct device *dev,
				   struct led_rgb *pixels,
				   size_t num_pixels)
{
	const struct ws2812_spi_cfg *cfg = dev_cfg(dev);
	const uint8_t one = cfg->one_frame, zero = cfg->zero_frame;
	struct spi_buf buf = {
		.buf = cfg->px_buf,
		.len = (cfg->length * 8 * cfg->num_colors + 2*WS2812_IDLE_DATA_SIZE),
	};
	const struct spi_buf_set tx = {
		.buffers = &buf,
		.count = 1
	};
	uint8_t *px_buf = cfg->px_buf;
	size_t i;
	int rc;

	/* Add initial Treset code time
	*/
	memset(px_buf, 0, WS2812_IDLE_DATA_SIZE);
	px_buf += WS2812_IDLE_DATA_SIZE;

	/*
	 * Convert pixel data into SPI frames. Each frame has pixel data
	 * in color mapping on-wire format (e.g. GRB, GRBW, RGB, etc).
	 */
	for (i = 0; i < num_pixels; i++) {
		uint8_t j;

		for (j = 0; j < cfg->num_colors; j++) {
			uint8_t pixel;

			switch (cfg->color_mapping[j]) {
			/* White channel is not supported by LED strip API. */
			case LED_COLOR_ID_WHITE:
				pixel = 0;
				break;
			case LED_COLOR_ID_RED:
				pixel = pixels[i].r;
				break;
			case LED_COLOR_ID_GREEN:
				pixel = pixels[i].g;
				break;
			case LED_COLOR_ID_BLUE:
				pixel = pixels[i].b;
				break;
			default:
				return -EINVAL;
			}
			ws2812_spi_ser(px_buf, pixel, one, zero);
			px_buf += 8;
		}
	}

	/*
	 * Fill the rest of the buffer with idle data.
	 * The WS2812 strip requires a certain amount of idle data (>200us)
	 * at the end of the frame to latch the pixel data.
	 * Giving 256 bytes of idle data is enough for all WS2812.
	 */
	memset(px_buf, 0, WS2812_IDLE_DATA_SIZE);
	px_buf += WS2812_IDLE_DATA_SIZE;

	/*
	 * Display the pixel data.
	 */
	rc = spi_write_dt(&cfg->bus, &tx);
	// ws2812_reset_delay(cfg->reset_delay);

	return rc;
}
`