zephyrproject-rtos/zephyr #72051: tests: test_mcuboot: minxrt1015_evk: mcu boot test failure

hakehuang, 2024-04-28:
**Describe the bug**
mimxrt1015_evk test_mcuboot failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1015_evk
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
9d7a3fb647e8ab7472ab67bb00f52e1cd918966f is the first bad commit
commit 9d7a3fb647e8ab7472ab67bb00f52e1cd918966f
Author: Daniel DeGrasse <daniel.degrasse@nxp.com>
Date:   Fri Aug 18 20:29:54 2023 +0000

    drivers: flash: flash_flexspi_nor: support SFDP probe

    Support SFDP probe in flexspi nor driver. This probe will allow the
    flash driver to dynamically configure quad spi flashes for 1-4-4 mode,
    expanding the flash chips supported with this driver.

    The following data is read from the SFDP header:
    - quad enable method
    - fast read command (1-4-4 is maximum supported)

    Fixes #55379

    Signed-off-by: Daniel DeGrasse <daniel.degrasse@nxp.com>

 drivers/flash/CMakeLists.txt           |   1 +
 drivers/flash/Kconfig.mcux             |   1 +
 drivers/flash/flash_mcux_flexspi_nor.c | 796 +++++++++++++++++++++++++++------
 drivers/memc/memc_mcux_flexspi.h       |   2 +
 4 files changed, 675 insertions(+), 125 deletions(-)

```

**To Reproduce**
Steps to reproduce the behavior:

```
scripts/twister --device-testing --hardware-map /home/ubuntu/nxp/mimxrt1015_evk/map.yaml --west-flash  -vv -T tests/boot/test_mcuboot/
```

**Expected behavior**
Test PASS

**Impact**

mcuboot

**Logs and console output**
no log output when this happen

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  v3.6.0-3128-gc059405da57

butok, 2024-04-29:
Hi @Hake
You are pointing to the August commit.
But the right mimxrt1015_evk/mimxrt1010_evk/mimxrt1040_evk McuBoot support was added & tested only in December 2023.
Be sure:
- Do mass erase before the test flashing, without it the test is failed.
- Do not use LinkServer runner (hex support is not merged yet).
- Do HW reset, very often no serial output without it.

butok, 2024-04-29:
This test requires the .hex support, otherwise it flashes an application image to a wrong place. 
So JLINK or Pyocd runners should be used.

butok, 2024-04-29:
@hakehuang @danieldegrasse 
Just checked on mimxrt1015_evk and the issue is real, there is no output on console!

dleach02, 2024-05-06:
@butok, I added you as a reviewer to @danieldegrasse PR

PR #72200 (2024-05-16): drivers: flash: flash_mcux_flexspi_nor: better handle legacy SFDP tables
Implement more robust handling for legacy SFDP tables, which may not implement some of the JEDEC defined DWORDS for SFDP data. Instead of failing to probe/initialize the flash when these DWORDS are not defined in the basic flash parameter table, revert to sane defaults for SPI flash.

Fixes #72051

This change was verified on the the following EVKs:
- RT1015 EVK (AT25SF128A QSPI flash)
- RT1060 EVKB (IS25WP064 QSPI flash)
- RT1040 EVK (W25Q64JV QSPI flash)
commit 1de7574d (2024-05-16): drivers: flash: flash_mcux_flexspi_nor: better handle legacy SFDP tables

Implement more robust handling for legacy SFDP tables, which may not
implement some of the JEDEC defined DWORDS for SFDP data. Instead of
failing to probe/initialize the flash when these DWORDS are not defined
in the basic flash parameter table, revert to sane defaults for SPI
flash.

Fixes #72051
closed 2024-05-16 by nashif