zephyrproject-rtos/zephyr #72068: `check_init_priorities` fails for `tests/kernel/device` with `native_sim`

pdgendt, 2024-04-29:
**Describe the bug**
Running `scripts/build/check_init_priorities.py` for `tests/kernel/device` fails when built for `native_sim` board.

**To Reproduce**
```shell
$ west build -b native_sim tests/kernel/device
$ ./scripts/build/check_init_priorities.py -f build/zephyr/zephyr.elf
```

**Expected behavior**
The script validates the device initialization priorities without raising an error.

**Impact**
Annoyance

**Logs and console output**
```
Traceback (most recent call last):
  File "/home/pdgendt/dev/zephyr-main/zephyr/./scripts/build/check_init_priorities.py", line 374, in <module>
    sys.exit(main(sys.argv[1:]))
             ^^^^^^^^^^^^^^^^^^
  File "/home/pdgendt/dev/zephyr-main/zephyr/./scripts/build/check_init_priorities.py", line 359, in main
    validator = Validator(args.elf_file, args.edt_pickle, log)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pdgendt/dev/zephyr-main/zephyr/./scripts/build/check_init_priorities.py", line 245, in __init__
    self._obj = ZephyrInitLevels(elf_file_path)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pdgendt/dev/zephyr-main/zephyr/./scripts/build/check_init_priorities.py", line 110, in __init__
    self._process_initlevels()
  File "/home/pdgendt/dev/zephyr-main/zephyr/./scripts/build/check_init_priorities.py", line 206, in _process_initlevels
    raise ValueError(f"no symbol at addr {addr:08x}")
ValueError: no symbol at addr 00000017
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04 (WSL)
 - Toolchain: Zephyr SDK
 - 1ad4b5c6170

pdgendt, 2024-04-29:
CC @fabiobaltieri

fabiobaltieri, 2024-04-29:
@pdgendt you need to use `zephyr.exe` for `native_sim`, something odd with how linking works there:

```console
~/zephyrproject/zephyr$ ./scripts/build/check_init_priorities.py -v -f build/zephyr/zephyr.elf 
ValueError: no symbol at addr 00000017
$ ./scripts/build/check_init_priorities.py -v -f build/zephyr/zephyr.exe 
INFO: check_init_priorities: build/zephyr/zephyr.exe
INFO: /fakedomain_0 <my_driver_sub_pri_0_init> POST_KERNEL+8 > /fakedomain_2 <my_driver_sub_pri_2_init> POST_KERNEL+7
INFO: /fakedomain_1 <my_driver_sub_pri_1_init> POST_KERNEL+9 > /fakedomain_0 <my_driver_sub_pri_0_init> POST_KERNEL+8
```

The buildsystem does that automatically (in fact it would fail the build if that was not working)

closed 2024-04-29 by pdgendt