zephyrproject-rtos/zephyr #72128: RISCV: fpu_sharling mode

SJC0719, 2024-04-30:
In the isr.S file in the riscv directory, after the fpu_sharling function is enabled, the system reads the value of the csr register mcause to determine whether the current exception belongs to the fpu operation. In this file, the system directly compares the value of mcause with that of 2 to determine whether to enter the jump. However, the actual mcause value is only used to determine the low 7bit, other bits may exist other values, resulting in the program can not jump normally

`#if defined(CONFIG_FPU_SHARING)
	/* determine if FPU access was disabled */
	li t1, MSTATUS_FS
	and t1, t1, t2
	bnez t1, no_fp
	/* determine if this is an Illegal Instruction exception */
	csrr t2, mcause
	and t2,t2,SOC_MCAUSE_EXP_MASK                                 //The low exception number should be performed or operated on here
	li t1, 2		/* 2 = illegal instruction */
	bne t1, t2, no_fp
	/* determine if we trapped on an FP instruction. */
	csrr t2, mtval		/* get faulting instruction */`

aescolar, 2024-04-30:
@SJC0719 Please use the template when filling issues

fkokosinski, 2024-05-09:
Hey @npitre, could you perhaps take a look at this?

PR #72523 (2024-05-14): arch: riscv: apply CONFIG_RISCV_MCAUSE_EXCEPTION_MASK to FPU code
Some implementations use bits outside of the mcause mask for other
purpose.

Fixes #72128
closed 2024-05-14 by henrikbrixandersen