zephyrproject-rtos/zephyr #72137: net: coap_client: GET with observe

tautologyclub, 2024-04-30:
Looked through the code and got the distinct impression that observe isn't supported (I'd be chuffed if someone told me I was mistaken though). It seems like a low-to-mid complexity feature to implement so I may end up taking a crack at it but I figured I'd ping here first so see what's up @rlubos  @jukkar

kartben, 2024-04-30:
Have you looked at the CoAP client and server samples? They both demonstrate the observe functionality.

tautologyclub, 2024-04-30:
@kartben I have but this enhancement is regarding the actual `coap_client`

tautologyclub, 2024-04-30:
For reference, I tried modifying `handle_response()` so that it doesn't unset `request_ongoing` flag when the request is an observe:

```
fail:
	client->response_ready = false;
	if (ret != 0 || !coap_request_is_observe(&internal_req->request)) {
		internal_req->request_ongoing = false;
	}
	return ret;
```

This worked beautifully. But there's still the architectural issues remaining. Since `coap_client` wasn't written to be a infinity-recv:ing daemon, there's no API:s to shut it down/cancel requests and so forth. That request will stick around forever. Not sure what the cleanest "cancel" API would be tho...

tautologyclub, 2024-04-30:
Please feel free to rip this apart: https://github.com/zephyrproject-rtos/zephyr/pull/72159 @rlubos @jukkar

tautologyclub, 2024-05-07:
Well, the PR is merged so which should theoretically mean this issue is closabe. But the PR was severely incomplete and should really be augmented so let's keep this open :P

tautologyclub, 2024-10-23:
The incomplete PR has been augmented so this is actually supported

closed 2024-10-23 by tautologyclub