zephyrproject-rtos/zephyr #72190: ESP32C3 failes to build with CONFIG_RTC_CLK_SRC_EXT_CRYS=y

fgihl, 2024-05-01:
Using zephyr main, you get build error when enabling external 32K clock on esp32c3.
To enable external 32K clock I use the Kconfig
`CONFIG_RTC_CLK_SRC_EXT_CRYS=y`

Repo:
Fetch zephyr according to Getting started guide.

Modifiy hello world sample by adding the line
`CONFIG_RTC_CLK_SRC_EXT_CRYS=y`
to prj.conf

Build hello world using the line 
`west build -p -b esp32c3_devkitm samples/hello_world`

The following  build error now happens:
/home/fgihl/tmp/zephyr-validation/modules/hal/espressif/components/bootloader_support/src/bootloader_clock_init.c: In function 'bootloader_clock_configure':
/home/fgihl/tmp/zephyr-validation/modules/hal/espressif/components/bootloader_support/src/bootloader_clock_init.c:85:31: error: 'CONFIG_ESP_SYSTEM_RTC_EXT_XTAL_BOOTSTRAP_CYCLES' undeclared (first use in this function)
   85 |         rtc_clk_32k_bootstrap(CONFIG_ESP_SYSTEM_RTC_EXT_XTAL_BOOTSTRAP_CYCLES);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/fgihl/tmp/zephyr-validation/modules/hal/espressif/components/bootloader_support/src/bootloader_clock_init.c:85:31: note: each undeclared identifier is reported only once for each function it appears in
[126/198] Building C object zephyr/arch/arch/riscv/core/CMakeFiles/arch__riscv__core.dir/fatal.c.obj
ninja: build stopped: subcommand failed.

sylvioalves, 2024-05-05:
@fgihl, would you re-try using the PR linked here?

fgihl, 2024-05-06:
Thanks @sylvioalves .
Yes, this fixes the compile problem.

PR #72327 (2024-05-08): soc: espressif: add misssing external xtal kconfig entry
External XTAL usage is missing a Bootstrap Cycle configuration
in Kconfig, causing build to failure when CONFIG_RTC_CLK_SRC_EXT_CRYS
is selected.

Fixes #72190
commit 3f07d4b6 (2024-05-08): soc: espressif: add misssing external xtal kconfig entry

External XTAL usage is missing a Bootstrap Cycle configuration
in Kconfig, causing build to failure when CONFIG_RTC_CLK_SRC_EXT_CRYS
is selected.

Fixes #72190
closed 2024-05-08 by nashif