zephyrproject-rtos/zephyr #72196: west update: `ERROR: update failed for project hal_espressif`

jvermillard, 2024-05-01:
Today, on commit:
* 27fa6d9163 - (HEAD -> main, upstream/main, upstream/HEAD) samples: littlefs: add support for H3ULCB and Salvator XS M3 (3 hours ago) <Mykola Kvach>

west update produce some error logs:

**To Reproduce**
Steps to reproduce the behavior:
1. in your zephyr directory: git pull
2. west update

**Expected behavior**
no error

**Impact**
I don't know what is the exact impact

**Logs and console output**
```

=== updating acpica (modules/lib/acpica):
HEAD is now at da5f2721e Merge pull request #905 from najumon1980/acpi_timer
=== updating cmsis (modules/hal/cmsis):
HEAD is now at 4b96cbb doc: Update documentations for CMSIS 5.9.0
=== updating cmsis-dsp (modules/lib/cmsis-dsp):
HEAD is now at 6489e771 do not generate warnings when scalar functions are used with MVE
=== updating cmsis-nn (modules/lib/cmsis-nn):
HEAD is now at 0c8669d Merge pull request #1 from XenuIsWatching/zephyr
=== updating edtt (tools/edtt):
HEAD is now at 64e5105 edttt_bsim: Avoid infinite exception loop
=== updating fatfs (modules/fs/fatfs):
HEAD is now at 427159b zephyr: fix ffconf.h override use of STRINGIZE
=== updating hal_altera (modules/hal/altera):
HEAD is now at 0d225dd zephyr: update include paths to use <zephyr/...>
=== updating hal_ambiq (modules/hal/ambiq):
HEAD is now at ff4ca35 Stimer: remove disabling stmer's COMPARE interrupts in delta_set() api
=== updating hal_atmel (modules/hal/atmel):
HEAD is now at d6221e7 asf: CMakeLists.txt: Use new family prefix
=== updating hal_espressif (modules/hal/espressif):
--- hal_espressif: fetching, need revision e554857b07d54e2152c6f1a9b39da6d6c13fe7f7
=== updating hal_ethos_u (modules/hal/ethos_u):
HEAD is now at 8e2cf75 Support timeout for interrupt semaphore
=== updating hal_gigadevice (modules/hal/gigadevice):
HEAD is now at 2994b7d README.md: add pllmf and pack exception for gd32a50x
=== updating hal_infineon (modules/hal/infineon):
HEAD is now at b1a4723 hal: infineon: Update for hwmv2
=== updating hal_intel (modules/hal/intel):
HEAD is now at 7b4c256 iut: add -Werror to make all warnings into errors
=== updating hal_microchip (modules/hal/microchip):
HEAD is now at 68575aa CMakeLists: use MEC15XX for the series option
=== updating hal_nordic (modules/hal/nordic):
HEAD is now at bdef8b6 nrfx: hal: nrf_cache: Add nrf_cache_lineaddr_get
=== updating hal_nuvoton (modules/hal/nuvoton):
HEAD is now at ab342e6 hal: nuvoton: add NuMaker RTC controller driver
=== updating hal_nxp (modules/hal/nxp):
HEAD is now at 8c354a91 mcux: hal_nxp: Update from mcux-sdk to include MKE1XZ9
=== updating hal_openisa (modules/hal/openisa):
HEAD is now at eabd530 fix-double-promotion in fsl_xcvr_trim
=== updating hal_quicklogic (modules/hal/quicklogic):
HEAD is now at b3a66fe Remove references to FPGA IP core drivers
=== updating hal_renesas (modules/hal/renesas):
HEAD is now at e3560c7 hal: renesas: Add sleep code
=== updating hal_rpi_pico (modules/hal/rpi_pico):
HEAD is now at fba7162 Merge pull request #5 from soburi/zephyr_timer_modification
=== updating hal_silabs (modules/hal/silabs):
HEAD is now at 442d0fb gecko/cmake: replace 'SOC_GECKO_SERIESx' with 'SOC_FAMILY_SILABS_Sx'
=== updating hal_st (modules/hal/st):
HEAD is now at b77157f drivers: sensor: st: Move to vendor subdirectory
=== updating hal_stm32 (modules/hal/stm32):
HEAD is now at 60c9634f dts: New batch of pinctrl.dtsi files including SWJ signals
=== updating hal_telink (modules/hal/telink):
HEAD is now at 38573af B91 ADC driver support
=== updating hal_ti (modules/hal/ti):
HEAD is now at b85f86e simplelink: slnetsock: fix sys/time.h include when sockets are enabled
=== updating hal_wurthelektronik (modules/hal/wurthelektronik):
HEAD is now at e5bcb2e Update WE sensors SDK to version 2.5.0
=== updating hal_xtensa (modules/hal/xtensa):
HEAD is now at a2d6585 zephyr: rename SoC overlay for NXP i.MX8ULP
=== updating hostap (modules/lib/hostap):
HEAD is now at 81bdd8343 [toup] zephyr: Fix the IPv4 address in net_if
=== updating libmetal (modules/hal/libmetal):
HEAD is now at 243eed5 lib: update libmetal to SHA f55c02a7b582
=== updating liblc3 (modules/lib/liblc3):
HEAD is now at 1a5938e meson: Bump to version 1.0.4
=== updating littlefs (modules/fs/littlefs):
HEAD is now at 408c16a Merge pull request #14 from de-nordic/update-to-v2.8.1
=== updating loramac-node (modules/lib/loramac-node):
HEAD is now at 1bf2120c mac: region: RegionCommon: fix implicit double promotion
=== updating lvgl (modules/lib/gui/lvgl):
HEAD is now at 2b498e6f Merge pull request #51 from faxe1008/upddate_lvgl_840
=== updating mbedtls (modules/crypto/mbedtls):
HEAD is now at 6ec4abdcd Merge pull request #52 from ceolin/mbedtls-352
=== updating mcuboot (bootloader/mcuboot):
HEAD is now at 78bfe750 docs: Fold additional release notes
=== updating mipi-sys-t (modules/debug/mipi-sys-t):
HEAD is now at a819419 library: allow building without wchar support
=== updating net-tools (tools/net-tools):
HEAD is now at cd2eb18 capture: Add CookedLinux support to net-capture.py
=== updating nrf_hw_models (modules/bsim_hw_models/nrf_hw_models):
HEAD is now at 123d37d RTC: Fix for TASKS_TRIGOVRFLW with stopped counter
=== updating open-amp (modules/lib/open-amp):
HEAD is now at da78aea lib: update open-amp lib to release v2023.10.0
=== updating openthread (modules/lib/openthread):
HEAD is now at 49c59ec51 [mbedtls] make debug logging independent from builtin (#9848)
=== updating percepio (modules/debug/percepio):
HEAD is now at 7f6fb3f Merge branch 'main' into zephyr
=== updating picolibc (modules/lib/picolibc):
HEAD is now at 764ef4e40 Version 1.8.6
=== updating segger (modules/debug/segger):
HEAD is now at b011c45 segger: config: Add CCM option for linker sections
=== updating tinycrypt (modules/crypto/tinycrypt):
HEAD is now at 1012a3e Fix warnings reported by UBSAN
=== updating trusted-firmware-m (modules/tee/tf-m/trusted-firmware-m):
HEAD is now at 0b898c9b7 Platform: nxp: Add initialization flag to the CMSIS USART driver.
=== updating trusted-firmware-a (modules/tee/tf-a/trusted-firmware-a):
HEAD is now at 421dc0502 Merge pull request #3 from SgrrZhf/topics/huizha01/2.9.0-updates
=== updating uoscore-uedhoc (modules/lib/uoscore-uedhoc):
HEAD is now at 150f4eb [zephyr fromlist] Adapt to changes in name generation in zcbor 0.8.0
=== updating zcbor (modules/lib/zcbor):
HEAD is now at 75d0880 tests: Update mps2 board name
ERROR: update failed for project hal_espressif
```

closed 2024-05-02 by fabiobaltieri

qwersem, 2024-05-02:
I faced with the same issue.

htibosch, 2024-10-15:
@jvermillard wrote:
~~~
To Reproduce
Steps to reproduce the behavior:

    in your zephyr directory: git pull
    west update
~~~
Today I still get mentioned error after cloning from main:
~~~
...
=== updating uoscore-uedhoc (modules/lib/uoscore-uedhoc):
HEAD is now at 84ef879 Merge pull request #4 from rlubos/upmerge-06-06-24
=== updating zcbor (modules/lib/zcbor):
HEAD is now at 47f34dd Update version to 0.9.0
ERROR: update failed for project hal_espressif
~~~
It doesn't seem to be related with [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/72186/files), because I see yet a different revision for `hal_espressif`.

aescolar, 2024-10-15:
> @jvermillard wrote....
> Today I still get mentioned error after cloning from main:

That may have been just a glitch, all repos seem to be fine at this point.

jvermillard, 2024-10-15:
Sometimes there is a hiccup when the sub-module is updated (maybe because it's updated in zephyr before being pushed on the esp side?)

htibosch, 2024-10-15:
OK, thank you both. As I don't use ESP, I removed the "hal_espressif" entry from west.yml to avoid the complaints.

PR #72186 (2024-05-02): west.yaml: update hal_espressif after SDHC porting
Update revision reference for PR #71165 as needed.

Fix #72196