zephyrproject-rtos/zephyr #72205: Build errors ESP32 when using display drivers SSD1306 or ST7789

zpm1066, 2024-05-02:
**Describe the bug**
Unable to build successfully ESP32-S3 firmware that uses Display Driver SSD1306 or ST7789. Previously, these built fine under Zephyr 3.6. The same code (with appropriate overlays) works fine for ARM boards.

**To Reproduce**
Steps to reproduce the behavior:
1. Build display driver sample for SSD1306 or ST7789 
2. Build fails.
3. See error below in "Logs and consoles output"

**Expected behavior**
Successful build.

**Impact**
Showstopper for firmware that uses ESP32 with SSD1306/ST7789 displays. 

**Logs and console output**
I get the following errors for several ESP32-S3 boards:

`warning: SSD1306 (defined at drivers/display/Kconfig.ssd1306:6) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: (DT_HAS_SOLOMON_SSD1306FB_ENABLED ||
DT_HAS_SINOWEALTH_SH1106_ENABLED) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SSD1306 and/or look up SSD1306 in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

...
...

  No SOURCES given to Zephyr library: drivers__display

  Excluding target from build.
`

**Environment (please complete the following information):**
 - macOS Big Sur and Monterey
 - Zephyr SDK
 - Latest Zephyr 3.6.99

**Additional context**
The same code with ARM boards (tested nRF52840, RP2040, STM42) builds and runs fine when used with SSD1306 or ST7789 displays.

ESP32-S3 boards with SSD1306 or ST7789 under Zephyr 3.5.xx works fine.

zpm1066, 2024-05-03:
After observing the introduction of qualifiers "esp32s3/procpu" in Zephyr 3.6.xx, in build of xiao_esp32s3 board, I renamed the "xiao_esp32s3.overlay" to "xiao_esp32s3_procpu.overlay". 

This has resolved the SSD1306 display devicetree issue but now throws the following errors when creating esp32s3 image.

`quark11:ssd1306_cfb_custom_fonts_v3.5 quark11$ west build -p -b xiao_esp32s3 -d build_xiao_esp32s3_3.6
-- west build: making build dir /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6 pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5
-- CMake version: 3.29.1
-- Found Python3: /usr/local/opt/python@3.11/bin/python3.11 (found suitable version "3.11.7", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/quark11/Library/Caches/zephyr
-- Zephyr version: 3.6.99 (/Users/quark11/zephyrproject360/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
CMake Warning at /Users/quark11/zephyrproject360/zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=xiao_esp32s3 specified, board automatically changed to:
  xiao_esp32s3/esp32s3/procpu.
Call Stack (most recent call first):
  /Users/quark11/zephyrproject360/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /Users/quark11/zephyrproject360/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/quark11/zephyrproject360/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Board: xiao_esp32s3, qualifiers: esp32s3/procpu
-- Found host-tools: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found Dtc: /usr/local/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6")
-- Found BOARD.dts: /Users/quark11/zephyrproject360/zephyr/boards/seeed/xiao_esp32s3/xiao_esp32s3_procpu.dts
-- Found devicetree overlay: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/boards/xiao_esp32s3_procpu.overlay
-- Generated zephyr.dts: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/include/generated/devicetree_generated.h
^C
quark11:ssd1306_cfb_custom_fonts_v3.5 quark11$ west build -p -b xiao_esp32s3/esp32s3/procpu -d build_xiao_esp32s3_3.6
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5
-- CMake version: 3.29.1
-- Found Python3: /usr/local/opt/python@3.11/bin/python3.11 (found suitable version "3.11.7", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/quark11/Library/Caches/zephyr
-- Zephyr version: 3.6.99 (/Users/quark11/zephyrproject360/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: xiao_esp32s3, qualifiers: esp32s3/procpu
-- Found host-tools: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found Dtc: /usr/local/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6")
-- Found BOARD.dts: /Users/quark11/zephyrproject360/zephyr/boards/seeed/xiao_esp32s3/xiao_esp32s3_procpu.dts
-- Found devicetree overlay: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/boards/xiao_esp32s3_procpu.overlay
-- Generated zephyr.dts: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/dts.cmake

Parsing /Users/quark11/zephyrproject360/zephyr/Kconfig
Loaded configuration '/Users/quark11/zephyrproject360/zephyr/boards/seeed/xiao_esp32s3/xiao_esp32s3_procpu_defconfig'
Merged configuration '/Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/prj.conf'
Configuration saved to '/Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/.config'
Kconfig header saved to '/Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/zephyr-sdk-0.16.5/xtensa-espressif_esp32s3_zephyr-elf/xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.16.5/xtensa-espressif_esp32s3_zephyr-elf/bin/xtensa-espressif_esp32s3_zephyr-elf-gcc
esptool path: /Users/quark11/zephyrproject360/modules/hal/espressif/tools/esptool_py/esptool.py
-- Using ccache: /usr/local/bin/ccache
-- Configuring done (7.5s)
-- Generating done (0.4s)
-- Build files have been written to: /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6
[... cut ...]
Info: inserting 4 bytes padding between device_area and _static_thread_data_area
Merged 17 ELF sections

A fatal error occurred: Segment loaded at 0x3c0134b8 lands in same 64KB flash mapping as segment loaded at 0x3c010000. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/quark11/Documents/2024_zephyr_dev/ssd1306_cfb_custom_fonts_v3.5/build_xiao_esp32s3_3.6
quark11:ssd1306_cfb_custom_fonts_v3.5 quark11$ `

fgihl, 2024-05-03:
Try adding `--sysbuild` to build command. That help me, when I had the same issue.

zpm1066, 2024-05-03:
@fgihl Thanks. Unfortunately, "--sybuild" didn't help. I tried a "git pull" and "west update", no change.

sylvioalves, 2024-05-05:
@fgihl CAn you share both .conf and .overlay files?

fgihl, 2024-05-06:
@sylvioalves 

Will it help you if you looked at my sample program? 
`west init -m git@github.com:fgihl/tenderizer.git --mr fgihl/esp32c3-sample`
`west build -p -b esp32c3_devkitm tenderize`

This compiles fine on Zephyr v3.6.0, but failes with Zephyr-main with the link error
`A fatal error occurred: Segment loaded at 0x3c0134b8 lands in same 64KB flash mapping as segment loaded at 0x3c010000. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.`

If I compile latest zephyr with the command 
`west build -p -b esp32c3_devkitm --sysbuild tenderizer`

With risk of hijacking the issue with other problems, I should probaby report is seperatly... 
But when starting I get the following runtime errors:
`W (370) tsens: legacy driver is deprecated, please migrate to `driver/temperature_sensor.h`
W (378) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h``
And neither ADC or internal tempsensor works.

zpm1066, 2024-05-06:
@sylvioalves 

Here are proj and overlay being used with target xiao_esp3s3 and Zephyr 3.6.99. The proj/overlay work fine with Zephyr 3.5.

```
CONFIG_HEAP_MEM_POOL_SIZE=16384
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_LOG=y
CONFIG_DISPLAY=y

CONFIG_GPIO=y
CONFIG_PRINTK=y
CONFIG_LOG_PRINTK=y

CONFIG_I2C=y
CONFIG_SSD1306=y

CONFIG_DISPLAY_LOG_LEVEL_ERR=y
CONFIG_STDOUT_CONSOLE=y

CONFIG_CFB_LOG_LEVEL_DBG=y
CONFIG_CHARACTER_FRAMEBUFFER=y
CONFIG_CHARACTER_FRAMEBUFFER_USE_DEFAULT_FONTS=n

# Preferred SHELL options
CONFIG_SHELL=y
#CONFIG_MCUMGR_CMD_SHELL_MGMT=y
CONFIG_BOOT_BANNER=y
CONFIG_DATE_SHELL=y
CONFIG_POSIX_CLOCK=y
CONFIG_DEVICE_SHELL=y
CONFIG_INIT_STACKS=y
CONFIG_KERNEL_SHELL=y
CONFIG_SHELL_STATS=n

# Enable RTT to replace UART
# set CONFIG_UART_CONSOLE=n in particle_xenon_defconfig
# error setting twice if below uncommented
# uncomment to enable uart console
# CONFIG_UART_CONSOLE=y
CONFIG_USE_SEGGER_RTT=n
CONFIG_SHELL_BACKEND_RTT=n
```

overlay for SSD1306:

```
 / {
	chosen {
		zephyr,display = &ssd1306;
	};

};

&pinctrl {
	i2c0_default: i2c0_default {
		group1 {
			pinmux = <I2C0_SDA_GPIO41>,
				 <I2C0_SCL_GPIO40>;
			bias-pull-up;
			drive-open-drain;
			output-high;
		};
	};
};

&i2c0 {
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	ssd1306: ssd1306@3c {
		compatible = "solomon,ssd1306fb";
		reg = <0x3c>;
		width = <128>;
		height = <64>;
		segment-offset = <0>;
		page-offset = <0>;
		display-offset = <0>;
		multiplex-ratio = <63>;
		segment-remap;
		com-invdir;
		prechargep = <0x22>;
	};
};

```

closed 2024-07-21 by github-actions[bot]

sylvioalves, 2024-08-26:
@fgihl, for some reason I missed following this. Have you got any success? We have made some improvements and that might have fixed the issue. Can you re-check?

zpm1066, 2024-08-26:
SSD1306 and ST7789 work under Zephyr 3.7.x. However, SSD1306 text is inverted. I had to correct it in ssd1306.c/ssd1306_init_device().

sylvioalves, 2024-08-26:
> com-invdir;

What about adding `inversion-on;` in SSD1306 .dts properties?

zpm1066, 2024-08-26:
> > com-invdir;
> 
> What about adding `inversion-on;` in SSD1306 .dts properties?

I did also try `inversion-on` in SSD1306 .dts properties but it didn't work.

rhvarrier, 2025-01-28:
Hello,
@zpm1066 and @sylvioalves  has there been any updates concerning this issue ?